thesofproject/sof #8006: [RFC] Use not-cached shared memory alias instead of coherent.h in case of shared buffers

marcinszkudlinski, 2023-08-03:
As everybody knows, not coherent cache is a serious complication we must deal with. 

There are 2 ways of share data between cores:
1) Using non-cached alias - read/write directly from L2 RAM 
2) Using a cached alias and a mechanism of writeback/invalidation to ensure coherent data between cores

Currently Zephyr uses solution 1, SOF uses solution 2

Shared memory in general has a lot of advantages - it is easy to use, the core using it is "obvious correct" instead of "correct". The structures may be used by several cores at the same time (standard multithread rules for sharing data apply of course) 

The question is - **how much it is slower?** 

I made some measurements on MTL RVP - access to the data using several methods:
1) not shared data. 2 ways were measured:
 - cached memory ensuring cache hit 
 - using coherent.h in not-shared mode

2) shared data - 3 ways were measured:
- not cached memory alias
- using coherent.h in shared mode
- single invalidation before operations / writeback after operations

all results in CPU cycles

NOT SHARED MEMORY

|ID|desc|Cached (hit)|coherent.h|
|--|--|--|--|
|1|read single dword|30|45|
|2|read double dword|31|44|
|3|read cacheline (16 dword)|80|98|
|4|read 64dwords|226|257|
|5|read 1024dword|3132|3449|
|6|read twice 1024|6234|6852|
|7|write single dword|30|44|
|8|write double dword|31|43|
|9|write cacheline (16dword)|84|43|
|10|write 64 dwords|195|211|
|11|wrtite 1024 dwords|2632|2748|
|12|twice write 1024 dwords|5535|5749|

no surprise here - using of direct access to memory is faster

SHARED MEMORY

|ID|desc|uncahced|cached (with inv/wb)|coherent.h (shared)|
|--|--|--|--|--|
|1|read single dword|35|38|119|
|2|read double dword|35|39|119|
|3|read cacheline (16 dword)|115|87|175|
|4|read 64dwords|344|234|336|
|5|read 1024dword|5034|3177|3575|
|6|read twice 1024|10036|6279|6972|
|7|write single dword|32|37|121|
|8|write double dword|32|40|119|
|9|write cacheline (16dword)|76|81|172|
|10|write 64 dwords|200|208|302|
|11|wrtite 1024 dwords|2636|2749|2955|

!!!!!!!!!!!!!!!!!! USING OF UNSHARED MEMORY ALIAS IS FASTER THAN COHERENT.H UNLESS WE WANT TO ACCESS MORE THAN 64 DWORDS!!!!!!!!!!!!!!!!!! 

"64 dwords" means - 64 accesses from acquire till release. Probably 99% (or even 100%) of all accesses to structures like comp_buffer are like this. 
**I don't mean access to the audio samples in buffers - only to control structures** common usage of buffer is:
```
sink_c = buffer_acquire(sink);
min_free_frames = MIN(min_free_frames, audio_stream_get_free_frames(&sink_c->stream));
buffer_release(sink_c);
```
in fact a worse case - 1 dword access

**CONCLUSIONS**

My proposition is to stop using coherent.h for comp_buffer/audio_stream, instead create those structures in shared (non-cached) or not-shared(cached) memory

Good news is that we do know if the buffer will be shared or not at allocation time. In current code marking a buffer as shared is performed by calling `coherent_shared_thread`, in my solution it would be by choosing proper memory alias

1) cross core safety - exactly same as in current solution. Currently if a buffer would be used on several cores without marking it as shared the system will crash. Same in my approach - cached must be used on single core exclusively
2) performance - better (!!!!) than currently 
 - in cased of not shared buffers the gain of performance is obvoius
 - even in case of shared buffers - probably 99% (or even 100%) of accesses will be smaller than 64 dword operations. That means we  
- shared buffers are rare in the system - currently 100% of buffers are not shared. Shared buffers will be used on cross core connected pipelines and in cross-core DP module ONLY. 
 - so - we do gain performance in every case
3) code is WAY simpler and WAY more easy to develop/debug. That means - less cache related defect in the future (and those defects are extremely hard to track)

Of course the above don't apply to audio samples in buffers - access to sample streams must be cached with proper inv/writebacks. BUT currently buffer_acquire and buffer_release DO NOT do any cache releated operations on audio buffers themselves. So change I'm proposing does not affect it at all.

marcinszkudlinski, 2023-08-07:
Correct me if I'm wrong - in IPC3 there's an IPC for buffer creation _SOF_IPC_TPLG_BUFFER_NEW_ and at this moment there's no information about components the buffer will be bind to?

lgirdwood, 2023-08-07:
@marcinszkudlinski good data. Its interesting to see the difference between coherent shared and the wb/inv columns as the difference should be spinlock and some asserts. 

Can you share your proposed new API where we have lockless buffer readers and writers.

Do we have any other (non buffer) cache shared data that can be changed to uncached (and made lockless where appropriate) ?

btian1, 2023-08-08:
for me, I definitely support remove those cache/uncache operations, it makes SOF hard to read and understand, especially current APIs have so much versions(ipc3/ipc4/xtensa).

From performance, actually, our current module performance is still too high, we need spend big effort to make it better and better, if there is no acquire/release/invalidate/writeback for most of modules, then I definitely believe performance will be better.

Another topic is module peak mcps, still no rootcause, one big concern is those cache related operations, another is logging, once those operation are removed, peak mcps rootcause will be easier to find.

@marcinszkudlinski , could you share your sample code here? I want to learn something from your code, since I did not have a chance to touch zephyr yet.

marcinszkudlinski, 2023-08-08:
@kv2019i 

The best approach is "use uncached for all data, cached for code". Nobody can deny that skipping cache is the easiest and 100% bug-proof way to solve all cache problems

This is exactly how Zephyr does it.

BUT - zephyr is SMP system, we have a perfect core separation, don't we? Threads, components, pipelines etc. are bind to single core and never moved from one to another.
That means - **almost everything may be cached**, as long as creation/usage/destruction is performed on the same core it is perfectly safe

All "cache sensitve" places are easy to identify:
 - cross core queues for DP and connected pipelines
 - IDC
 - system operations like loading modules, memory drop to IMR
 - DMA operations
 
and that's all. Those may either be uncached OR have smart cache operations. And again - unless there's a significant performance drop - the most easy and most bug-resistant way is to use it uncached

you say "use in diffent way" - that's why there **MUST NOT** be any extra "smart" cache operations in the code, 99% of the code must have a ready to use and safe pointers. The modules **MUST NOT** even know that the pointers are cached or uncached - this is up to queue/IDC/driver code

this way won't give any chance to "use diffrently"

lyakh, 2023-08-14:
Thanks @marcinszkudlinski for the data and the write-up! One interesting aspect of the data I noticed that I haven't seen being mentioned in the discussion yet - is a discrepancy between reading and writing, e.g. lines 5 and 11 represent the same test with the only difference being using reads instead of writes, and it looks like with (the most deterministic) uncached access reads are twice as slow as writes? Is some write buffering taking place there?
As for the coherent API use - as I started working with it, I immediately raised a concern, that it's very complex and therefore expensive in terms of developer time and error-prone, and potentially not even beneficial for performance. I also thought that simple uncached access to shared objects like buffer control structures would be both much simpler to work with and possibly not even slower. However we had cycle counts, needed for SRAM and for cache access and it has been estimated that using the coherent API would bring performance benefits. So, in general - yes, I absolutely agree that the coherent API is complex, expensive to maintain and error prone, and if it doesn't provide performance gains, we probably don't need it.
OTOH I think it would be useful to actually do some close to real use-case measurements: can we benchmark some simple capture or playback streaming with the coherent API, while forcing the `.shared` flag to `true` and with uncached access to `struct comp_buffer`?

lgirdwood, 2023-08-16:
@marcinszkudlinski btw, is there a high level FEATURE tracking all these parts and dependecies ?

cujomalainey, 2023-09-22:
@marcinszkudlinski excellent experiment! Would you be able to share the code to do the test? I feel we should validate on other SoCs. That we can validate this is either a global truth or a subjective one to design.

cujomalainey, 2023-09-22:
And by other I mean @thesofproject/amd @thesofproject/nxp and @thesofproject/mediatek

lyakh, 2023-09-25:
> @marcinszkudlinski excellent experiment! Would you be able to share the code to do the test? I feel we should validate on other SoCs. That we can validate this is either a global truth or a subjective one to design.

@cujomalainey it was my impression that Intel DSP was the only architecture with incoherent caches, this my impression is based on the definitions by all other architectures:
```
#define uncache_to_cache(address)	address
#define cache_to_uncache(address)	address
```
In which case all the Intel cache-uncache handling should only be relevant for Intel?

lgirdwood, 2023-09-29:
@marcinszkudlinski I guess we are now finished with this update. i.e. all PRs now merged ?

marcinszkudlinski, 2023-09-29:
yes, task completed!

closed 2023-09-29 by lgirdwood