thesofproject/sof #8022: [BUG] [MTL] Playback clips captured from mock RTC AEC device is double speeded

yongzhi1, 2023-08-09:
**Describe the bug**

Playback the file captured from "device 27: DMIC0 RTC AEC" with CONFIG_GOOGLE_RTC_AUDIO_PROCESSING_MOCK=y, and hear the speed is doubled.

**To Reproduce**

Start spk playback:
```
aplay -Dhw:0,1 -c2 -r48000 -fS16_LE /home/LR.wav -vv
```
Start spk feedback capture:
```
arecord -Dhw:0,29 /tmp/tmp.wav -c2 -r48000 -fs32_le -vv
```
Start AEC capture:
```
arecord -Dhw:0,27 /tmp/mock_aec.wav -c4 -r48000 -fs32_le -vv -d20
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1}](https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1%7D)
    * SOF: {MTL-005.0.3 release} https://github.com/thesofproject/sof/releases/tag/releases%2Fmtl%2Fv5.0.3
2) Name of the topology file
    * Topology: {sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL/Rex}

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). 

[mtrace.txt](https://github.com/thesofproject/sof/files/12302788/mtrace.txt)
[dmesg-mock-aec.txt](https://github.com/thesofproject/sof/files/12302795/dmesg-mock-aec.txt)
[mock_aec.zip](https://github.com/thesofproject/sof/files/12302803/mock_aec.zip)

lgirdwood, 2023-08-16:
@yongzhi1 can you confirm if its SPK feedback or AEC that is double speed or both ?
@mwasko @abonislawski fyi.

abonislawski, 2023-08-21:
Please verify on newest chromeos, I cannot reproduce on main and 005

RDharageswari, 2023-08-22:
hi Adrian,
i can repro the issue still on mtl-005

serhiy-katsyuba-intel, 2023-08-28:
Looks like the problem is caused by wrong host-copier.27.capture output format configuration. It is configured to receive 2-channel audio on source and output 4-channel audio to host. See dmesg log:
```
[  666.200835] sof-audio-pci-intel-mtl 0000:00:1f.3: Init input audio formats for host-copier.27.capture
[  666.200836] sof-audio-pci-intel-mtl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002) buffer size 384
[  666.200837] sof-audio-pci-intel-mtl 0000:00:1f.3: Output audio format for host-copier.27.capture
[  666.200838] sof-audio-pci-intel-mtl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffff3210 ch_cfg 4 interleaving_style 0 fmt_cfg 0x2004) buffer size 768
[  666.200839] sof-audio-pci-intel-mtl 0000:00:1f.3: copier host-copier.27.capture, IPC size is 84
[  666.200839] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC match in the firmware file's manifest (ibs/obs: 384/768)
[  666.217456] sof-audio-pci-intel-mtl 0000:00:1f.3: host-copier.27.capture: ibs / obs / cpc: 384 / 768 / 0
```

Current version of copier does not support remapping of channels. So, basically, host-copier.27.capture just copy its input data to output DMA buffer and so host receives 2 times less data when expected.

Is it a mistake that host-copier.27.capture is configured with 4-channel output instead of 2? Or is this intestinal and for some reason 4 channels are required?

cc: @ranj063.

lgirdwood, 2023-08-30:
@RDharageswari I'm assuming the 4 channel output is not needed by userspace here ?

ranj063, 2023-08-31:
@yongzhi1 @RDharageswari Do you still see this issue with #8101

ranj063, 2023-09-01:
@serhiy-katsyuba-intel the current aec pipeline in topology is missing support for 4ch capture. I am working with @yongzhi1 to fix this.

yongzhi1, 2023-09-06:
With https://github.com/thesofproject/sof/pull/8101, https://github.com/thesofproject/sof/pull/8172 and https://github.com/thesofproject/linux/pull/4550, the issue can be closed, thanks @ranj063 for the prompt support!!

closed 2023-09-06 by yongzhi1