thesofproject/sof #8054: [BUG][sparse] mm_drv_intel_adsp_tlb.c:147:20: warning: incorrect type in initializer (different address spaces)

kv2019i, 2023-08-17:
Seen in CI since this week:
mm_drv_intel_adsp_tlb.c:147:20: warning: incorrect type in initializer (different address spaces)

https://github.com/thesofproject/sof/actions/runs/5878948715/job/15942028924?pr=7986

kv2019i, 2023-08-17:
@jxstelter @lyakh Not sure yet why this is popping up now, but let's try to fix asap so more sparse issued don't slip through (while CI is red on this).

jxstelter, 2023-08-17:
> @jxstelter @lyakh Not sure yet why this is popping up now, but let's try to fix asap so more sparse issued don't slip through (while CI is red on this).

The warnings looks strange. Zephyr is set to this commit:
commit 431108d89e1750dbea05dce64dbb7fd329aa610e (HEAD -> Branch_431108d8)

Code lines marked shows even comment space (eg. line 204)

lyakh, 2023-08-18:
> @jxstelter @lyakh Not sure yet why this is popping up now, but let's try to fix asap so more sparse issued don't slip through (while CI is red on this).

@kv2019i isn't it because we've now enabled the MM remapping on TGL? https://github.com/thesofproject/sof/pull/7986/commits/22f1562050c7d5262b7cfbc8c0ce421308ffa548 @ujfalusi

ujfalusi, 2023-08-18:
The SOF side fix (seen in MTL/LNL builds): https://github.com/thesofproject/sof/pull/8058

ujfalusi, 2023-08-21:
The Zephyr side: https://github.com/zephyrproject-rtos/zephyr/pull/61630

kv2019i, 2023-08-29:
Issue fixed, closing.

closed 2023-08-29 by kv2019i