thesofproject/sof #8073: [BUG] sof_ri_info.oy fails on Intel MTL+ binaries

kv2019i, 2023-08-22:
**Describe the bug**
Following error is seen when using the script on recent MTL binaries:

$ ./sof_ri_info.py sof-mtl.ri 
Traceback (most recent call last):
  File "./sof_ri_info.py", line 1717, in <module>
    main(ARGS)
  File "./sof_ri_info.py", line 1694, in main
    fw_bin = parse_fw_bin(args.sof_ri_path, args.no_cse, args.verbose)
  File "./sof_ri_info.py", line 906, in parse_fw_bin
    parsed_bin.add_comp(parse_cse_manifest(reader))
  File "./sof_ri_info.py", line 681, in parse_cse_manifest
    entry_file_offset + entry_length)
  File "./sof_ri_info.py", line 704, in parse_css_manifest
    return parse_css_manifest_4(css_mft, reader, limit)
  File "./sof_ri_info.py", line 769, in parse_css_manifest_4
    assert reader.get_offset() == size_limit # wrong extension length
AssertionError

**To Reproduce**

**Reproduction Rate**

**Expected behavior**

**Impact**

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: n/a
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL

marc-hb, 2023-08-24:
Known issue: 842c39fb0f69

kv2019i, 2023-08-24:
Issue rootcaused, fix in https://github.com/thesofproject/rimage/pull/173

kv2019i, 2023-08-25:
Reopening until rimage update is done to SOF main with this fix.

closed 2023-08-28 by CBala21

marc-hb, 2023-08-28:
Thanks @kv2019i ! Enabling `sof_ri_info.py` on MTL and LNL at last in:
- #8107