thesofproject/sof #8075: [BUG] [MTL] AEC functionality depends on Echo reference endpoint to be started

RDharageswari, 2023-08-22:
With the AEC topology on main, the following endpoints get listed.
i see the following o/p:
rex-rev1 ~ # arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: Headset () []
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 2: Bluetooth () []
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 27: DMIC0 RTC AEC () []
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 28: DMIC0 RTC AEC NS () []
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 29: EchoRef () []
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 99: DMIC Raw () []
Subdevices: 1/1
Subdevice #0: subdevice #0
rex-rev1 ~ #

This sequence doesnt work:
1)start speaker playback
2)start dmic capture: arecord -Dhw:0,27 /tmp/tmp.wav -c4 -r48000 -fs32_le -vv

there is a strong dependency on echo reference endpoint to be started.
While this works:
1)start speaker playback
2)start echo ref capture
3)start dmic capture: arecord -Dhw:0,27 /tmp/tmp.wav -c4 -r48000 -fs32_le -vv.

There is no need for the echo reference endpoint to be opened from the user space, while speaker playback the data can be fed to AEC without the FE.Kind of hostless stream.

lgirdwood, 2023-08-23:
@RDharageswari the inention is that connected PCM endpoints are meant to be started in any order with any timing, since its all driven by userspace. There is a topology flag that's meant to tell any share modules to ignore internal XRUN errors in such cases and I'm wondering if we dont have this set (as this was a common issue when enabled with IPC3 + M4).

kv2019i, 2023-08-23:
@RDharageswari I need to investigate the topology definitions, but it does sound this is working as expceted. The long history and rationale is documented here:
https://github.com/thesofproject/sof/issues/2564

But, but, if the AEC is on the DSP, then we actually should not have the user-space echo ref PCM in topology (or it can be there, but then it has to be used).

kv2019i, 2023-08-23:
@yongzhi1 It seems the speaker ref part of the topology is from you? Is the speaker ref to user-space here intentionally?

ranj063, 2023-08-24:
@kv2019i I know the root cause for this issue. I am in the provess of removing the host PCM 27 for echo ref adn this should take of this issue as well.

yongzhi1, 2023-08-25:
> @yongzhi1 It seems the speaker ref part of the topology is from you? Is the speaker ref to user-space here intentionally?
Yes, the original code is to enable feedback path from speaker, same idea for SDW_AMP_FEEDBACK.

ranj063, 2023-08-25:
@yongzhi1 @RDharageswari PR for the modified topology is here: https://github.com/thesofproject/sof/pull/8101

This change also requires a kernel change. I will update when I post the PR

RanderWang, 2023-09-22:
Please include these patches for this feature
Rimage side: https://github.com/thesofproject/rimage/pull/180
Topology: https://github.com/thesofproject/sof/pull/8101, https://github.com/thesofproject/sof/pull/8246, https://github.com/thesofproject/sof/pull/8204
Kernel: https://github.com/thesofproject/linux/pull/4593

mengdonglin, 2023-09-25:
> Please include these patches for this feature Rimage side: [thesofproject/rimage#180](https://github.com/thesofproject/rimage/pull/180) Topology: #8101, #8246, #8204 Kernel: [thesofproject/linux#4593](https://github.com/thesofproject/linux/pull/4593)

@RDharageswari @ranj063 FYI

kv2019i, 2023-10-10:
@RDharageswari @RanderWang @ranj063 can we close this as all the patches referred here are now merged?

RanderWang, 2023-10-12:
Let me close it

closed 2023-10-12 by RanderWang