thesofproject/sof #8091: [BUG] Build failed on MTL/LNL with github action

aiChaoSONG, 2023-08-25:
**Describe the bug**

In PR build, github action report below build failures:

```
ERROR: /soc/ssp@2d000 POST_KERNEL 42 88 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/ssp@2c000 POST_KERNEL 42 87 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/ssp@2b000 POST_KERNEL 42 86 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/ssp@2a000 POST_KERNEL 42 85 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/ssp@29000 POST_KERNEL 42 84 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/ssp@28000 POST_KERNEL 42 83 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/dmic1@10000 POST_KERNEL 70 75 < /soc/dfpmccu@71b00/hub_ulp_domain POST_KERNEL 75 73
ERROR: /soc/dmic0@10000 POST_KERNEL 70 74 < /soc/dfpmccu@71b00/hub_ulp_domain POST_KERNEL 75 73
ERROR: /soc/dma@72400 POST_KERNEL 40 66 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
ERROR: /soc/dma@7e000 POST_KERNEL 40 72 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/dma@7d000 POST_KERNEL 40 71 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
ERROR: /soc/dma@7c000 POST_KERNEL 40 82 < /soc/dfpmccu@71b00/hub_ulp_domain POST_KERNEL 75 73
ERROR: /soc/dma@72c00 POST_KERNEL 40 69 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
ERROR: /soc/dma@72800 POST_KERNEL 40 68 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
ERROR: /soc/dma@72600 POST_KERNEL 40 67 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
```

Link: https://github.com/thesofproject/sof/actions/runs/5966542268/job/16186380637?pr=8090
https://github.com/thesofproject/sof/actions/runs/5964429072/job/16179610729

aiChaoSONG, 2023-08-25:
I am not able to reproduce it locally, even run the same command `./sof/scripts/xtensa-build-zephyr.py --no-interactive --cmake-args=-DEXTRA_CFLAGS=-Werror --cmake-args=-DEXTRA_CXXFLAGS=-Werror --cmake-args=-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings --cmake-args=--warn-uninitialized mtl`

marc-hb, 2023-08-30:
More info in duplicate #8096.

I think this has been fixed and can be closed now.

marc-hb, 2023-08-30:
What is worrying is why you could not reproduce. Did you run west update at the time?

aiChaoSONG, 2023-08-31:
fixed, see #8096, close

closed 2023-08-31 by aiChaoSONG