thesofproject/sof #8104: [BUG] capture test failed with dai XRUN

keqiaozhang, 2023-08-28:
**Describe the bug**
Observed this issue when testing tplg `development/sof-hda-benchmark-generic-mtl.tplg`.  it failed immediately with dai XRUN. No errors in kernel log.

**To Reproduce**
arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null -v -q

**Reproduction Rate**
100%

**TPLG**

[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[1083fe342f46](https://github.com/thesofproject/linux/commit/1083fe342f4654ba18446d9d4955d02560849807)}
    * SOF: {[7b52664a6a65](https://github.com/thesofproject/sof/commit/7b52664a6a655d295c42fb7bfb4d235a52bb48e2)}
2) Name of the topology file
    * Topology: {development/sof-hda-benchmark-generic-mtl.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTLP-RVP-HDA}

**Screenshots or console output**
```
2023-08-28 07:34:16 UTC [REMOTE_COMMAND] arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null -v -q
Hardware PCM card 0 'sof-hda-dsp' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
arecord: pcm_read:2221: read error: Input/output error
2023-08-28 07:34:17 UTC [REMOTE_ERROR] arecord on PCM hw:0,0 failed at 1/3.
```

**mtrace**
```
[    0.380988] <inf> mem_allocator: heap_alloc_aligned: heap allocated: 25224 free: 825672 max allocated: 39760
[    0.381000] <inf> mem_allocator: heap_alloc_aligned: heap allocated: 25256 free: 825624 max allocated: 39760
[    0.381010] <inf> mem_allocator: heap_alloc_aligned: heap allocated: 25288 free: 825576 max allocated: 39760
[    0.381016] <inf> copier: copier_prepare: comp:0 0x40000 copier_prepare()
[    0.381030] <inf> pipe: pipeline_trigger: pipe:0 0x0 pipe trigger cmd 7
[    0.381040] <inf> ll_schedule: zephyr_ll_task_schedule_common: task add 0xa0110f80 0x201e0U priority 0 flags 0x0
[    0.381068] <inf> ll_schedule: zephyr_domain_register: zephyr_domain_register domain->type 1 domain->clk 3 domain->ticks_per_ms 38400 period 1000
[    0.381243] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.387245] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.388180] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.389255] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_common_copy(): Copy_bytes 0 + free bytes 0 < period bytes 384, possible glitch
[    0.389260] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.390170] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_common_copy(): Copy_bytes 0 + free bytes 0 < period bytes 384, possible glitch
[    0.390183] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.391183] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_common_copy(): Copy_bytes 0 + free bytes 0 < period bytes 384, possible glitch
[    0.391188] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.392171] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_common_copy(): Copy_bytes 0 + free bytes 0 < period bytes 384, possible glitch
[    0.392193] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
[    0.393183] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_common_copy(): Copy_bytes 0 + free bytes 0 < period bytes 384, possible glitch
[    0.393195] <wrn> dai_comp: dai_common_copy: comp:0 0x40000 dai_zephyr_copy(): nothing to copy
```
[dmesg.txt](https://github.com/thesofproject/sof/files/12451891/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/12451892/mtrace.txt)

aiChaoSONG, 2023-08-29:
#8105 didn't fix the issue. the issue is a little bit complex:

If do capture after playback, there is io error on capture pcm, but if only do capture after reboot, no io error (xrun) happenned whatever times capture is started. After IO error on capture, do playback still works fine.

lgirdwood, 2023-08-30:
> #8105 didn't fix the issue. the issue is a little bit complex:
> 
> If do capture after playback, there is io error on capture pcm, but if only do capture after reboot, no io error (xrun) happenned whatever times capture is started. After IO error on capture, do playback still works fine.
> 
> If run aplay while arecord is running on HDA Analog, aplay will get io error. and thereafter, aplay/arecord could never run again.

@aiChaoSONG can you confirm if RTD3 transition fixes the issue or does it still persist ? What about if SOF modules reloaded ? 
i.e. is HW getting into an error state OR (more likely) is the error logic that handles the XRUN missing IP programming sequqnces to cleanly shut down HDA link.

mengdonglin, 2023-09-01:
@lgirdwood @ranj063   I learned from @aiChaoSONG that ARIA module doesn't use ipc4_base_module_cfg_ext but its own module-specific extension, so current module initialization is wrong for ARIA. 

Since ARIA is not used by Chrome, so we can lower the issue priority.

mengdonglin, 2023-11-08:
@keqiaozhang @ranj063 No longer reproduced in MTL daily test. Closing.

closed 2023-11-08 by mengdonglin