thesofproject/sof #8115: [BUG] IPC timeout check mismatch between kernel and fw

RanderWang, 2023-08-29:
Linux driver set ipc timeout duration to 500ms, but FW wait 5000ms in cpu_disable_core for SET_DX ipc message. This makes driver stop but fw is still waiting.  And why we need 5000ms to disable core 1 ~ n ? Can we adopt 400 or 300 ms ?

````c
void cpu_disable_core(int id) {
       .....
	uint64_t timeout = k_cycle_get_64() +
		k_ms_to_cyc_ceil64(CONFIG_SECONDARY_CORE_DISABLING_TIMEOUT); // 5000ms

	/* Waiting for secondary core to enter idle state */
	while (arch_cpu_active(id) && (k_cycle_get_64() < timeout))
		idelay(PLATFORM_DEFAULT_DELAY);

	if (arch_cpu_active(id)) {
		tr_err(&zephyr_tr, "core %d did not enter idle state", id);
		return;
	}
     .....
}
````

kv2019i, 2023-08-29:
Labeling P2 for investigation, need to check with git blame on what is history of this long timeout.

RanderWang, 2023-08-29:
normally only needs less than 1ms
````
[    9.832556] <err> zephyr: cpu_disable_core: ~~~~~~~~ core[2] entry
[    9.832568] <inf> os: pm_state_set: --------------- power off core[2] 
[    9.832585] <err> zephyr: cpu_disable_core: ^^^^^^^^^^ core[2] exit

[    9.835976] <err> zephyr: cpu_disable_core: ~~~~~~~~ core[1] entry
[    9.835986] <inf> os: pm_state_set: --------------- power off core[1] 
[    9.836003] <err> zephyr: cpu_disable_core: ^^^^^^^^^^ core[1] exit
````

kv2019i, 2023-08-29:
@RanderWang git blame is your friend.. looking at history, the 5000 comes from this commit from @tmleman 
```
commit b6a4bb682cade0f4eb7a9ffe872f6fe46e099a2e
Author: Tomasz Leman <tomasz.m.leman@intel.com>
Date:   Fri Apr 29 16:37:07 2022 +0200

    platform: Kconfig: add timeout for disabling secondary core
    
    Count of loops used when waiting for secondary core to enter OFF state
    in transition to D3 state.
```

This is a generic setting applying to all platforms, so it's probably conservative because of that. Do you @tmleman any specific reason to have such a long time.

We should reduce this to under 500ms as this will now complicated debug. If the secondary core power-down fails, the kernel will hit IPC timeout sooner (and bring down the DSP as  error recovery). It'd be better to error out on the FW side and return a proper error that can be displayed in kernel log.

tmleman, 2023-08-29:
> Do you @tmleman any specific reason to have such a long time.

@kv2019i  timeout is set so high to avoid a situation where it is constantly being increased. It should never get to a timeout, so such a high value seemed appropriate. If we want to reduce this value, we should bear in mind that in the development environment (FPGA) this value will probably be higher than on silicon.

RanderWang, 2023-08-31:
> It looks like we need a Kconfig debug option were we multiple all delays by 10x (or whatever) to cater for all different development environments. 5 seconds is too long today as this could block userspace (frozen app on audio driver state transition). Can we lower to the real limit.

It is configurable in Kconfig. The default value is 5000, so we can change it when building fw

plbossart, 2023-08-31:
All the timeouts we use in the kernel are defined based on the recommendation of hardware folks. Specifically for SPA/CPA power-up, we have a recommendation to loop 100 times and wait for 10usec each time.

Clearly 5s timeout for core power management was not a hardware recommendation....

closed 2023-09-06 by kv2019i

lgirdwood, 2023-09-11:
@mwasko fyi needed for mtl06