thesofproject/sof #8137: [BUG] Reloading SOF module fails after playback on AMD Mendocino / Rembrandt

coolstar, 2023-08-31:
**Describe the bug**

AMD Mendocino / Rembrandt fails to load firmware if module / driver is reloaded after playing audio once through SOF.

What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
1. Load snd_sof_amd_rembrandt
2. Play audio
3. Unload snd_sof_amd_rembrandt
4. Reload snd_sof_amd_rembrandt

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

Happens all the time so long as audio is played. Note that if I skip `SOF_IPC_STREAM_TRIG_START` in my non-Linux custom driver (but leave the rest of the PCM setup / teradown in place) then the driver can successfully unload and reload just fine

**Expected behavior**
A clear and concise description of what you expected to happen.

Firmware should successfully reload

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

Trying to port SOF to a non Linux OS and enable runtime power management (which SOF on AMD appears to be missing, but even without that, hot reloading the driver is useful for development). Except it's not working if audio is played.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Ubuntu v6.5 mainline PPA as well as custom non-Linux driver based on [fd01a15164a15328fd96f9ce820f0fc9f700f00b](https://github.com/torvalds/linux/commit/fd01a15164a15328fd96f9ce820f0fc9f700f00b)
    * SOF: Pulled from chromeos_15436.63.0
2) Name of the topology file
    * Topology: sof-smb-rt5682s-rt1019.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: AMD Mendocino

**Screenshots or console output**

SOF driver reloaded at 55.86.8939 (with modprobe -r snd_sof_amd_rembrandt && modprobe snd_sof_amd_rembrandt)

```[   31.153976] snd_sof_amd_rembrandt 0000:05:00.5: unknown sof_ext_man header type 3 size 0x30
[   31.153986] snd_sof_amd_rembrandt 0000:05:00.5: Firmware info: version 2:5:0-b30d2
[   31.153989] snd_sof_amd_rembrandt 0000:05:00.5: Firmware: ABI 3:26:0 Kernel ABI 3:23:0
[   31.156854] snd_sof_amd_rembrandt 0000:05:00.5: Firmware info: version 2:5:0-b30d2
[   31.156866] snd_sof_amd_rembrandt 0000:05:00.5: Firmware: ABI 3:26:0 Kernel ABI 3:23:0
[   31.162873] snd_sof_amd_rembrandt 0000:05:00.5: Topology: ABI 3:26:0 Kernel ABI 3:23:0
[   31.163046] sof_mach rt5682s-hs-rt1019: ASoC: Parent card not yet available, widget card binding deferred
[   31.163053] snd_sof_amd_rembrandt 0000:05:00.5: ACP_HS config ACP1 channel 8 rate 48000 tdm_mode 1
[   31.163058] snd_sof_amd_rembrandt 0000:05:00.5: ACP_DMIC config ACP0 channel 4 rate 48000
[   31.163066] snd_sof_amd_rembrandt 0000:05:00.5: ACP_HS config ACP0 channel 8 rate 48000 tdm_mode 1
[   31.163073] snd_sof_amd_rembrandt 0000:05:00.5: ACP_HS config ACP0 channel 8 rate 48000 tdm_mode 1
[   31.173488]  acp-headset-codec: codec dai name = rt5682s-aif1
[   31.176819] sof_mach rt5682s-hs-rt1019: ASoC: driver name too long 'sof-rt5682s-hs-rt1019' -> 'sof-rt5682s-hs-'
[   31.178556] input: sof-rt5682s-hs-rt1019 Headset Jack as /devices/pci0000:00/0000:00:08.1/0000:05:00.5/rt5682s-hs-rt1019/sound/card1/input33
[   55.868939] snd_sof_amd_rembrandt 0000:05:00.5: unknown sof_ext_man header type 3 size 0x30
[   55.868948] snd_sof_amd_rembrandt 0000:05:00.5: Firmware info: version 2:5:0-b30d2
[   55.868951] snd_sof_amd_rembrandt 0000:05:00.5: Firmware: ABI 3:26:0 Kernel ABI 3:23:0
[   55.915606] snd_sof_amd_rembrandt 0000:05:00.5: SHA DMA Failed to Transfer Length c0
[   55.915616] snd_sof_amd_rembrandt 0000:05:00.5: SHA DMA transfer failed status: -110
[   55.915620] snd_sof_amd_rembrandt 0000:05:00.5: failed pre fw run op
[   55.915623] snd_sof_amd_rembrandt 0000:05:00.5: error: failed to boot DSP firmware -110
[   55.918056] snd_sof_amd_rembrandt 0000:05:00.5: error: sof_probe_work failed err: -110```

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2023-08-31:
@thesofproject/amd fyi

coolstar, 2023-08-31:
Issue is still present with SOF built from upstream v2.6.1 tag https://github.com/thesofproject/sof/commit/aa2ca7f6aef78fb1dac4314e178e61f46d3291ef

vijendarmukunda, 2023-09-05:
@coolstar 

Could you please try removing below line mentioned in the link?

https://elixir.bootlin.com/linux/latest/source/sound/soc/sof/amd/pci-rmb.c#L37

i.e remove acp_clkmux_sel field from rembrandt_chip_info structure?

coolstar, 2023-09-05:
@vijendarmukunda just tested that here (my driver in the other OS is synced with latest thesofproject/linux) and can confirm that the issue is fixed!

lgirdwood, 2023-09-11:
@coolstar @vijendarmukunda can we close ?

MrChromebox, 2023-09-11:
@lgirdwood is there any commit to make the above referenced change?

lgirdwood, 2023-09-11:
> @lgirdwood is there any commit to make the above referenced change?

@thesofproject/amd can you answer here ? Is the commit on a AMD branch only today ?

lgirdwood, 2023-09-13:
> will raise a PR for this change. This ticket can be closed as the issue is resolved.

ok, will close when PR is merged.

closed 2023-09-25 by vijendarmukunda