thesofproject/sof #8162: [BUG][MTL] High Peak/Average MCPS ratio for gain components

mengdonglin, 2023-09-06:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Run check-performance test case on the target platform

**Reproduction Rate**
100%

**Expected behavior**
Peak to average MCPS ratio should be no more than 3

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev [55bbc7075e68](https://github.com/thesofproject/linux/commit/55bbc7075e686692362b8ffc9c07f5eacdd20ba0)
    * SOF: main [d964b7bf518d](https://github.com/thesofproject/sof/commit/d964b7bf518d05654cfb2d04825cc2b42972898d)

2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL RVP HDA

**Screenshots or console output**
https://sof-ci.ostc.intel.com/#/result/planresultdetail/31304?model=MTLP_RVP_HDA-ace1_0-ipc4&testcase=check-performance
[image]

btian1, 2023-09-11:
issue rootcaused:
1. peak mcps is caused by volume ramp
2. further, take tgl platform as example, driver set ramp time as 20ms, then sof will select 125us as ramp step, means 8 times per ms(1000/125), means for a 48k playback, 6 frames will have a ramp volume, this is somehow too short.

I made below PR for as DRAFT for review, and will listen to reviewer's comments on this.
https://github.com/thesofproject/sof/pull/8183

kv2019i, 2023-09-11:
I'm not fully convinced a high peak/avg ratio is a problem if there is a clear design reason (like there is here). An algorithm can have modalities where it has a high-consumption and low-consumption states. For power consumption, the average cost is what matters. The peak consumption affects pipeline construction (peak consumption across use-case components cannot exceed DSP capacity, or audio glitches will happen). But e.g. with gain as a low-cost component, a high peak/avg ratio is not a problem.

A peak that can not be explained with algorithm design, is another matter, and suggest a bug in system behaviour or component implement. That's of course a different matter.

btian1, 2023-09-11:
we should evaluate each algorithm's Computational complexity to determine whether it need those cycles or not.
regarding volume ramp, power calculation already fastened by look up table, left thing is: determine the ramp stepping.
this is somehow quite subjective, we may need hear and listen other people's voice.

lgirdwood, 2023-09-11:
adding @singalsu.
Lets make sure we solve MCPS vs quality with Kconfig options to allow the user to choose his/her preference for build time config. Additionally we can use topology to allow teh user to select runtime config.

singalsu, 2023-09-12:
> adding @singalsu. Lets make sure we solve MCPS vs quality with Kconfig options to allow the user to choose his/her preference for build time config. Additionally we can use topology to allow teh user to select runtime config.

The first is possible, but I'm afraid there is no room to decrease the ramp gain update rate without negative audible impact. Setting a worse quality ramp would hide an issue if the peak really is an issue.

For the latter the topology can't impact the update rate directly. The ramp duration impacts the internally selected rate though. There's also a bug/feature that if windows fade ramp is enabled in kconfig that no other ramp type can be selected.

Ways to to improve this

- we can make capture ramps longer and have lower gain update rate there
- we don't need initial ramp in every volume instance if there are many volumes in series, should have a way to skip initial ramp but still have ramp for user adjust or mute/unmute
- we could use in Linux and Chrome OS a lighter linear ramp type, Windows fade is only requirement for Windows OS
- find optimizations to windows fade equation

btian1, 2023-09-13:
we don't need initial ramp in every volume instance if there are many volumes in series, should have a way to skip initial ramp but still have ramp for user adjust or mute/unmute

we can first try to implement this, need find a way to find which volume need ramp, which does not.

singalsu, 2023-10-16:
PRs remaining to close this issue are #8322 and #8183.

btian1, 2023-10-20:
https://github.com/thesofproject/sof/pull/8213
https://github.com/thesofproject/sof/pull/8183
https://github.com/thesofproject/sof/pull/8207
https://github.com/thesofproject/sof/pull/8322

full list, all are merged, closing.

btian1, 2023-11-15:
I re-opened this, as we are still seeing some module(mainly volume with ratio > 5) have this high ratio, a lot of test and triage is ongoing, trying to figure out the real root-cause.

kv2019i, 2023-11-24:
Move to v2.9. I think we have good progress on this, I'll let you  @btian1  update, but we can close this in v2.9 context.