thesofproject/sof #8164: [BUG][MTL] no suspend if ssh logs in after driver probing

lyakh, 2023-09-06:
**Describe the bug**

If a user logs in to an MTL system after drivers are loaded and plays audio, after that the DSP isn't powered down by runtime PM.

**To Reproduce**

```
[terminal 1] ssh device
[terminal 1] modprobe snd-sof-pci-intel-mtl
[terminal 2] ssh device
[terminal 2] aplay -d 5
```

Observe no D3 suspend. However, when doing

```
[terminal 1] ssh device
[terminal 2] ssh device
[terminal 1] modprobe snd-sof-pci-intel-mtl
[terminal 2] aplay -d 5
```
After that the DSP is suspended.

**Reproduction Rate**

100%

**Expected behavior**

In both cases the DSP should be suspended.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: daily from 30.08 linux-image-6.5.0-rc6-ge327169bf752
    * SOF: daily from 03.09
2) Name of the topology file
    * Nocodec
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL

kv2019i, 2023-11-24:
@lyakh This is the strangest bug report I've seen in a while. Are you sure somebody else didn't login to the machine locally in the lab and start doing something else? :) Not sure what to make of it, but let's try to reproduce and decice whether this is v2.8 material (and if yes, priority most certainly needs to be bumped)...

lyakh, 2023-11-27:
> Are you sure somebody else didn't login to the machine locally in the lab and start doing something else? :)

@kv2019i yes, I am sure, this is 100% reproducible, and it would be too weird for someone to follow me around and to wait for me to log in to the system to trigger this behaviour. It's likely down to some pulseaudio / pipewire weirdness

closed 2025-07-24 by github-actions[bot]