thesofproject/sof #8165: [BUG][Needs Check] IPC worker thread can run on secondary cores

lyakh, 2023-09-06:
**Describe the bug**

SOF isn't an SMP system, its use of cores is hard-wired. Most management tasks are only supposed to run on the primary core, only pre-assigned audio processing tasks should run on secondary cores. One of such management tasks is the EDF scheduler work handler. To fix it to the primary core a special work queue is created, which is set to only run on the primary core, and the delayed work, implementing the scheduler, is submitted to that queue.

Whereas the IPC delayed work is run on the default work queue, which on Zephyr can run on any core.

**Expected behavior**

The IPC delayed work should only run on the primary core.

**Impact**

Running on a secondary core without proper code synchronisation can lead to data corruption.

lgirdwood, 2023-09-13:
@marcinszkudlinski are you also aligned with this proposal ?

kv2019i, 2023-11-20:
Nobody assigned, less than a week to branch, moving to v2.9.

kv2019i, 2024-03-04:
Stable-v2.9 branched, this didn't make the cut, bumping to 2.10.

kv2019i, 2025-01-03:
@lyakh Can you check this and if still valid, ensure this is covered for for v2.13 ? Bumping now as v2.12 has been already branched.

PR #9764 (2025-02-04): ipc: move delayed IPC sending to the primary core
Low level IPC processing should be confined to the primary core. Move delayed IPC sending to a dedicated work queue with core 0 affinity.

Fixes: #8165
commit 85f1dbea (2025-02-04): ipc: move delayed IPC sending to the primary core

Low level IPC processing should be confined to the primary core. Move
delayed IPC sending to a dedicated work queue with core 0 affinity.

Fixes: #8165