thesofproject/sof #8174: [BUG] MTL: multiple-pause-resume.sh takes 10 times longer with _NOCODEC

marc-hb, 2023-09-07:
MTLP_RVP_NOCODEC takes TEN times longer (!) to run multiple-pause-resume-50.sh than MTLP_RVP_SDW or MTLP_RVP_HDA

All tests seem slower on _NOCODEC but `multiple-pause-resume-50.sh` is especially slower.

This is "bottlenecking" and slowing down the entire validation.

Examples:

https://sof-ci.01.org/linuxpr/PR4577/build6527/devicetest/index.html
multiple-pause-resume-50.sh
MTLP_RVP_NOCODEC: 30 minutes
MTLP_RVP_SDW, MTLP_RVP_HDA: 3 minutes
 
Same in https://sof-ci.01.org/sofpr/PR8161/build12679/devicetest/index.html
and in every `mtl_pr` test run I've looked at in the last few days.

singalsu, 2023-09-07:
So average 36s vs. 3.6s per suspend-resume cycle, that's a big difference!

marc-hb, 2023-09-07:
Thanks @plbossart : 

> The answer is in the topology: there are just more PCM devices in the
nocodec topology and the script blindly tests them all. In theory the
test duration should be a linear factor of the number of PCM devices,
but that could be impacted by firmware transitions since some PCM
devices go into mixers and pipelines are activated with a refcount.

closed 2023-09-07 by marc-hb