thesofproject/sof #8178: [FEATURE] remove SOF build time dependency to rimage headers

kv2019i, 2023-09-07:
**Is your feature request related to a problem? Please describe.**
rimage was originally moved to a different repository as it is an independent tool from SOF. 

With changes since SOF2.4, we have more and more direct dependencies from SOF to rimage. IPC4 protocol requires metadata that is defined in firmware manifest (data maintained in toml files, rimage encodes this to firmware binary file).
Most recently, loadable module support was added to SOF firmware and this code depends on firmware file structure definitions that are needed both by rimage and the new SOF module loader.

The build dependencies are causing issues to downstream SOF users (e.g. https://github.com/zephyrproject-rtos/zephyr/pull/62262  ).

Having the rimage manifest changes (like https://github.com/thesofproject/rimage/pull/177) not tested by full SOF CI, is causing operational issues (problems caused by rimage changes are detected too late).

**Describe the solution you'd like**
Merge rimage back to sof repository as it is now more integrate part of SOF. 

**Describe alternatives you've considered**
- alternative: duplicate manifest headers in SOF (these are part of the loadable modules ABI)
      - cons: does not help with SOF CI woes with rimage PRs

kv2019i, 2023-09-07:
FYI @nashif @lgirdwood @marc-hb @softwarecki @aiChaoSONG

marc-hb, 2023-09-07:
cc: @andyross , @dbaluta , @iuliana-prodan , @dcpleung 

> rimage was originally moved to a different repository as it is an independent tool from SOF.

Anyone remembers the rationale for a separate rimage repo? We don't want to go back and forth and back. I know booting Zephyr tests on Intel/NXP hardware is possible *with* rimage and *without* SOF. Could have this been a reason?

Having more git repos for no good reason makes everything more complicated. Among others and as mentioned above, it makes CI especially more complicated: how do you submit an rimage change (or Zephyr) to SOF testing? You can:
- https://github.com/thesofproject/rimage/blob/main/README.md
- https://docs.zephyrproject.org/latest/develop/modules.html#process-for-submitting-changes-to-existing-modules

... BUT it's tedious and experience shows most people don't take the time.

https://en.wikipedia.org/wiki/Conway%27s_law aside (does not apply here), the ability to **make arbitrary combinations** is probably the only good reason to have multiple repos. For instance we want the latest SOF-TEST branch to be able to test most SOF branches out of the box. We don't want to branch SOF-TEST every time that we branch SOF, that would be much _more_ work. So: a different repo for SOF-TEST.

Similarly, we want the latest SOF docker image to be able to test any SOF branch, which shows `sof/scripts/docker_build/` should be in a separate repo:
- #7552
- #5073

Back to rimage: do we need the ability to freely combine any rimage version with any SOF version? I guess not.

PS: https://en.wikipedia.org/wiki/Monorepo (@ Google, Facebook, Microsoft)

plbossart, 2023-09-07:
2019 change in 94956778b0c35ec95c2690f04bffe7cd19fef6b0

At some point there was a decision that tools don't belong in the SOF tree. That was after we merged them into SOF from https://github.com/thesofproject/soft to avoid the sort of problems we now have.

This clearly means we don't have a clear idea of what we are doing :-)

marc-hb, 2023-09-07:
gitarcheology of the past moves with links to the corresponding discussions. I hope I did not miss anything.

The most relevant link is https://github.com/thesofproject/sof/pull/2086 , this is where most of the rimage split discussion took place.

**October 2016**: creation of new sof-tools.git with rimage import as the first commit
- https://github.com/thesofproject/soft/commit/04a535d508

**February 2018**: rimage is moved from sof-tools.git/rimage/ to sof.git/rimage/ to "remove a cyclic dependency between these two repos"
- https://github.com/thesofproject/sof/commit/05ef4343d112 ADD
- https://github.com/thesofproject/soft/commit/72d797d9ddae7 REMOVE

Git history is not transferred... yet! See next step.
The sof-tools.git/tools/rimage/keys/ subdirectory is copied but not removed and also left behind for some unknown reason.

**November 2018**: the rest of sof-tools.git follows and is also merged into sof.git to "reduce complexity".
- https://github.com/thesofproject/sof/pull/624
- https://github.com/thesofproject/sof/commits/ebf4f52652b4

This time the sof-tools git history is merged into sof.git thanks to @cujomalainey. This funnily enough resurrects the old sof-tools.git/rimage/ history in sof.git: https://github.com/thesofproject/sof/commits/72d797d9ddae7

sof-tools.git is frozen/archived.

  --- SHORT-LIVED, SINGLE SOF REPO NIRVANA !? --

**Nov 2019 - May 2020**: rimage split from sof.git to a new rimage submodule (ick) for (at least) some Zephyr reason:
- https://github.com/thesofproject/sof/pull/2086  <= discussed mostly here
- https://github.com/thesofproject/rimage/pull/1 
- https://github.com/thesofproject/rimage/pull/3

rimage history is not transferred.

> This clearly means we don't have a clear idea of what we are doing :-)

In the links above I think I spotted some "plbossart" guy who already said that at the time.

marc-hb, 2023-09-13:
> the ability to make arbitrary combinations is probably the only good reason to have multiple, independent repos. 

In a meeting it was noted that rimage has recently gained more features which has made it more SOF-dependent than it was when it was split in May 2020.

So I ran some basic `git log --oneline ... | wc` commands on https://github.com/thesofproject/sof/commits/main/rimage and https://github.com/thesofproject/rimage/commits/main to quickly gauge the `rimage` activity. It has roughly doubled since January 2023:

Between June 2020 (right after it was split) and January 2023

- 133 rimage commits in 30 months
- submodule bumped 26 times in 30 months.

Since January 2023:

- 81 commits in 9 months
- submodule bumped 12 times in 9 months.

Of course these are just commit numbers, they don't directly measure dependencies.

cujomalainey, 2023-09-15:
Thanks for the history @marc-hb I agree, anytime we are expressing the same information across multiple repos we are in for a bad time. Now that the manifest contains data from the modules I think the split is not acceptable anymore. In reality I would rather the manifest derive from the C definition of the component so it's a single file but that is a different problem entirely.

kv2019i, 2023-09-19:
FYI @pjdobrowolski

aiChaoSONG, 2023-09-19:
IMHO: rimage could be in a separate repo, or we could move it under sof/tools, just like other tools, but those toml files should stay with sof.

I have implemented the code to copy Intel IPC4 toml files to sof/config, and split to platform related toml to `sof/config/platform/`, module related toml to sof/config/module/`, files still names as PLATFORM.toml.

Further split could be done for module toml, we could have one toml for each module, and only include the toml if it is enabled in kconfig. I could raise an RFC once https://github.com/thesofproject/sof/pull/8212 got merged.

abonislawski, 2023-09-19:
> IMHO: rimage could be in a separate repo, or we could move it under sof/tools, just like other tools, but those toml files should stay with sof.
> Further split could be done for module toml, we could have one toml for each module, and only include the toml if it is enabled in kconfig.

Looks like a good idea, rimage repo + in SOF toml per module, just like all the rest module files in SOF, no need to maintain one giant toml in another repo

kv2019i, 2023-09-25:
I'm tagging this as "unclear" as we don't have a consensus whether this is a problem to begin with.

lgirdwood, 2023-09-29:
@nashif do you need rimage moved into SOF with unified headers ?

lgirdwood, 2023-09-29:
@kv2019i @aiChaoSONG - let move rimage back into SOF and deprecate the rimage repo for v2.8.

marc-hb, 2023-10-02:
The main question is whether we move rimage.git back into `sof/tools/rimage` (smooth option) or into `sof/rimage` (brutal option).

The brutal option is faster but it creates a "flag day" that can't be `git bisect`ed across and it will likely break other git commands for submodule users = all XTOS users + some Zephyr users.

The smoother, `sof/tools/rimage` option requires a transition period and a few scripting changes but it shouldn't break much. I think it's the safer choice.

```shell
cd sof/rimage

# We will need a ref to fetch and merge
git fetch thesofproject/
git checkout -t thesofproject/main

# Smooth option 
mkdir tools/rimage
find . -maxdepth 1 ! -name . ! -name .git ! -name rimage -exec  git mv {} tools/rimage/ \;

# Brutal option
mkdir  rimage/
find . -maxdepth 1 ! -name . ! -name .git ! -name rimage -exec  git mv {} rimage/ \;

git commit -m 'move everything down to prepare git merge'

# Fetch rimage history into sof history
git remote add rimage-repo rimage
git fetch rimage-repo

git submodule deinit rimage || true
rm -rf ../sof/rimage # we already fetched it

git merge --allow-unrelated-histories -m 'Merging rimage history back into sof/tools/' rimage-repo/main

# re-init tomlc99 submodule
sed '/path.*tomlc99/ s#tomlc99#rimage/tomlc99#' tools/rimage/.gitmodules >> .gitmodules 
git rm tools/rimage/.gitmodules
git commit -m 're-add tomlc99 submodule'

git submodule update --init

# Change rimage location in sof/west.yml
west update

(build)
```

marc-hb, 2023-10-04:
I'm on it, so far it seems easier than I thought it would be (famous last words). I will hopefully submit a PR this week.

marc-hb, 2023-10-09:
rimage transfer ready for prime time in:
- #8297