thesofproject/sof #8191: [BUG] [v2.7-rc1] IPC timed out for 0x47000000|MOD_SET_DX when suspending on ADLP_RVP_NOCODEC

mengdonglin, 2023-09-12:
**Describe the bug**
IPC timed out happens on suspending on ADLP_RVP_NOCODEC, in stress test of v2.7-rc1.

**To Reproduce**
check-suspend-resume-with-capture-100.sh

**Reproduction Rate**
Low. Observed in v2.7 stress test once
https://sof-ci.ostc.intel.com/#/result/planresultdetail/31509

```
[ 1019.963176] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget host-copier.0.capture freed
[ 1019.963189] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[ 1019.963361] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[ 1019.963450] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[ 1019.963462] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[ 1020.470524] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[ 1020.470547] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 1020.470557] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1020.470583] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[ 1020.470593] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1020.470618] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0x0|0x0|0x80000000, ctl: 0x3
[ 1020.470630] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1020.470637] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1020.470642] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 1020.470648] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 1020.470671] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xffffffff: unknown ROM status value
[ 1020.470769] kernel: snd_sof:sof_ipc4_find_debug_slot_offset_by_type: sof-audio-pci-intel-tgl 0000:00:1f.3: Slot type 0x4c455400 is not available in debug window
[ 1020.470778] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1020.470789] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
```

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev), [dd526472a2da](https://github.com/thesofproject/linux/commit/dd526472a2dac76455a21f6351b8b286cad5b889)
    * SOF: [stable-v2.7](https://github.com/thesofproject/sof/tree/stable-v2.7), [2d834e938c99](https://github.com/thesofproject/sof/commit/2d834e938c99978d4d02364588169e9289f7302d)
2) Name of the topology file
    * Topology: sof-adl-nocodec (single DSP core)
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP_RVP_NOCODEC

keqiaozhang, 2023-09-13:
Detected this issue on main branch with IMR disabled. 
https://sof-ci.ostc.intel.com/#/result/planresultdetail/31668?model=ADLP_RVP_NOCODEC-cavs2_5-ipc4-nocodec&testcase=check-suspend-resume-with-capture-100

mengdonglin, 2023-09-15:
Put it to P3 as the very low reproduce rate and low impact to user experience.

RanderWang, 2023-09-27:
@mengdonglin When we find it in CI test with multicore enabled https://sof-ci.01.org/sofpr/PR8240/build13473/devicetest/index.html?model=ADLP_RVP_NOCODEC_IPC4ZPH&testcase=multiple-pipeline-all

fredoh9, 2023-10-02:
Found same issue with main branch.
This is Intel internal daily test link: planresultdetail/32586?model=ADLP_RVP_NOCODEC-cavs2_5-ipc4-nocodec&testcase=check-suspend-resume-with-capture-5

- SOF Branch: 
- [main](https://github.com/thesofproject/sof/tree/main)
- SOF Commit: 
- [405d48dbeaf7](https://github.com/thesofproject/sof/commit/405d48dbeaf7f5ff27fb37cb05558585a4c03264)
- Zephyr Commit: 
- [492517b918d2](https://github.com/zephyrproject-rtos/zephyr/commit/492517b918d267f553688cd6b9d59b92ffc10f91)

fredoh9, 2023-10-05:
Found same error again with ADLP_RVP_NOCODEC.

- SOF Branch: [main](https://github.com/thesofproject/sof/tree/main)
- SOF Commit: [51159d80e4c3](https://github.com/thesofproject/sof/commit/51159d80e4c329f20c9c402f8a4758c79f9e430c)
- Zephyr Commit: [8c4eec7ac6e3](https://github.com/zephyrproject-rtos/zephyr/commit/8c4eec7ac6e37be89af89e021c6f5c96e1ac1e0a)

Intel Internal daily test link:
planresultdetail/32701?model=ADLP_RVP_NOCODEC-cavs2_5-ipc4-nocodec&testcase=check-suspend-resume-with-capture-5

keqiaozhang, 2023-10-26:
Found this issue on TGLU_RVP_NOCODEC-ipc4. Inner test ID:33668

RanderWang, 2023-11-01:
https://github.com/zephyrproject-rtos/zephyr/pull/64541 and  https://github.com/zephyrproject-rtos/zephyr/pull/62522 can improve this issue. I do many stress test

mengdonglin, 2023-11-08:
Thank you @RanderWang!  This issue cannot be reproduced in recent IP4 daily test. Closing.

closed 2023-11-08 by mengdonglin