thesofproject/sof #8193: [BUG] sof/installer does not support Zephyr builds with xtensa-build-zeprhy.py

kv2019i, 2023-09-13:
**Describe the bug**
sof/installer/GNUmakefile uses xtensa-build-all.sh to build the firmwares. This onlye covers targets witH XTOS support, not Zephyr.

**To Reproduce**

**Reproduction Rate**

**Expected behavior**

**Impact**
sof-bin releases need to be manually constructed.

**Environment**

marc-hb, 2023-09-13:
This is "by design", the Zephyr build is really too different. Yes this means XTOS and Zephyr builds need to be combined manually, tarball_multi_releases.bash should be able to do that.

kv2019i, 2023-09-13:
Ack @marc-hb , let's close this. I now realzie xtensa-build-zephyr.py actually does the staging for the binary, but we are missing a staging step for the topologies (or I've not found it yet).

closed 2023-09-13 by kv2019i