thesofproject/sof #8195: [FEATURE] Mute light support

EvanCarroll, 2023-09-13:
It would be nice if there was mute light support in sof. [I wrote about this in detail here](https://unix.stackexchange.com/questions/756443/how-can-i-activate-my-keyboards-microphone-mute-light-when-using-set-source-mut), but it seems _something_ in the stack (and I'm not aware of where this happening), is triggering my audio light on mute, when I run

    pactl set-sink-mute   @DEFAULT_SINK@ toggle

But that functionality is weirdly absent in,

    pactl set-source-mute @DEFAULT_SOURCE@ toggle

But that would be tremendously useful for writing a simply global-push to talk with a visual indicator of when you're not hot.

-----

(This very very likely doesn't belong on your board, I just don't know where to put it. Feel free to point me in another direction if that functionality is not in sof and shouldn't be homed there. I see in some places they suggest things like this `options snd-hda-intel model=dell-s14` to enable the mute light in some laptop builds. So it seems like this may be the right place. It would be nice if there was an option for `lenovo-x1c`.

plbossart, 2023-09-13:
I can't comment on pactl, it's been too many years since I used it.

At the SOF level, we have an ALSA control to mute the DMIC, and there's a token to define whether the control is associated with an LED. that's about the extent of the involvement of the SOF topology+driver wrt the LED mutes.

The bulk of the work is done in UCM, there is a mapping from the ALSA control to LEDs, see e.g. ucm2/Intel/sof-hda-dsp/sof-hda-dsp.conf

it's entirely possible that this LED only worked on Dell devices.

Adding @perexg for more comments since he's probably the most knowledgeable on this LED support

huyhoang8398, 2025-05-01:
well, i have to install alsa-utils package to make the lighting work, not sure why