thesofproject/sof #8209: [BUG] Headphone playback/capture not working on AMD Mendocino Chrome device

WeirdTreeThing, 2023-09-17:
**Describe the bug**
When attempting to play audio through the headphones, there is no output. The same thing happens when capturing audio from the headset mic, nothing gets captured.

When using the same userspace and sof fw/tplg but with the ChromeOS kernel built from release-R115-15474.B-chromeos-5.15, headphone playback/capture works as expected.

**To Reproduce**
1. Play audio on headphones
2. Listen for any output

**Reproduction Rate**
Always

**Expected behavior**
Audio output on headphones should work.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev@cffcac3
    * SOF: amd-rmb-stable
2) Name of the topology file
    * Topology: sof-acp-rmb-tdm8ch-dmic4ch-rt5682-rt1019.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: AMD Mendocino

**Screenshots or console output**
[Full log](https://github.com/thesofproject/sof/files/12643756/sof-log.txt)

Snippet from headphone playback
```
[ 3226.853111] snd_sof:sof_pcm_open: snd_sof_amd_rembrandt 0000:05:00.5: pcm: open stream 0 dir 0
[ 3226.853119] snd_sof:sof_pcm_open: snd_sof_amd_rembrandt 0000:05:00.5: period min 192 max 16384 bytes
[ 3226.853121] snd_sof:sof_pcm_open: snd_sof_amd_rembrandt 0000:05:00.5: period count 2 max 16
[ 3226.853124] snd_sof:sof_pcm_open: snd_sof_amd_rembrandt 0000:05:00.5: buffer max 65536 bytes
[ 3226.853501] snd_sof:sof_ipc3_pcm_dai_link_fixup: snd_sof_amd_rembrandt 0000:05:00.5: AMD_HS channel_max: 8 rate_max: 48000
[ 3226.853514] snd_sof:sof_ipc3_pcm_dai_link_fixup: snd_sof_amd_rembrandt 0000:05:00.5: AMD_HS channel_max: 8 rate_max: 48000
[ 3226.853531] snd_sof:sof_pcm_hw_params: snd_sof_amd_rembrandt 0000:05:00.5: pcm: hw params stream 0 dir 0
[ 3226.853565] snd_sof_utils:snd_sof_create_page_table: snd_sof_amd_rembrandt 0000:05:00.5: generating page table for 000000003c8a5d45 size 0x10000 pages 16
[ 3226.853570] snd_sof:sof_ipc3_pcm_hw_params: snd_sof_amd_rembrandt 0000:05:00.5: stream_tag 1
[ 3226.853573] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc tx: 0x60010000
[ 3226.854488] snd_sof_amd_acp:acp_set_stream_data_offset: snd_sof_amd_rembrandt 0000:05:00.5: pcm: stream dir 0, posn mailbox offset is 4324
[ 3226.855386] snd_sof:sof_pcm_trigger: snd_sof_amd_rembrandt 0000:05:00.5: pcm: trigger stream 0 dir 0 cmd 1
[ 3226.855395] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc tx: 0x60040000
[ 3226.954923] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3226.954968] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.040924] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.040970] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.126935] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.126981] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.212897] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.212933] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.298895] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.298929] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.384656] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.384699] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.470912] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.470953] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.556934] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.556980] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.642934] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.642979] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.728913] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.728953] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.814899] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.814934] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.900936] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.900981] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3227.986885] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3227.986920] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3228.072912] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3228.072954] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3228.158933] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3228.158979] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3228.245171] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3228.245211] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3228.330909] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx: 0x600a000e
[ 3228.330950] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc rx done: 0x600a000e
[ 3228.331001] snd_sof:sof_pcm_trigger: snd_sof_amd_rembrandt 0000:05:00.5: pcm: trigger stream 0 dir 0 cmd 0
[ 3228.331008] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc tx: 0x60050000
[ 3228.336016] snd_sof:ipc3_log_header: snd_sof_amd_rembrandt 0000:05:00.5: ipc tx: 0x60030000
[ 3228.336480] snd_sof:sof_pcm_hw_free: snd_sof_amd_rembrandt 0000:05:00.5: pcm: free stream 0 dir 0
[ 3228.337477] snd_sof:sof_pcm_close: snd_sof_amd_rembrandt 0000:05:00.5: pcm: close stream 0 dir 0
```

lgirdwood, 2023-09-22:
@thesofproject/amd fyi.

WeirdTreeThing, 2023-09-25:
I am using a Frostflow Chromebook with rt5682s and rt1019 on mainline linux.

vijendarmukunda, 2023-09-26:
We are able to reproduce the issue. We will get back.

Venkata-Prasad-Potturu, 2023-09-29:
Hi All,
With the recent machine driver changes in dapm widgets, SPK widget is not been used. So, UCM configuration file need be updated.
Manually setting the mixer controls able to get audio (Headset playback and capture). Below are the attached working amixer settings for headset playback and capture. Similarly speaker mixer controls need to on for speaker playback.
[Headset_amixer_settings_working.txt](https://github.com/thesofproject/sof/files/12763528/Headset_amixer_settings_working.txt)

Speaker mixer settings:- 
		 numid=15,iface=MIXER,name='Right Spk Switch'
		  ; type=BOOLEAN,access=rw------,values=1
		  : values=on
		 numid=14,iface=MIXER,name='Left Spk Switch'
		  ; type=BOOLEAN,access=rw------,values=1
		  : values=on

WeirdTreeThing, 2023-09-29:
Thanks for the response. Unfortunately, after applying those mixer settings, headset playback and capture still doesn't work.

Venkata-Prasad-Potturu, 2023-09-29:
Could you try below amixer commands and share your amixer contents as well.

amixer -c 1 cset numid=14 0
amixer -c 1 cset numid=15 0

amixer -c 1 cset numid=4 50
amixer -c 1 cset numid=18 1
amixer -c 1 cset numid=20 1
amixer -c 1 cset numid=21 1
amixer -c 1 cset numid=27 1
amixer -c 1 cset numid=31 2
amixer -c 1 cset numid=11 1
amixer -c 1 cset numid=41 1
amixer -c 1 cset numid=42 1

WeirdTreeThing, 2023-09-29:
Running those commands gave me the same result. 
[amixer-contents.txt](https://github.com/thesofproject/sof/files/12765161/amixer-contents.txt)

Venkata-Prasad-Potturu, 2023-10-04:
Sorry for the late response :),
Your Amixer commands are same as mine.
Could you please check and share me **evtest** results to ensure headset is connected properly.

WeirdTreeThing, 2023-10-04:
Here you go. 
[mdn-headset-evtest.txt](https://github.com/thesofproject/sof/files/12803253/mdn-headset-evtest.txt)

Venkata-Prasad-Potturu, 2023-10-04:
Please try to increase the volume using below mixer control
amixer -c 1 cset numid=1 127
Attached amixer settings are working in my setup.
[Working_amixer_settings.txt](https://github.com/thesofproject/sof/files/12803362/Working_amixer_settings.txt)

WeirdTreeThing, 2023-10-04:
Unfortunately, I still get no output from the headset. Here's my mixer settings [amixer-mdn.txt](https://github.com/thesofproject/sof/files/12809140/amixer-mdn.txt)
Can you let me know which kernel version is working for you?

Venkata-Prasad-Potturu, 2023-10-20:
Hi,

>>Can you let me know which kernel version is working for you?
We are using V5.15 kernel as frostflow uses v5.15 kernel, on top of that we have added latest machine driver changes.
When we added latest machine drvier patches audio is not working then we have removed SPK dapm widget in the UCM config then
it is working fine.

Could you please revert the latest machine driver changes mentioned below on your kernel and try, we will get to know 
whether issue is with spk dapm widget or not.
	ASoC: amd: acp: Add kcontrols and widgets per-codec in common code
	ASoC: amd: acp: Map missing jack kcontrols
	ASoC: amd: acp: Add machine driver support for max98388 codec
	ASoC: amd: acp: Add machine driver support for nau8821 codec.
	
Is the issue happening with aplay command also after entering amixer commands for headset?

WeirdTreeThing, 2024-04-13:
Looks like a recent kernel update fixed this.

closed 2024-04-13 by WeirdTreeThing