thesofproject/sof #8224: [BUG][mtl-006] I/O error when running aplay on speaker pipeline

keqiaozhang, 2023-09-19:
**Describe the bug**
It seems that there's a new regression on MTL-SDW(single core) platforms. there's an I/O error when checking aplay on speaker pipeline at every first time. The reproduce rate is 100%. No critical errors in kernel log, but there're many `dai_zephyr_multi_endpoint_copy(): nothing to copy` and XRUNs detected in mtrace.

```
[    0.103653] <inf> pipe: add_pipeline_cps_consumption: Sum of KCPS consumption: 44306, core: 0
[    0.103656] <inf> pipe: add_pipeline_cps_consumption: Registering module: 0x20000 KCPS consumption: 4996, core: 0
[    0.103661] <inf> pipe: add_pipeline_cps_consumption: Sum of KCPS consumption: 49302, core: 0
[    0.103671] <inf> ll_schedule: zephyr_ll_task_schedule_common: task add 0xa010b800 0x201e0U priority 0 flags 0x0
[    0.104108] <wrn> dai_comp: dai_zephyr_multi_endpoint_copy: comp:1 0x40001 dai_zephyr_multi_endpoint_copy(): nothing to copy
[    0.104171] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 0, free_samples: 192
[    0.105105] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.105111] <wrn> dai_comp: dai_zephyr_multi_endpoint_copy: comp:1 0x40001 dai_zephyr_multi_endpoint_copy(): nothing to copy
[    0.105120] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 0, free_samples: 192
[    0.106116] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.106121] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.106125] <wrn> dai_comp: dai_zephyr_multi_endpoint_copy: comp:1 0x40001 dai_zephyr_multi_endpoint_copy(): nothing to copy
[    0.106135] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 0, free_samples: 192
[    0.107111] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.107116] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.107120] <wrn> dai_comp: dai_zephyr_multi_endpoint_copy: comp:1 0x40001 dai_zephyr_multi_endpoint_copy(): nothing to copy
[    0.107130] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 0, free_samples: 192
[    0.108150] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.108155] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.108158] <wrn> dai_comp: dai_zephyr_multi_endpoint_copy: comp:1 0x40001 dai_zephyr_multi_endpoint_copy(): nothing to copy
[    0.108168] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 0, free_samples: 192
[    0.109111] <err> dma_dw_common: dw_dma_get_status: xrun detected
[    0.109116] <err> dma_dw_common: dw_dma_get_status: xrun detected
```

**To Reproduce**
 aplay   -Dhw:0,2 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Linux Commit: [cffcac365eac](https://github.com/thesofproject/linux/commit/cffcac365eac0d60dc38959cadaa5078bc924b8e)
    * SOF: {[mtl-006-drop-stable](https://github.com/thesofproject/sof/tree/mtl-006-drop-stable)
    * SOF Commit: [37bd2817e898](https://github.com/thesofproject/sof/commit/37bd2817e898e28ca3becf721794ea65966a13b5)}
2) Name of the topology file
    * Topology: {sof-ace-tplg/sof-mtl-rt711-l0-rt1316-l23-rt714-l1.tplg/sof-ace-tplg/sof-mtl-rt1318-l12-rt714-l0.tplg }
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL-SDW}

**Screenshots or console output**
```
2023-09-18 22:04:09 UTC [REMOTE_COMMAND] aplay   -Dhw:0,2 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q
Hardware PCM card 0 'sof-soundwire' device 2 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
aplay: pcm_write:2127: write error: Input/output error
2023-09-18 22:04:11 UTC [REMOTE_ERROR] aplay on PCM hw:0,2 failed at 1/1.
2023-09-18 22:04:11 UTC [REMOTE_INFO] Starting func_exit_handler(1)
```
[dmesg.txt](https://github.com/thesofproject/sof/files/12654826/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/12654834/mtrace.txt)

keqiaozhang, 2023-09-19:
This issue is introduced by https://github.com/thesofproject/sof/pull/8194

mengdonglin, 2023-09-19:
> This issue is introduced by #8194

@serhiy-katsyuba-intel Can you help check this issue?

serhiy-katsyuba-intel, 2023-09-19:
PR with the fix for mtl-006-drop-stable branch https://github.com/thesofproject/sof/pull/8226

The problem indeed was triggered by new mixin/mixout re-implementation (support for cross-core connection). However, the problem was really in dai-zephyr. If sink or source buffer is full and so no space to copy data out/in of DMA buffer, dai-zephyr just log "nothing to copy" and skipped dma_realod(). There are multiple "nothing to copy" in the log before XRUN. Skipping dma_reload() for a long time leads to XRUN.
Probably new mixin_prepare() or mixout_prepare() takes somewhat more time then previous version (but not too terribly more).  Seems previously we were just some short time before XRUN. With new mixin_prepare()/mixout_prepare() first dma_reload() happens somewhat leter and that was enough to catch XRUN.
The fix calls dma_realod() when "nothing to copy" happens, i.e., no space in sink of source buffer.

plbossart, 2023-09-19:
This does not look like a fix to me. If the sink is full, then you are already toast. in a 'normal' situation the sink should not be full, ever.
sorry for being blunt, but we're seen this dma-reload before when the capture failed because of host DMA issues, this seems to be the go-to solution when we don't really know what's going on...

marc-hb, 2023-09-20:
> No critical errors in kernel log, but there're many dai_zephyr_multi_endpoint_copy(): nothing to copy and XRUNs detected in mtrace.

The very small https://github.com/thesofproject/sof-test/pull/1075/files PR will make every sof-test fail if any `<err>` is found in the FW logs.

mengdonglin, 2023-09-20:
This issue cannot be reproduced with mtl v6.1 (https://github.com/thesofproject/sof/releases/tag/releases%2Fmtl%2Fv6.1) Closing . 

We don't spot this issue on SOF main branch atm.

closed 2023-09-20 by mengdonglin