thesofproject/sof #8236: [BUG][MTL][HDA] Sporadically Audio glitches on on-board speakers upon unplug of HDMI/DP external display cable

syedk008, 2023-09-20:
**Describe the bug**
Sporadically glitches were observed on speakers during playback when HDMI external display was disconnected.

**To Reproduce**

- Play a local audio via default audio player.
- Plug-in HDMI/DP external display. Audio should be routed to external display
- Unplug HDMI/DP external display. 
- Repeat plug/unplug HDMI/DP external display several times

**Reproduction Rate**
50%

**Expected behavior**
No glitches should be present during playback.

**Impact**
showstopper

**Environment**
1) * Kernel: 6.4.1 with backporting patches from latest/sof-dev kernel(70d2c6
    * SOF: mtl-006
2) * Topology: sof-hda-generic-4ch.tplg
3)  * Platform: mtl

**log**

[dmesg.log](https://github.com/thesofproject/sof/files/12677712/dmesg.log)

mengdonglin, 2023-09-21:
@syedk008  Thanks for reporting this issue!
Does the audio glitch on speaker disappear in a few seconds, or persist until HDMI/DP external display is plug-in again?
Can this issue be reproduced by using module parameter to ignore the CPC value (snd-sof ipc4_ignore_cpc=1)?
Can you help check if this issue can be reproduced with previous mtl-005 FW with the same kernel?

ujfalusi, 2023-09-21:
The `0xe040000` is the CHAIN_DMA message with allocate and enable = 0, it is sent when the playback is stopped.
The firmware responds with '6' which is really odd as it can only happen if the `memcpy_s` fails [1] or the second ipc_get_comp_by_id() fail to get the component [2]

[1] https://github.com/thesofproject/sof/blob/main/src/ipc/ipc4/handler.c#L616
[2] https://github.com/thesofproject/sof/blob/main/src/ipc/ipc4/handler.c#L649

can you enable more verbose prints via dyndbg in kernel and also provide a firmware log? I'm not sure what `0xe040101` and `0xe040202` are.

syedk008, 2023-09-22:
> As so far I have not heard this glitch, while I've been playing with MTL RVP and display-audio. It would be nice know some details on how it sounds, or even have a sample of it. Also the connected display/AV-amplifier maybe relevant.
Video file is attached to the HSD:https://hsdes.intel.com/resource/15014143962 . Please check. I will get the details of the display.

syedk008, 2023-09-22:
@jsarha display is acer kg281k.

Steps to reproduce the issue: Reboot with the external display attached, Switch in the audio settings between the internal speaker and the external display. Repeat this several times.

@mengdonglin issue can be reproduced with  mtl-005 FW as well. currently this parameter (snd-sof ipc4_ignore_cpc=1) is set.
once the glitches are observed on on-board speakers, they persist until reboot the board. glitches disappear once switch to the external display and glitches are back when switch back to on-board speakers.

@ujfalusi dmesg and firmware logs are attached. Please let me know if you need more logs.
[firmware_log.txt](https://github.com/thesofproject/sof/files/12703909/firmware_log.txt)
[dmesg_log.txt](https://github.com/thesofproject/sof/files/12703914/dmesg_log.txt)

ujfalusi, 2023-09-25:
@syedk008, so the garbled audio is on the analog speaker and _not_ on the HDMI?
Or you hear the garbled audio on the display when you have switched to the analog speaker?

In the kernel log there is a whole lot going on than just switching the playback back and forth between HDMI and analog speaker.
There is capture switching from HDA to/from DMIC (with overlap)
```
$ rg "0000:00:1f.3: pcm: " Downloads/dmesg_log.txt 
764:[    7.231986] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 5 dir 0
770:[    7.232168] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 5 dir 0
781:[    7.246965] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 4 dir 0
786:[    7.247097] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 4 dir 0
796:[    7.258714] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 3 dir 0
801:[    7.258868] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 3 dir 0
825:[    7.601051] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 0 dir 0
831:[    7.601183] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 0 dir 0
934:[    7.608330] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 0 dir 1
940:[    7.608395] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 0 dir 1
1009:[    7.614273] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 6 dir 1
1014:[    7.614353] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 6 dir 1
1122:[    7.625407] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 5 dir 0
1130:[    7.625769] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 5 dir 0
1131:[    7.625802] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 4 dir 0
1139:[    7.626180] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 4 dir 0
1140:[    7.626199] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 3 dir 0
1148:[    7.626497] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 3 dir 0
1149:[    7.626514] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 0 dir 0
1177:[    7.629082] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 0 dir 0
1178:[    7.629111] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 0 dir 1
1200:[    7.631264] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 0 dir 1
1201:[    7.631287] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 6 dir 1
1227:[    7.633754] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 6 dir 1
1228:[    7.634857] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 5 dir 0
1233:[    7.634964] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 5 dir 0
1246:[    7.639133] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 5 dir 0 cmd 1
1253:[    7.641484] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 4 dir 0
1258:[    7.641552] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 4 dir 0
1270:[    7.644809] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 4 dir 0 cmd 1
1276:[    7.646944] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 3 dir 0
1281:[    7.647095] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 3 dir 0
1292:[    7.651797] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 1
1298:[    7.653445] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 0 dir 0
1304:[    7.653879] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 0 dir 0
1416:[    7.677370] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
1427:[    7.689572] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 0 dir 1
1433:[    7.689640] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 0 dir 1
1499:[    7.699665] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 1
1503:[    7.699803] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 6 dir 1
1508:[    7.699912] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 6 dir 1
1632:[    7.739713] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 1
1667:[   12.805404] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 0
1672:[   12.816267] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 6 dir 1
1698:[   12.821041] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 6 dir 1
1704:[   12.831939] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[... cut ...]
2123:[   48.762235] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 1
2132:[   58.434658] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
2138:[   58.436208] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 3 dir 0
2146:[   58.436890] snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 3 dir 0
```
playback stream 0 is HDA analog
playback stream 3, 4 and 5 is HDMI/DP
capture stream 0 is HDA analog
capture stream 6 is DMIC

syedk008, 2023-09-25:
@ujfalusi yes, we are observing the issue only on analog speakers. 
I think, during the system bootup all pcm streams are opening/closing. please check the attached bootup log.
Also when settings->audio page opens, both playback and capture pcm stream starts.
[dmesg_bootup.txt](https://github.com/thesofproject/sof/files/12718233/dmesg_bootup.txt)

ujfalusi, 2023-09-26:
@syedk008, when the audio is corrupted you can try to use the probes to extract the audio from the firmware at a given point to hear where the glitch is added.
https://thesofproject.github.io/latest/developer_guides/debugability/probes/index.html

I will create an update for the page, but I also followed it more or less.

to get the ID that you need to set (sample from upx-i11 and for the analog playback path):
```
dmesg | grep "tplg: widget "

[13826.489918] snd_sof:sof_widget_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: widget 2 (gain.1.1) is ready [type: 6, pipe: 1, pins: 1 / 1, stream: none]
[13826.489948] snd_sof:sof_widget_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: widget 3 (mixin.1.1) is ready [type: 4, pipe: 1, pins: 1 / 3, stream: none]
[13826.489975] snd_sof:sof_widget_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: widget 4 (pipeline.1) is ready [type: 32, pipe: 1, pins: 0 / 0, stream: none]
[13826.489996] snd_sof:sof_widget_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: widget 5 (codec0_in) is ready [type: 0, pipe: 1, pins: 0 / 0, stream: none]
[13826.490020] snd_sof:sof_widget_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: widget 6 (iDisp2 Tx) is ready [type: 7, pipe: 1, pins: 0 / 0, stream: none]
[13826.490058] snd_sof:sof_widget_ready: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: widget 7 (dai-copier.HDA.Analog.playback) is ready [type: 27, pipe: 2, pins: 1 / 0, stream: Analog]
```
I want to extract the audio from the first gain widget, so the number is **2**, because:
the number is "tplg: widget **2** (gain.1.1) is ready [type: 6, pipe: 1, pins: 1 / 1, stream: none]"
`echo 2,0,0 > /sys/kernel/debug/sof/probe_points`

The important thing is: first crecord, then aplay, then echo
You will have the /tmp/extract.dat file created, copy that to the host and run:
<sof root>/tools/build_tools/probes/sof-probes -p extract.dat
it will create a buffer_**2**.wav (for me in the example)
For some reason the wav header is incorrect, but you can play it back with `mpv` or import the file with audacity as raw and use these settings:
[image]

ujfalusi, 2023-09-26:
fwiw, with https://github.com/thesofproject/sof/pull/8263 the wav file should be correct, no need for the import as raw trick.

syedk008, 2023-10-05:
@ujfalusi, are below options enabled by default in the FW? We are using a production key signed Fw, so we can't change the settings and build the same at our end. I will check if we have any pre-production si boards available.
CONFIG_PROBE=y                  # enable probes
CONFIG_PROBE_POINTS_MAX=16      # max probepoints
CONFIG_LOG_BACKEND_SOF_PROBE=y
CONFIG_ZEPHYR_LOG=y

lgirdwood, 2023-10-05:
> @ujfalusi, are below options enabled by default in the FW? We are using a production key signed Fw, so we can't change the settings and build the same at our end. I will check if we have any [...]

Is this for v2.7 or mtl-06 as Kconfig may be different, but usually debug items are disabled.

syedk008, 2023-10-05:
> Is this for v2.7 or mtl-06 as Kconfig may be different, but usually debug items are disabled.

It's for v2.7. Currently, we do not have a pre-production si board and are in the process of getting one to collect logs for this issue.

marc-hb, 2023-10-05:
> Currently, we do not have a pre-production si board and are in the process of getting one to collect logs for this issue.

I don't know if it's possible in this case but you should also  ask for a system that accepts BOTH firmware images signed with production keys and non production keys. It maybe be possible because I've seen it in at least in one example in the past. It certainly should be possible.

BTW "pre-production" is very poor terminology. It should be called "development" which obviously does not stop the minute production starts.

jsarha, 2023-10-10:
I think we managed to reproduce this on the MTL RVP we have at Espoo lab. There is no need to plug and unplug HDMI cable. Just play 16-bit audio first to analog PCM (hw:0,0), then to HDMI (hw:0,3), and again to analog. Usually on the first cycle the 
audio gets severely distorted on the third playback. Sometimes it breaks only on second round, again at analog. After the audio on analog output is broken, its still Ok on HDMI, as long as its 16-bit audio. 32-bit audio is broken on HDMI too.

Here is one example sequence and the related kernel log. The audio was Ok on third playback, but was broken on 5th:

```
ubuntu@mtl-rvp-hda:~$` aplay -Dhw:0,0 -d5 vanhalen48.wav 
Playing WAVE 'vanhalen48.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
ubuntu@mtl-rvp-hda:~$ aplay -Dhw:0,3 -d5 vanhalen48.wav 
Playing WAVE 'vanhalen48.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
ubuntu@mtl-rvp-hda:~$ aplay -Dhw:0,0 -d5 vanhalen48.wav 
Playing WAVE 'vanhalen48.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
ubuntu@mtl-rvp-hda:~$ aplay -Dhw:0,3 -d5 vanhalen48.wav 
Playing WAVE 'vanhalen48.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
ubuntu@mtl-rvp-hda:~$ aplay -Dhw:0,0 -d5 vanhalen48.wav 
Playing WAVE 'vanhalen48.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
```

[last_dmesg.txt](https://github.com/thesofproject/sof/files/12854809/last_dmesg.txt)

ujfalusi, 2023-10-11:
A fix is submitted: https://github.com/zephyrproject-rtos/zephyr/pull/63784

ujfalusi, 2023-10-11:
@syedk008, can you try that patch on your setup?

lgirdwood, 2023-10-20:
@syedk008 are we pedning any further testing or can we close ?

keqiaozhang, 2023-10-23:
Fix has been merged on Zephyr main branch, closing this bug.

closed 2023-10-23 by keqiaozhang