thesofproject/sof #8238: [BUG] ramp time extended by 0.03s after clock source change for all SSPs

keqiaozhang, 2023-09-21:
**Describe the bug**
Observed this issue on all IPC4-nocodec(single core) platforms.  it caused alsabat test failures.
The ramp time extended by 0.03s after the clock source change(https://github.com/thesofproject/sof/commit/0912756595d5b89fca02a475c07d19d58c372e0d):
[image]

**To Reproduce**
check the alsabat wav file

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[cffcac365eac](https://github.com/thesofproject/linux/commit/cffcac365eac0d60dc38959cadaa5078bc924b8e)}
    * SOF: {[0912756595d5](https://github.com/thesofproject/sof/commit/0912756595d5b89fca02a475c07d19d58c372e0d)}
2) Name of the topology file
    * Topology: {nocodec tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {NOCODEC-IPC4 platforms}

marc-hb, 2023-09-21:
The commit message says "This change increases the ramp duration from 20 ms to 50 ms"

So it's not a bug, it's a feature? :-)

keqiaozhang, 2023-09-21:
> So it's not a bug, it's a feature? :-)

Not sure about this, but alsabat cannot survive from such long ramp time.

singalsu, 2023-09-21:
Ouch, but there's justification for this change. 1) capture artefacts mitigation accidentally dropped from tplg2 vs. tplg1 2) we can avoid a big MCPS peak with this lower rate ramp.

Is there a way to ask alsabat to ignore a bit long audio from beginning?

singalsu, 2023-09-21:
If not possible to change alsabat we can have a test topology with short ramp?

singalsu, 2023-09-21:
I could not find from alsabat command line options to ignore audio from very beginning of test. I don't like at all to change something because of limitation in our test procedure but this might help https://github.com/thesofproject/sof/pull/8242 .

lgirdwood, 2023-09-22:
@singalsu ack, the topology used for the test can have its ramp shortened, providing its not a end user topology. i.e. testing only topology.

marc-hb, 2023-09-25:
Let's wait for validation in the next daily test run.

Stupid Github auto close anti-feature: https://github.com/orgs/community/discussions/17308

keqiaozhang, 2023-09-26:
Confirmed that alsabat test passed after applying PRhttps://github.com/thesofproject/sof/pull/8242 on nocodec platforms. Closing this bug.

PR #8242 (2023-09-25): Tools: Topology2: Remove 50 ms curve duration set from nocodec
We need to care about audio user experience and peak MCPS usage in production topologies.

The alsabat test is disturbed by longer ramp so it can be removed from nocodec topologies since the codec is never used by end users and also the peak MCPS mitigation is not relevant for a test topology, as long as higher MCPS is not triggering error reports. The curve duration becomes without explicit set the default 20 ms.

Fixes: #8238
Fixes: d0d74a477f64a83654bccb3905175584d05d9257
       ("Tools: Topology2: Change in capture gain
        [... cut]
commit 2cdd11a9 (2023-09-25): Tools: Topology2: Remove 50 ms curve duration set from nocodec

We need to care about audio user experience and peak MCPS
usage in production topologies.

The alsabat test is disturbed by the longer ramp so the
change can be reverted from nocodec topologies. Those
topologies are never used by end users. Also the peak MCPS
mitigation is not relevant for test topologies, as long as
higher MCPS is not triggering error reports. The curve
duration is restored without explicit set to the default 20 ms.

Fixes: #8238
Fixes: d0d74a477f64a83654bccb3905175584d05d9257
       ("Tools: Topology2: Change in [... cut]
closed 2023-09-26 by keqiaozhang