thesofproject/sof #8241: [BUG] Null uid member in static const struct comp_driver comp_eq_iir

singalsu, 2023-09-21:
**Describe the bug**
We have seen randomly in CI testbench test case unrelated valgrind fails. E.g. in #8230 in cavs2.5-001-drop-stable

```
==20381== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==20381== Command: ../../testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -c 2 -n 2 -b S16_LE -t ../../build_tools/test/topology/test-playback-ssp5-mclk-0-I2S-eq-iir-s16le-s16le-48k-24576k-codec.tplg -i zeros_in.raw -o eqiir_out.raw
==20381== 
(ipc-common.c:239) ipc_init()
(ll_schedule.c:310) ll_scheduler_init()
(edf_schedule.c:115) edf_scheduler_init()
(helper.c:149) get_drv(), found driver type 1, uuid (nil)U
(helper.c:309) comp new (nil)U type 1 id 1.0
==20381== Thread 2:
==20381== Invalid read of size 8
==20381==    at 0x486AA79: get_drv (helper.c:130)
==20381==    by 0x486AA79: comp_new (helper.c:299)
==20381==    by 0x486B9EF: ipc_comp_new (helper.c:656)
==20381==    by 0x48B00[22](https://github.com/thesofproject/sof/actions/runs/6238577593/job/16934715708?pr=8230#step:7:23): load_process (process.c:507)
==20381==    by 0x48AEB55: load_widget (widget.c:146)
==20381==    by 0x1101B7: parse_topology (topology.c:472)
==20381==    by 0x10C5F7: test_pipeline_load (testbench.c:659)
==20381==    by 0x10C5F7: pipline_test (testbench.c:765)
==20381==    by 0x48C0608: start_thread (pthread_create.c:477)
==20381==    by 0x49FA1[32](https://github.com/thesofproject/sof/actions/runs/6238577593/job/16934715708?pr=8230#step:7:33): clone (clone.S:95)
==20[38](https://github.com/thesofproject/sof/actions/runs/6238577593/job/16934715708?pr=8230#step:7:39)1==  Address 0x0 is not stack'd, malloc'd or (recently) free'd

```

It caused by effort to memcpy to a null pointer:

[image]

The compile time initialized value in the struct from C preprocessor and macro is SOF_RT_UUID(eq_iir_uuid) is` ((void *)0)`. The macro is possibly incorrect for testbench in this cavs2.5 branch.

**To Reproduce**
scripts/host-testbench.sh

**Reproduction Rate**
Unknown, possibly only with cavs2.5-001-drop-stable

**Expected behavior**
Pass the valgrind test

**Impact**
Annoyance, false failures in CI unless proven it can happen outside testbench env.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: 85732fbdd79e6c1b725096ef7ecbd09123ee775d
2) Name of the topology file
    * Topology: test-playback-ssp5-mclk-0-I2S-eq-iir-s16le-s16le-48k-24576k-codec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: testbench x86

**Screenshots or console output**

==========================================================
test eqiir with ./eqiir_run.sh 16 16 48000 zeros_in.raw eqiir_out.raw
eqiir test failed!
Command:         ../../testbench/build_testbench/install/bin/testbench
Argument:         -r 48000 -R 48000 -c 2 -n 2 -b S16_LE -t ../../build_tools/test/topology/test-playback-ssp5-mclk-0-I2S-eq-iir-s16le-s16le-48k-24576k-codec.tplg  -i zeros_in.raw -o eqiir_out.raw
LD_LIBRARY_PATH=../../testbench/build_testbench/sof_ep/install/lib:../../testbench/build_testbench/sof_parser/install/lib
==363525== Memcheck, a memory error detector
==363525== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==363525== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==363525== Command: ../../testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -c 2 -n 2 -b S16_LE -t ../../build_tools/test/topology/test-playback-ssp5-mclk-0-I2S-eq-iir-s16le-s16le-48k-24576k-codec.tplg -i zeros_in.raw -o eqiir_out.raw
==363525== 
(ipc-common.c:239) ipc_init()
(ll_schedule.c:310) ll_scheduler_init()
[... 154 lines cut ...]
==363525==    by 0x4004B2B: _dl_find_object_update (dl-find_object.c:805)
==363525==    by 0x400D01F: dl_open_worker_begin (dl-open.c:735)
==363525==    by 0x40014DC: _dl_catch_exception (dl-catch.c:237)
==363525==    by 0x400C2E9: dl_open_worker (dl-open.c:782)
==363525==    by 0x40014DC: _dl_catch_exception (dl-catch.c:237)
==363525==    by 0x400C6BB: _dl_open (dl-open.c:884)
==363525==    by 0x4962BEB: dlopen_doit (dlopen.c:56)
==363525==    by 0x40014DC: _dl_catch_exception (dl-catch.c:237)
==363525==    by 0x4001602: _dl_catch_error (dl-catch.c:256)
==363525==    by 0x49626BE: _dlerror_run (dlerror.c:138)
==363525==    by 0x4962CA0: dlopen_implementation (dlopen.c:71)
==363525==    by 0x4962CA0: dlopen@@GLIBC_2.34 (dlopen.c:81)
==363525== 
==363525== LEAK SUMMARY:
==363525==    definitely lost: 0 bytes in 0 blocks
==363525==    indirectly lost: 0 bytes in 0 blocks
==363525==      possibly lost: 3,349 bytes in 9 blocks
==363525==    still reachable: 17,749 bytes in 31 blocks
==363525==         suppressed: 0 bytes in 0 blocks
==363525== Reachable blocks (those to which a pointer was found) are not shown.
==363525== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==363525== 
==363525== For lists of detected and suppressed errors, rerun with: -s
==363525== ERROR SUMMARY: 10 errors from 10 contexts (suppressed: 0 from 0)
./comp_run.sh: line 110: 363525 Segmentation fault      (core dumped) $VALGRIND_CMD $CMD

singalsu, 2023-09-21:
@marc-hb FYI

lgirdwood, 2023-09-21:
@singalsu this is probably the tplg parser doing a bzero() on the data and not populating a member under certain conditions.

marc-hb, 2023-09-22:
Reproduced again in https://github.com/thesofproject/sof/actions/runs/6277952750/job/17050714146?pr=8253
(#8253)

Interestingly, qemu CNL seems to fail at the same time. Coincidence?