thesofproject/sof #8244: [BUG] Fujitsu FMV LOOX tablet needs the sof-hda-generic-2ch-pdm1 topology

auspicacious, 2023-09-22:
Thank you for your very helpful [Suggestions before filing an SOF bug](https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html) page.

Using the steps there, I was able to confirm that the [Fujitsu FMV LOOX FMVWGL17 tablet released in 2022](https://www.fmworld.net/overseas/en/loox/2206/) uses the PDM1 controller for its digital microphones. By setting:

```
options snd-sof-pci tplg_filename=sof-hda-generic-2ch-pdm1.tplg
```

in the file `/etc/modprobe.d/alsa-base.conf`, I was able to get my device's internal microphone working.

I'm attaching the output of `./alsa-info.sh`; please let me know if I can do anything else to help. Thanks again.

[alsa-info.txt](https://github.com/thesofproject/sof/files/12697190/alsa-info.txt)

lgirdwood, 2023-09-25:
@plbossart @ujfalusi do we publish this fix anywhere ?

plbossart, 2023-09-25:
The workaround is documented here: https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#digital-mic-issues

We are not going to do anything here since we don't have any means to discover the hardware, and no desire to start a DMI quirk table.

auspicacious, 2023-09-27:
I see. This sentence in the "Suggestions" page had led me to think that filing a bug was the right thing to do:

> These PDM1 issues are tracked in GitHub with the label ‘DMIC-PDM1’ in the [firmware issues](https://github.com/thesofproject/sof/issues?q=is%3Aissue+label%3ADMIC-PDM1) and in the [Linux issues](https://github.com/thesofproject/linux/issues?q=is%3Aopen+is%3Aissue+label%3ADMIC-PDM1).

In any case, opening this GitHub issue hopefully means that anyone else with this same hardware will more easily find a solution in the future.

closed 2025-07-24 by github-actions[bot]