thesofproject/sof #8247: ADL: Load tplg file and return lots of errors from SOF driver

davidrau-renesas-opensource, 2023-09-22:
**1. Environment:**
- Intel 12th ADL platform + **CS42L42-SDW** codec
- Linux v6.5

**2. Situation:**
I  tried the available tplg files (for ADL) to bring-up the onboard codec and then got lots of error message from SOF driver.
(e.g. sof-adl-nocodec.tplg, )

[   15.592511] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[   15.592762] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[   15.592782] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[   15.593274] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   15.593476] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   15.600749] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[   15.771673] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   15.771674] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[   15.975370] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:0:0-b678a
[   15.975374] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:20:0 Kernel ABI 3:23:0
[   15.975382] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   16.085714] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:0:0-b678a
[   16.085731] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:20:0 Kernel ABI 3:23:0
[   16.205352] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:20:0 Kernel ABI 3:23:0
[   16.205416] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI ALH2.OUT stream SDW0-Playback
[   16.207010] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : ALH2.OUT stream SDW0-Playback
[   16.210194] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI ALH3.IN stream SDW0-Capture
[   16.211317] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 28 name : ALH3.IN stream SDW0-Capture
[   16.213025] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI HDA3.OUT stream iDisp4
[   16.213423] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : HDA3.OUT stream iDisp4
[   16.214678] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI SSP2.OUT stream SSP2-BT
[   16.215092] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : SSP2.OUT stream SSP2-BT
[   16.216325] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI SSP2.IN stream SSP2-BT
[   16.216741] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 28 name : SSP2.IN stream SSP2-BT
[   16.218060] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink ALH2.OUT not found
[   16.219458] sof-audio-pci-intel-tgl 0000:00:1f.3: error: source ALH3.IN not found
[   16.220794] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink HDA3.OUT not found
[   16.220796] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink SSP2.OUT not found
[   16.220798] sof-audio-pci-intel-tgl 0000:00:1f.3: error: source SSP2.IN not found
[   16.226628] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30030000 (msg/reply size: 16/0): -22
[   16.226631] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc3_route_setup: route BUF14.0 -> PCM14C failed
[   16.226632] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc3_set_up_all_pipelines: route set up failed
[   16.226633] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[   16.226638] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[   16.226640] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22 

Some questions as following.
- Would you please let me know how to solve this errors to make the codec become workable?
- Are there any available tplg file support this codec in ADL platform?
- May I refer to the existing tplg file (*.m4) to create a new one to access this codec correctly?

Thank you.

Best,
David

lgirdwood, 2023-09-22:
@charleskeepax @rfvirgil @bardliao is there a topology file for CS42L42 ?

lgirdwood, 2023-09-22:
@davidrau-renesas-opensource there may also be driver updates needed too, best to align with @charleskeepax and @rfvirgil here.

davidrau-renesas-opensource, 2023-09-25:
Hi @lgirdwood ,
Thanks for the prompt response.

> @davidrau-renesas-opensource there may also be driver updates needed too, best to align with @charleskeepax and @rfvirgil here.

Hi charleskeepax and Richard,
Nice to meet you. 
This is David from Renesas.

My current test environment bases on Linux_v6.5.
I took some time to bring-up the on-board codec CS42L42-SDW but still failed of parsing the tplg.

Would you please let me know if any available tplg file for configuring this codec with SoundWire interface correctly in ADL platform?
Or any sugeestions to fix it on my own?

Thanks.

Best,
David

bardliao, 2023-09-25:
> @charleskeepax @rfvirgil @bardliao is there a topology file for CS42L42 ?

We have sof-mtl-sdw-cs42l42-l0-max98363-l2.tplg for MTL, but not for ADL.
https://github.com/thesofproject/sof/blob/main/tools/topology/topology2/sof-ace-tplg/tplg-targets.cmake#L71

@davidrau-renesas-opensource Please rename [sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/12712905/sof-dyndbg.conf.txt) to `sof-dyndbg.conf` and put it in the `/etc/modprobe.d/` folder and get the dmesg again.

BTW, do you have the audio HW configuration in detail? For example, cs42l42 in sdw link 0, max98363 in sdw link 2.

davidrau-renesas-opensource, 2023-09-26:
@bardliao 
Thanks for the kind response.
Please refer to the following info about my current modifications to probe CS42L42-SDW in ADL.

According to the info of "sdw_intel_ctx", the default ids is configured for RT711.
In order to pass the default ids checking mechanism in sof-hda driver, I created my own fake id table in ADL ACPI code.
`static const struct snd_soc_acpi_adr_device cs42l42_adr[] = {`
`        {`
`                .adr = 0x000120025D071101ull,`
`                .num_endpoints = 1,`
`                .endpoints = &single_endpoint,`
`                .name_prefix = "cs42l42-sdw"`
`        }`
`};`

The attachment is my complete bootup log message after applying sof-dyndbg.conf.
[cs42l42.log](https://github.com/thesofproject/sof/files/12726133/cs42l42.log)

Please provide me any suggestions about this log message.
Thanks  a lot. :)

bardliao, 2023-09-26:
@davidrau-renesas-opensource It will be easier to connect the cs42l42 on sdw link 0.
Please make sure sdw link 0 and only sdw link 0 is enabled in the BIOS.
Please follow the instruction in https://github.com/thesofproject/acpi-scripts to add cs42l42 sdw adr.

`sudo ./install_hooks`
`sudo acpi-add <directory>/*.asl`

Here is an example for adl cs42l42 [adl-cs42l42.asl.txt](https://github.com/thesofproject/sof/files/12727469/adl-cs42l42.asl.txt)
Here is the kernel patch
[0001-ASoC-Intel-soc-acpi-intel-adl-match-add-cs42l42-supp.patch](https://github.com/thesofproject/sof/files/12727470/0001-ASoC-Intel-soc-acpi-intel-adl-match-add-cs42l42-supp.patch)

And you can rename sof-adl-rt711-4ch.tplg and sof-adl-rt711.tplg to sof-adl-cs42l42-4ch.tplg and sof-adl-cs42l42.tplg

Please share the dmesg with those changes.

davidrau-renesas-opensource, 2023-09-27:
@bardliao 
Thanks for the prompt response.
According to on my "HD Audio configuration" of BIOS setting, there are "SNDW #1" and "SNDW #2" options.
So I enabled  "SNDW #1" only and the log message shows the link mask is "0x2" as below.
`[   16.206603] hda_sdw_machine_select() - link_mask = 0x2`
=> Does "link_mask = 0x2" mean the link#1 is connected in my dev-kit?

If my understanding is correct, the related setting should be confiuged to "link#1". Is't it?  :)
E.g. (adl-cs42l42.asl)
`Scope (_SB.PC00.HDAS.SNDW) {`
`	Device (CS1) // CS42L42 on link1`
`        {`
`            Name (_ADR, 0x00000001fa424200)  // _ADR: Address`
`        }`
`  }`

And the following modificatins of ADL ACPI code are also needed?
`.link_mask = 0x2`
`.mask = BIT(1)`

davidrau-renesas-opensource, 2023-09-27:
@bardliao 
I tried to run acpi-add command with adl-cs42l42.asl (with link#1) and integrate you patch into my test environment.
There are 2 log messages to let you know my current test cases.
1. acpi-add command + integrate kernel patch: 
[cs42l42-add-acpi-1.txt](https://github.com/thesofproject/sof/files/12737148/cs42l42-add-acpi-1.txt)
=> The default ids for RT711 is requested and then load the default hda_dsp_generic tplg.

2. acpi-add command + integrate kernel patch + bypass default ids checking in hda.c:
[cs42l42-add-acpi-2.txt](https://github.com/thesofproject/sof/files/12737150/cs42l42-add-acpi-2.txt)
=>  sof-adl-sdw-cs42l42-4ch.tplg is loaded but slave Entry not found.

And ideas about my test result?
Thanks.

bardliao, 2023-09-28:
> @bardliao Thanks for the prompt response. According to on my "HD Audio configuration" of BIOS setting, there are "SNDW #1" and "SNDW #2" options. So I enabled "SNDW #1" only and the log message [...]

It means your cs42l43 should be connected to SNDW_DAT1 and SNDW_CLK1. And it looks like there is no SNDW #0 option in your BIOS. And my patch assumes the cs42l42 is connected to SNDW_DAT0 and SNDW_CLK0

Let's connect the cs42l42 on SNDW_DAT0 and SNDW_CLK0 and use kernel option to overwrite the mask to 0x1.
`options snd_intel_sdw_acpi sdw_link_mask=0x1`

> If my understanding is correct, the related setting should be confiuged to "link#1". Is't it? :) E.g. (adl-cs42l42.asl) `Scope (_SB.PC00.HDAS.SNDW) {` ` Device (CS1) // CS42L42 on link1` ` {` ` Name [...]

Connecting your cs42l42 to link 0 can reuse the existing topology. Yes, we can use link 1, and all above change is correct. .You need to change adr to 0x000**1**0001fa424200ull, in ` cs42l42_0_adr[]`, too.  And you need to connect your cs42l42 to SNDW_CLK1 and SNDW_DAT1. However, in this case, you will need to change the topology. 
```patch
diff --git a/tools/topology/topology1/sof-tgl-rt711-rt1308.m4 b/tools/topology/topology1/sof-tgl-rt711-rt1308.m4
index 73f3bc7b67ae..ed92fe67ad28 100644
--- a/tools/topology/topology1/sof-tgl-rt711-rt1308.m4
+++ b/tools/topology/topology1/sof-tgl-rt711-rt1308.m4
@@ -166,14 +166,14 @@ dnl     deadline, priority, core, time_domain)
 # playback DAI is ALH(SDW0 PIN2) using 2 periods
 # Buffers use s24le format, with 48 frame per 1000us on core 0 with priority 0
 DAI_ADD(sof/pipe-mixer-volume-dai-playback.m4,
-       1, ALH, 2, SDW0-Playback,
+       1, ALH, 258, SDW0-Playback,
        NOT_USED_IGNORED, 2, s24le,
        1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER, 2, 48000)

 # capture DAI is ALH(SDW0 PIN2) using 2 periods
 # Buffers use s24le format, with 48 frame per 1000us on core 0 with priority 0
 DAI_ADD(sof/pipe-dai-capture.m4,
-       2, ALH, 3, SDW0-Capture,
+       2, ALH, 259, SDW0-Capture,
        PIPELINE_SINK_2, 2, s24le,
        1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)
```
Rebuild the topology and use the sof-adl-rt711-4ch.tplg

davidrau-renesas-opensource, 2023-10-03:
@bardliao 

> You need to change adr to 0x00010001fa424200ull, in  cs42l42_0_adr[]

I also modified the _ADR in "adl-cs42l42.asl" as below.
` Device (CS1) // CS42L42 on link1`
`        {`
`            Name (_ADR, 0x00010001fa424200)  // _ADR: Address`
`        }`

According to the boot message, the ACPI table is already applied to the sysytem successfully.
`[    0.020462] ACPI: SSDT ACPI table found in initrd [kernel/firmware/acpi/adl-cs42l42.aml][0x66]`

And then the following messages tell us the previous modifications we did are correct. doesn't?
`[   15.091262] hda_sdw_machine_select() - link_mask = 0x2`
`[   15.091263] mfg_id = 1fa, part_id = 4242, link_id = 1, version = 0`
`[   15.091264] ids[0].link_id = 1, ids[0].id.part_id = 711, ids[0].id.mfg_id = 25d, ids[0].id.sdw_version = 2`
`[   15.091265] ids[1].link_id = 1, ids[1].id.part_id = 4242, ids[1].id.mfg_id = 1fa, ids[1].id.sdw_version = 0`
`[   15.091267] snd_sof_intel_hda_common:link_slaves_found: sof-audio-pci-intel-tgl 0000:00:1f.3: found 4242 at link 1`

Next, the SOF driver loads the tplg file we created "sof-adl-sdw-cs42l42-4ch.tplg".
It seems "ALH258.out" is not workable in my current environment.
`[   15.711478] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI ALH258.OUT stream SDW0-Playback`
`[   15.712989] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : ALH258.OUT stream SDW0-Playback`
`[   15.714277] sof_sdw sof_sdw: ASoC: failed to load widget ALH258.OUT`
`[   15.715698] sof_sdw sof_sdw: ASoC: topology: could not load header: -22`

And ideas?

The complete log message is for your reference.
[cs42l42-link1.txt](https://github.com/thesofproject/sof/files/12788478/cs42l42-link1.txt)

Thank. you.

bardliao, 2023-10-03:
@davidrau-renesas-opensource Sorry, I missed something in the pervious topology patch. We need to change the BE name, too.
Please use the new patch.
```patch
diff --git a/tools/topology/topology1/sof-tgl-rt711-rt1308.m4 b/tools/topology/topology1/sof-tgl-rt711-rt1308.m4
index 73f3bc7b67ae..fc192f981d8c 100644
--- a/tools/topology/topology1/sof-tgl-rt711-rt1308.m4
+++ b/tools/topology/topology1/sof-tgl-rt711-rt1308.m4
@@ -166,14 +166,14 @@ dnl     deadline, priority, core, time_domain)
 # playback DAI is ALH(SDW0 PIN2) using 2 periods
 # Buffers use s24le format, with 48 frame per 1000us on core 0 with priority 0
 DAI_ADD(sof/pipe-mixer-volume-dai-playback.m4,
-       1, ALH, 2, SDW0-Playback,
+       1, ALH, 258, SDW1-Playback,
        NOT_USED_IGNORED, 2, s24le,
        1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER, 2, 48000)

 # capture DAI is ALH(SDW0 PIN2) using 2 periods
 # Buffers use s24le format, with 48 frame per 1000us on core 0 with priority 0
 DAI_ADD(sof/pipe-dai-capture.m4,
-       2, ALH, 3, SDW0-Capture,
+       2, ALH, 259, SDW1-Capture,
        PIPELINE_SINK_2, 2, s24le,
        1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)

@@ -276,12 +276,12 @@ PCM_PLAYBACK_ADD(HDMI 4, 8, PIPELINE_PCM_9)

 #ALH dai index = ((link_id << 8) | PDI id)
 #ALH SDW0 Pin2 (ID: 0)
-DAI_CONFIG(ALH, 2, 0, SDW0-Playback,
-       ALH_CONFIG(ALH_CONFIG_DATA(ALH, 2, 48000, 2)))
+DAI_CONFIG(ALH, 258, 0, SDW1-Playback,
+       ALH_CONFIG(ALH_CONFIG_DATA(ALH, 258, 48000, 2)))

 #ALH SDW0 Pin3 (ID: 1)
-DAI_CONFIG(ALH, 3, 1, SDW0-Capture,
-       ALH_CONFIG(ALH_CONFIG_DATA(ALH, 3, 48000, 2)))
+DAI_CONFIG(ALH, 259, 1, SDW1-Capture,
+       ALH_CONFIG(ALH_CONFIG_DATA(ALH, 259, 48000, 2)))

 ifdef(`EXT_AMP',
 `
```

davidrau-renesas-opensource, 2023-10-03:
@bardliao 
Thanks for the prompt response.

After applying the new patch of tplg file, it shows the different error message about "DAI HDA3.OUT stream iDisp4".
`[   16.516344] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI HDA3.OUT stream iDisp4`
`[   16.517698] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : HDA3.OUT stream iDisp4`
`[   16.519085] sof_sdw sof_sdw: ASoC: failed to load widget HDA3.OUT`
`[   16.520064] sof_sdw sof_sdw: ASoC: topology: could not load header: -22`

So I took a try to disable the related definitations of "iDisp4" and then the new error mesage is about "SSP2.OUT stream SSP2-BT"
`[   16.896588] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI SSP2.OUT stream SSP2-BT`
`[   16.897910] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : SSP2.OUT stream SSP2-BT`
`[   16.899346] sof_sdw sof_sdw: ASoC: failed to load widget SSP2.OUT`
`[   16.900003] sof_sdw sof_sdw: ASoC: topology: could not load header: -22`

Any ideas?
Thanks.

bardliao, 2023-10-03:
@davidrau-renesas-opensource It looks like there is no iDisp4 in your device. Please remove the HDA 3 stuff from the topology.
[sof-tgl-rt711-rt1308.diff.txt](https://github.com/thesofproject/sof/files/12790439/sof-tgl-rt711-rt1308.diff.txt)

davidrau-renesas-opensource, 2023-10-03:
@bardliao 
HDA 3 stuff is already removed in the tplg now.
The remaining error message shows "SSP2.OUT stream SSP2-BT" is not avaiable in my device.
`[   16.611837] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI SSP2.OUT stream SSP2-BT`
`[   16.613204] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : SSP2.OUT stream SSP2-BT`
`[   16.614456] sof_sdw sof_sdw: ASoC: failed to load widget SSP2.OUT`
`[   16.615222] sof_sdw sof_sdw: ASoC: topology: could not load header: -22`

Would you please let me know how to remove this stuff correctly?

Thanks.

davidrau-renesas-opensource, 2023-10-03:
@bardliao 
I tried to disable this stuff by removie "-DBT_OFFLOAD" in the CMakeLists.txt. Is it correct?
Now the error message as following.
`[   16.193301] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30030000 (msg/reply size: 16/0): -22`
`[   16.194218] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc3_route_setup: route PGA9.0 -> BUF9.1 failed`
`[   16.195513] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc3_set_up_all_pipelines: route set up failed`
`[   16.196801] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22`
`[   16.197797] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22`

Any ideas?
Thanks.

bardliao, 2023-10-03:
> @bardliao I tried to disable this stuff by removie "-DBT_OFFLOAD" in the CMakeLists.txt. Is it correct? 

Yes, that is correct.

> Now the error message as following. `[ 16.193301] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30030000 (msg/reply size: 16/0): -22` `[ 16.194218] sof-audio-pci-intel-tgl 0000:00:1f.3: [...]

Please remove
```
# Low Latency playback pipeline 9 on PCM 8 using max 2 channels of s32le.
# Schedule 48 frames per 1000us deadline with priority 0 on core 0
PIPELINE_PCM_ADD(sof/pipe-volume-playback.m4,
        9, 8, 2, s32le,
        1000, 0, 0,
        48000, 48000, 48000)
```
for this error.

davidrau-renesas-opensource, 2023-10-04:
@bardliao
Thanks for the prompt response.

After removing this stuff,  the current error message shows something wrong when adding cs42l42 routes. 
`[   15.714875] sof_sdw sof_sdw: ASoC: Failed to add route cs42l42 HP(*) -> Headphone`
`[   15.716262] sof_sdw sof_sdw: ASoC: Failed to add route Headset Mic -> cs42l42 HS(*)`
`[   15.717411] sof_sdw sof_sdw: cs42l42 map addition failed: -19`
`[   15.718416]  SDW1-Playback: ASoC: error at snd_soc_link_init on SDW1-Playback: -19`
`[   15.720196] sof_sdw sof_sdw: snd_soc_register_card failed -19`

I guess there are still incompatible configurations between the current tplg we applied and sof_sdw_cs42l42 helper code.
(The similar definitions are also included in sof_sdw_rt711 helper code.)

Do you have any suggestions to modify below dapm route in "sof_sdw_cs42l42.c" to fix this error?
`static const struct snd_soc_dapm_route cs42l42_map[] = {`
`        /* HP jack connectors - unknown if we have jack detection */`
`        {"Headphone", NULL, "cs42l42 HP"},`
`        /* other jacks */`
`        {"cs42l42 HS", NULL, "Headset Mic"},`
`};`

The complete log message is for your reference.
[cs42l42-add-route.txt](https://github.com/thesofproject/sof/files/12798867/cs42l42-add-route.txt)

Thanks.

bardliao, 2023-10-04:
@davidrau-renesas-opensource sof_sdw_cs42l42.c is shared with other machine drivers. What we need to do is to change the name_prefix to "cs42l42".
```
static const struct snd_soc_acpi_adr_device cs42l42_0_adr[] = {
	{
		.adr = 0x00000001fa424200ull,
		.num_endpoints = 1,
		.endpoints = &single_endpoint,
		.name_prefix = "cs42l42"
	}
};
```

davidrau-renesas-opensource, 2023-10-05:
@bardliao 
Thanks a lot for the kind support.
Now the CS42L42 codec is registered into Linux system successfully.

When verifying playback function, I executed aplay command but no output from headphone.
`$ sudo aplay -Dplughw:0,0 music.wav -v`
`Playing WAVE 'music.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo`
`Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)`
`Converter: linear-interpolation`
`Protocol version: 10003`
`Its setup is:`
`  stream       : PLAYBACK`
`  access       : RW_INTERLEAVED`
`  format       : S16_LE`
`  subformat    : STD`
`  channels     : 2`
`  rate         : 44100`
`  exact rate   : 44100 (44100/1)`
`  msbits       : 16`
`  buffer_size  : 15052`
`  period_size  : 3763`
`  period_time  : 85333`
`  tstamp_mode  : NONE`
`  tstamp_type  : MONOTONIC`
`  period_step  : 1`
`  avail_min    : 3763`
`  period_event : 0`
`  start_threshold  : 15052`
`  stop_threshold   : 15052`
`  silence_threshold: 0`
`  silence_size : 0`
`  boundary     : 4236761349448794112`
`Slave: Hardware PCM card 0 'sof-soundwire' device 0 subdevice 0`
`Its setup is:`
`  stream       : PLAYBACK`
`  access       : MMAP_INTERLEAVED`
`  format       : S16_LE`
`  subformat    : STD`
`  channels     : 2`
`  rate         : 48000`
`  exact rate   : 48000 (48000/1)`
`  msbits       : 16`
`  buffer_size  : 16384`
`  period_size  : 4096`
`  period_time  : 85333`
`  tstamp_mode  : NONE`
`  tstamp_type  : MONOTONIC`
`  period_step  : 1`
`  avail_min    : 4096`
`  period_event : 0`
`  start_threshold  : 16384`
`  stop_threshold   : 16384`
`  silence_threshold: 0`
`  silence_size : 0`
`  boundary     : 4611686018427387904`
`  appl_ptr     : 0`
`  hw_ptr       : 0`
Do you have any suggestions to configure CS42L42 to output audio data correctly?

About recording function, there is error message about hw parameters.
`$ sudo arecord -D plughw:0,1 -r48000 -c 2 -f S16_LE record-test.wav`
`Recording WAVE 'record-test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo`
`arecord: set_params:1416: Unable to install hw params:`
`ACCESS:  RW_INTERLEAVED`
`FORMAT:  S16_LE`
`SUBFORMAT:  STD`
`SAMPLE_BITS: 16`
`FRAME_BITS: 32`
`CHANNELS: 2`
`RATE: 48000`
`PERIOD_TIME: (85333 85334)`
`PERIOD_SIZE: 4096`
`PERIOD_BYTES: 16384`
`PERIODS: 4`
`BUFFER_TIME: (341333 341334)`
`BUFFER_SIZE: 16384`
`BUFFER_BYTES: 65536`
`TICK_TIME: 0`
It looks like some configurations are incompatible for CS42L42 capture function.
Would you please provide me any suggestions about this issue?

Thanks.

bardliao, 2023-10-05:
@deb-intel Can you share the "amixer contents" result? And please share the dmesg log while playback and capture.

davidrau-renesas-opensource, 2023-10-06:
@bardliao 
Please refer to the following info and the complete dmesg log.
[amixer-contents.txt](https://github.com/thesofproject/sof/files/12826420/amixer-contents.txt)
[cs42l42-playback-dmesg.txt](https://github.com/thesofproject/sof/files/12826422/cs42l42-playback-dmesg.txt)
[cs42l42-capture-dmesg.txt](https://github.com/thesofproject/sof/files/12826424/cs42l42-capture-dmesg.txt)

When running capture function by arecord command, the below error shows in the dmesg log.
`[  148.028626] cs42l42-sdw sdw:1:01fa:4242:00: Chn prep failed for port 1: -110`
`[  148.028711] soundwire sdw-master-0: Prepare port(s) failed ret = -110`
`[  148.028758]  SDW1-Capture: ASoC: error at snd_soc_link_prepare on SDW1-Capture: -110`
`[  148.028805]  SDW1-Capture: ASoC: error at __soc_pcm_prepare on SDW1-Capture: -110`
`[  148.028846]  CS42L42 Jack In: ASoC: error at dpcm_be_dai_prepare on CS42L42 Jack In: -110`
`[  148.028935]  CS42L42 Jack In: ASoC: error at dpcm_fe_dai_prepare on CS42L42 Jack In: -110`
`[  148.029355] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 1 dir 1`
`[  148.029368] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000`
`[  148.029709] sdw_deprepare_stream: subdevice #0-Capture: inconsistent state state 1`

Thanks.

bardliao, 2023-10-06:
@davidrau-renesas-opensource Regarding the amixer, you need to set below settings
```
amixer cset name='Headset Mic Switch' 1
amixer cset name='Headphone Switch' 1
amixer cset name='PGA2.0 2 Master Capture Switch' 1
amixer cset name='PGA2.0 2 Master Capture Volume' 50
```
For "cs42l42-sdw sdw:1:01fa:4242:00: Chn prep failed for port 1: -110", that is cs42l42 failed to prepare. Can you contact Cirrus Logic folks for help?

plbossart, 2023-10-06:
@rfvirgil @charleskeepax FYI

plbossart, 2023-10-06:
````
[  148.028626] cs42l42-sdw sdw:1:01fa:4242:00: Chn prep failed for port 1: -110
[  148.028711] soundwire sdw-master-0: Prepare port(s) failed ret = -110
````

this is another instance of a ETIMEDOUT response to a prepare step, but this time on the channel prepare. it's the same NotFinished type of logic, so that's suspicious.

davidrau-renesas-opensource, 2023-10-12:
> @davidrau-renesas-opensource Regarding the amixer, you need to set below settings
> 
> ```
> amixer cset name='Headset Mic Switch' 1
> amixer cset name='Headphone Switch' 1
> amixer cset name='PGA2.0 2 Master Capture Switch' 1
> amixer cset name='PGA2.0 2 Master Capture Volume' 50
> ```

@bardliao 
Thanks for the kind support always.
Now I can hear the audio output from CS42L42 normally after setting up the below configurations:
`amixer cset name='Headphone Switch' 1`
`amixer set 'cs42l42 Mixer' 30`
(The default setting of "CS42L42_MIXER_CHA_VOL" and "CS42L42_MIXER_CHB_VOL" are muted)

lgirdwood, 2023-10-16:
@davidrau-renesas-opensource pls do upstream any topology or UCM changes needed to get this config working.
@rfvirgil @charleskeepax you guys will need to review UCM changes.

davidrau-renesas-opensource, 2023-10-17:
> @davidrau-renesas-opensource pls do upstream any topology or UCM changes needed to get this config working. 
@lgirdwood 
Thanks for the reminder. :)
Do you think it'd be better to do upstream topology file after cs42l42 recording function is verified completely?

lgirdwood, 2023-10-17:
> > @davidrau-renesas-opensource pls do upstream any topology or UCM changes needed to get this config working.
> 
> @lgirdwood Thanks for the reminder. :) Do you think it'd be better to do upstream topology file after cs42l42 recording function is verified completely?

Yes, and also once @rfvirgil and @charleskeepax also approve the UCM and codec config.

davidrau-renesas-opensource, 2023-10-24:
Already created a pull request: [8370](https://github.com/thesofproject/sof/pull/8370)
It is okay for me to close this ticket if no further needed actions.
Thanks a lot for all your great support. :)

closed 2025-07-24 by github-actions[bot]