thesofproject/sof #8250: [BUG] [Windows CI] CMake was unable to find a build program corresponding to "Ninja"

marc-hb, 2023-09-22:
**Describe the bug**

The Windows build workflow recently started failing intermittently in Github.

```
CMake Error: CMake was unable to find a build program corresponding to "Ninja".
   CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
In dir: D:\a\sof\sof\workspace; running command:
    cmake -B build-rimage -G Ninja -S 'D:\a\sof\sof\workspace\sof\rimage'
```

**To Reproduce**

Unknown

**Reproduction Rate**

Started rarely bug getting much worse:

- September 20 daily
1/5 build-windows fail
https://github.com/thesofproject/sof/actions/runs/6255625447/job/16985259003

- September 21 daily
5/5 build-windows fail
https://github.com/thesofproject/sof/actions/runs/6268589962

marc-hb, 2023-09-22:
I has a look at the logs of about 30 recent Windows builds.

- All the ones that passed listed Windows image version **20230910.1** in their initial "Setup" step.
https://github.com/actions/runner-images/blob/win22/20230910.1/images/win/Windows2022-Readme.md

- All the ones that failed listed more recent Windows image version **20230918.1** in their initial "Setup" step.
https://github.com/actions/runner-images/blob/win22/20230918.1/images/win/Windows2022-Readme.md

marc-hb, 2023-09-22:
`choco install ninja` seems to work

- #8254

kv2019i, 2023-09-25:
CI tests now green, closing.

closed 2023-09-25 by kv2019i

marc-hb, 2023-10-25:
In the same run https://github.com/thesofproject/sof/actions/runs/6639325877/job/18037461117?pr=8387

- 20231016.1.0 ninja found
- 20231023.1.0 not found

(stable-v2.7 branch)