thesofproject/sof #8260: [FEATURE][Zephyr] Remove duplication of CMake rules between Zephyr and XTOS

kv2019i, 2023-09-25:
**Is your feature request related to a problem? Please describe.**
When a software module is added to SOF that can be used both in XTOs and in Zephyr builds, the module sources need to be added to CMake rules twice:
 - first in sof/src/foo/CMakeLists.txt (for XTOS), and 
 - second time to sof/zephyr/CMakeLists.txt (for Zephyr)

This split of Cmake rules dates back to the time when Zephyr support was introduced and codebase was not cleanly split between operating systems. Now we have a cleaner split at top-level source directories, so it should be possible to start cleaning up this overlap in definitions.

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**
Platforms where transition to Zephyr is still in progress might cause some extra challenges to doing this makefile cleanup. E.g. see [XTOS-only a](https://github.com/thesofproject/sof/issues/7248)

kv2019i, 2023-09-25:
FYI @dbaluta @lgirdwood @marc-hb @marcinszkudlinski -- not tagging to v2.8 yet. A bit more feasibility and work effort analysis is needed to gauge the effort and options. But sooner or later we need to get rid of the duplication.

lyakh, 2023-10-26:
the best would be of course if we could re-use CMakeLists.txt in subdirectories for both XTOS and Zephyr. If this is impossible, we could move XTOS specific code to xtos.cmake, add Zephyr to zephyr.cmake and put any common code in CMakeLists.txt

marc-hb, 2023-10-26:
> if we could re-use CMakeLists.txt in subdirectories for both XTOS and Zephyr.

I noticed `sof/zephyr/CMakeLists.txt` uses `zephyr_sources()`, not the plain `target_sources()`. I'm not sure why. I found these:
- https://github.com/zephyrproject-rtos/zephyr/issues/8439
- https://github.com/zephyrproject-rtos/zephyr/discussions/43650

`zephyr/samples/**/CMakeLists.txt` and `zephyr/tests/**/CMakeLists.txt` use `target_sources()`.

lgirdwood, 2023-10-30:
Can we drop target_sources everywhere and use zephyr sources e.g. somethimg like for xtos
```
#define zephyr_sources(x) target_sources(x)
```

marc-hb, 2023-10-31:
> I agree this location is inconvenient. Let me think about it.

I have something promising, stay tuned.

marc-hb, 2023-10-31:
Submitted
- #8422

lgirdwood, 2023-11-10:
@marc-hb what do you think the next steps are now that #8422 is merged ?

marc-hb, 2023-11-10:
My plan is to wait ~1 week for the sky to fall (e.g: people compiling SOF in some unusual Zephyr configurations) and if that has not happened then just keep at it: moving more and more files gradually. It should only take a few shots.

kv2019i, 2023-11-24:
This is well under way, but I think completing the task will go beyond SOF2.8, bumping milestone.

marc-hb, 2024-01-12:
This is still high on my list but unlikely to happen for v2.9

lgirdwood, 2024-01-17:
> This is still high on my list but unlikely to happen for v2.9

We can do this incrementally, module by module if needed.

marc-hb, 2024-01-17:
> This is still high on my list but unlikely to happen for v2.9

Very bad phrasing sorry, I meant unlikely to be _completed_ for v2.9. I will hopefully have time to make some progress until v2.9.

kv2019i, 2024-09-06:
No owner, moving to v2.12

marc-hb, 2024-09-06:
https://github.com/thesofproject/sof/pull/8892 should still work

kv2019i, 2024-12-13:
Feature cutoff for v2.12, moving this to v2.13.

lgirdwood, 2024-12-13:
@kv2019i one for your planning.

kv2019i, 2025-03-13:
Resuming development for this with first PR submitted today.

kv2019i, 2025-04-02:
Status update:
- https://github.com/thesofproject/sof/pull/9918 merged
- https://github.com/thesofproject/sof/pull/9931 submitted
- https://github.com/thesofproject/sof/issues/8606 will be handled separately (this not related to Zephyr/XTOS but is now localized to how the src/audio/CMakeLists.txt is organized)

kv2019i, 2025-04-04:
With https://github.com/thesofproject/sof/pull/9950 this issue is now closed.

The platform code rules are still in the sof/zephyr/CMakeLists.txt, but these should in the end disappear as we move to native Zephyr, so let's track this effort in https://github.com/thesofproject/sof/issues/7248 

Thanks for @marc-hb for doing the baseline work and coming up with the general approach how to do this!

closed 2025-04-04 by kv2019i