thesofproject/sof #8265: [BUG] Unable to load tplg with mixer or DRC component

iuliana-prodan, 2023-09-26:
**Describe the bug**
Unable to load tplg with mixer or DRC component.
This is the dmesg log:

```
[   13.541416] platform sof-sound-wm8960: deferred probe pending
[   13.709372]     fdt_file=RODEV--MX8MP_evk--1_imx8mp-evk-sof-wm8960.dtb
[   20.411653] sof-audio-of-imx8m 3b6e8000.dsp: DT DSP detected
[   20.419352] sof-audio-of-imx8m 3b6e8000.dsp: sof_debug value: 0x1
[   20.568582] sof-audio-of-imx8m 3b6e8000.dsp: unknown sof_ext_man header type 3 size 0x30
[   20.584024] sof-audio-of-imx8m 3b6e8000.dsp: Firmware info: version 2:7:99-cbc04
[   20.599106] sof-audio-of-imx8m 3b6e8000.dsp: Firmware: ABI 3:28:0 Kernel ABI 3:23:0
[   20.621005] sof-audio-of-imx8m 3b6e8000.dsp: Firmware info: version 2:7:99-cbc04
[   20.628419] sof-audio-of-imx8m 3b6e8000.dsp: Firmware: ABI 3:28:0 Kernel ABI 3:23:0
[   20.763141] sof-audio-of-imx8m 3b6e8000.dsp: Topology: ABI 3:27:0 Kernel ABI 3:23:0
[   20.773756] asoc-simple-card sof-sound-wm8960: ASoC: Parent card not yet available, widget card binding deferred
[   20.789377] sof-audio-of-imx8m 3b6e8000.dsp: tplg: config SAI3 fmt 0x1 mclk 12288000 width 32 slots 2 mclk id 0
[   20.800647] sof-audio-of-imx8m 3b6e8000.dsp: tplg: config SAI3 fmt 0x1 mclk 12288000 width 32 slots 2 mclk id 0
[   20.817712] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx error for 0x30010000 (msg/reply size: 224/0): -22
[   20.826973] sof-audio-of-imx8m 3b6e8000.dsp: Failed to setup widget DRC1.0
[   20.834102] sof-audio-of-imx8m 3b6e8000.dsp: error: tplg component load failed -22
[   20.845282] sof-audio-of-imx8m 3b6e8000.dsp: error: failed to load DSP topology -22
[   20.855015] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at snd_soc_component_probe on 3b6e8000.dsp: -22
[   20.864608] asoc-simple-card sof-sound-wm8960: ASoC: failed to instantiate card -22
[   20.872475] asoc-simple-card: probe of sof-sound-wm8960 failed with error -22
```
or
```
[   18.757734] sof-audio-of-imx8m 3b6e8000.dsp: DT DSP detected
[   18.767291] sof-audio-of-imx8m 3b6e8000.dsp: sof_debug value: 0x1
[   18.841884] sof-audio-of-imx8m 3b6e8000.dsp: unknown sof_ext_man header type 3 size 0x30
[   18.852175] sof-audio-of-imx8m 3b6e8000.dsp: Firmware info: version 2:7:99-cbc04
[   18.860637] sof-audio-of-imx8m 3b6e8000.dsp: Firmware: ABI 3:28:0 Kernel ABI 3:23:0
[   18.871370] sof-audio-of-imx8m 3b6e8000.dsp: Firmware info: version 2:7:99-cbc04
[   18.888841] sof-audio-of-imx8m 3b6e8000.dsp: Firmware: ABI 3:28:0 Kernel ABI 3:23:0
[   18.909813] sof-audio-of-imx8m 3b6e8000.dsp: Topology: ABI 3:27:0 Kernel ABI 3:23:0
[   18.922941] asoc-simple-card sof-sound-wm8960: ASoC: Parent card not yet available, widget card binding deferred
[   18.943412] sof-audio-of-imx8m 3b6e8000.dsp: tplg: config SAI3 fmt 0x1 mclk 12288000 width 16 slots 2 mclk id 0
[   18.954296] sof-audio-of-imx8m 3b6e8000.dsp: tplg: config SAI3 fmt 0x1 mclk 12288000 width 16 slots 2 mclk id 0
[   18.969501] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx error for 0x30010000 (msg/reply size: 80/0): -22
[   18.978950] sof-audio-of-imx8m 3b6e8000.dsp: Failed to setup widget MIXER1.0
[   18.979236] sof-audio-of-imx8m 3b6e8000.dsp: error: tplg component load failed -22
[   18.979254] sof-audio-of-imx8m 3b6e8000.dsp: error: failed to load DSP topology -22
[   18.979264] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at snd_soc_component_probe on 3b6e8000.dsp: -22
[   18.979318] asoc-simple-card sof-sound-wm8960: ASoC: failed to instantiate card -22
[   18.979420] asoc-simple-card: probe of sof-sound-wm8960 failed with error -22

```

**To Reproduce**
Boot the fw with a mixer or DRC tplg file.

**Reproduction Rate**
100%

**Expected behavior**
Tplg is loaded correctly.

**Impact**
Mixer/DRC not working

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.6.0-rc1
    * SOF: sof/main - cbc04be53
2) Name of the topology file
    * Topology: For example: sof-imx8mp-drc-wm8960.tplg or sof-imx8mp-wm8960-mixer.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: imx8mp, imx8qxp, imx8qm

**Screenshots or console output**
```
root@imx8mpevk:~# ./sof-logger -l /lib/firmware/imx/sof-gcc/sof-imx8m.ldc 
 TIMESTAMP         (us)          [  671.118572] sof-audio-of-imx8m 3b6e8000.dsp: Copying cached debugfs data
    DELTA  C# COMPONENT          LOCATION                      CONTENT	ktime=671.113s  @  2023-09-26 11:10:16 +0000 UTC
[   816940871.131436] (           0.000000) c0 dma-trace             src/trace/dma-trace.c:352  INFO SHM: FW ABI 0x301c000 DBG ABI 0x5003000 tags SOF:v2.5-stable-branch-1077-ge640248551d9 src hash 0xeefd2a1b (ldc hash 0xeefd2a1b)
[     3274965.026115] (     3274965.000000) c0 dma-trace          ......./module_adapter.c:60   ERROR module_adapter_new() start
[     3275273.151102] (         308.125000) c0 dma-trace          ......./module_adapter.c:104  ERROR module_adapter_new() config->type = 5
[     3275808.307331] (         130.729156) c0 dma-trace          ......./module/generic.c:86   ERROR module_init() start
[     3292991.483732] (       17183.175781) c0 dma-trace          ......./module_adapter.c:60   ERROR module_adapter_new() start
[     3293315.181635] (         323.697906) c0 dma-trace          ......./module_adapter.c:104  ERROR module_adapter_new() config->type = 5
[     3293839.504531] (         124.374992) c0 dma-trace          ......./module/generic.c:86   ERROR module_init() start
[     3299414.348060] (        5574.843750) c0 dma-trace             src/trace/dma-trace.c:663  ERROR dtrace_add_event(): number of dropped logs = 1
[     3307495.597739] (        8081.249512) c0 dma-trace          ......./module_adapter.c:60   ERROR module_adapter_new() start
[     3307763.045645] (         267.447906) c0 dma-trace          ......./module_adapter.c:104  ERROR module_adapter_new() config->type = 5
[     3308190.337294] (         104.322914) c0 dma-trace          ......./module/generic.c:86   ERROR module_init() start
[     3320061.378489] (       11871.041016) c0 dma-trace          ......./module_adapter.c:60   ERROR module_adapter_new() start
[     3320342.784728] (         281.406250) c0 dma-trace          ......./module_adapter.c:104  ERROR module_adapter_new() config->type = 5
[     3320782.368044] (         104.374992) c0 dma-trace          ......./module/generic.c:86   ERROR module_init() start
[     3323340.180442] (        2557.812500) c0 dma-trace          ......./module_adapter.c:60   ERROR module_adapter_new() start
[     3323611.742932] (         271.562500) c0 dma-trace          ......./module_adapter.c:104  ERROR module_adapter_new() config->type = 6
[     3323798.722091] (          87.916664) c0 dma-trace          ......./module_adapter.c:142  ERROR module_adapter_new() size = 0 data = 0
[     3323892.003337] (          93.281250) c0 dma-trace          ......./module_adapter.c:159  ERROR module_adapter_new() = > goto err

[     3324039.763748] (         147.760406) c0 dma-trace          ......./module_adapter.c:206  ERROR module_adapter_new() 17
[     3324221.117907] (         181.354156) c0 dma-trace             src/ipc/ipc3/helper.c:341  ERROR comp_new(): unable to create the new component
[     3324314.711654] (          93.593750) c0 dma-trace             src/ipc/ipc3/helper.c:653  ERROR ipc_comp_new(): component cd = NULL
[     3324423.097066] (         108.385414) c0 dma-trace            src/ipc/ipc3/handler.c:1270 ERROR ipc: pipe 1 comp 0 creation failed -22
[   104061466.177469] (   100737040.000000) c0 dma-trace             src/trace/dma-trace.c:663  ERROR dtrace_add_event(): number of dropped logs = 170
Skipped 3252 bytes after the last statement.
root@imx8mpevk:~# 
```

Did a git bisect and the "bad" commit is https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496

iuliana-prodan, 2023-09-26:
I have a fix for MIXER - create a ipc_config_mixer struct as we have for volume or src or asrc:
```
diff --git a/src/audio/mixer/mixer.c b/src/audio/mixer/mixer.c
index 1d4f6cb2b..60a0b8b95 100644
--- a/src/audio/mixer/mixer.c
+++ b/src/audio/mixer/mixer.c
@@ -45,10 +45,20 @@ static int mixer_init(struct processing_module *mod)
 {
 	struct module_data *mod_data = &mod->priv;
 	struct comp_dev *dev = mod->dev;
+
+	struct module_config *cfg = &mod_data->cfg;
+	struct ipc_config_mixer *mixer = cfg->data;
+
 	struct mixer_data *md;
 
 	comp_err(dev, "mixer_init()");
 
+	if (!mixer || cfg->size != sizeof(*mixer)) {
+		comp_err(dev, "mixer_init(): No configuration data or bad data size %u",
+			 cfg->size);
+		return -EINVAL;
+	}
+
 	md = rzalloc(SOF_MEM_ZONE_RUNTIME, 0, SOF_MEM_CAPS_RAM, sizeof(*md));
 	if (!md)
 		return -ENOMEM;
diff --git a/src/audio/module_adapter/module_adapter.c b/src/audio/module_adapter/module_adapter.c
index 7d56ca538..732818a1e 100644
--- a/src/audio/module_adapter/module_adapter.c
+++ b/src/audio/module_adapter/module_adapter.c
@@ -133,6 +133,16 @@ struct comp_dev *module_adapter_new(const struct comp_driver *drv,
 		data = spec;
 		break;
 	}
+	case SOF_COMP_MIXER:
+	{
+		const struct ipc_config_mixer *ipc_mixer = spec;
+		size = sizeof(*ipc_mixer);
+		data = spec;
+		break;
+	}
 	default:
 	{
 		const struct ipc_config_process *ipc_module_adapter = spec;
diff --git a/src/include/sof/audio/ipc-config.h b/src/include/sof/audio/ipc-config.h
index 8bedb7002..8ef386cda 100644
--- a/src/include/sof/audio/ipc-config.h
+++ b/src/include/sof/audio/ipc-config.h
@@ -132,6 +132,13 @@ struct ipc_config_tone {
 	int32_t ramp_step;
 };
 
+/* generic mixer component */
+struct ipc_config_mixer {
+	uint32_t no_sources; /* number of sources supported by the module */
+	uint32_t source_bytes;
+	uint32_t sink_bytes;
+};
+
 /* generic "effect", "codec" or proprietary processing component */
 struct ipc_config_process {
 	uint32_t size;	/**< size of bespoke data section in bytes */
diff --git a/src/ipc/ipc3/helper.c b/src/ipc/ipc3/helper.c
index f7b731ffd..14874c7d3 100644
--- a/src/ipc/ipc3/helper.c
+++ b/src/ipc/ipc3/helper.c
@@ -186,6 +186,7 @@ union ipc_config_specific {
 	struct ipc_config_src src;
 	struct ipc_config_asrc asrc;
 	struct ipc_config_tone tone;
+	struct ipc_config_mixer mixer;
 	struct ipc_config_process process;
 	struct ipc_comp_file file;
 } __attribute__((packed, aligned(4)));
-- 
```
but I'm not sure if is the right one.
Also, for DRC will have to create another structure?

iuliana-prodan, 2023-09-26:
@btian1 can you please explain more why this patch was added: https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496
There were unanswered questions on the PR as well - https://github.com/thesofproject/sof/pull/7992/files#r1286322041

iuliana-prodan, 2023-09-26:
Another fix I'm trying is to have the SOF_COMP_MIXER as the others:
```
diff --git a/src/ipc/ipc3/helper.c b/src/ipc/ipc3/helper.c
index f7b731ffd..fb0d4b8db 100644
--- a/src/ipc/ipc3/helper.c
+++ b/src/ipc/ipc3/helper.c
@@ -275,6 +275,7 @@ static int comp_specific_builder(struct sof_ipc_comp *comp,
        case SOF_COMP_DCBLOCK:
        case SOF_COMP_SMART_AMP:
        case SOF_COMP_MODULE_ADAPTER:
+       case SOF_COMP_MIXER:
        case SOF_COMP_NONE:
                config->process.type = proc->type;
                config->process.size = proc->size;
@@ -283,9 +284,11 @@ static int comp_specific_builder(struct sof_ipc_comp *comp,
 #else
                config->process.data = proc->data;
 #endif
```
but I'm getting an invalid size when trying to load module - https://github.com/thesofproject/sof/blob/main/src/audio/module_adapter/module/generic.c#L50

iuliana-prodan, 2023-09-26:
@lgirdwood @singalsu @ujfalusi any ideas
Have you checked mixer with IPC3?

singalsu, 2023-09-26:
I tested DRC module conversion back then with IPC3 and it was working. I don't think new IPC structs should be added since they all follow the generic processing component IPC. Also there should be no need for IPC size adjustments like exception handling. @btian1 can the issue that was fixed by https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496 be addressed differently?

iuliana-prodan, 2023-09-26:
> I tested DRC module conversion back then with IPC3 and it was working. I don't think new IPC structs should be added since they all follow the generic processing component IPC. Also there should be no need for IPC size adjustments like exception handling. @btian1 can the issue that was fixed by [e847c8b](https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496) be addressed differently?
> 
> All IPC3 build has been removed from git main for Intel platforms so we are not able to test it ourselves any more.

Thanks @singalsu
Yes, reverting https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496 fixes the tplg load, but I wanted to know why this was added (from PR and commit message is not quite clear).

iuliana-prodan, 2023-09-26:
> > I tested DRC module conversion back then with IPC3 and it was working. I don't think new IPC structs should be added since they all follow the generic processing component IPC. Also there should be no need for IPC size adjustments like exception handling. @btian1 can the issue that was fixed by [e847c8b](https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496) be addressed differently?
> > All IPC3 build has been removed from git main for Intel platforms so we are not able to test it ourselves any more.
> 
> Thanks @singalsu Yes, reverting [e847c8b](https://github.com/thesofproject/sof/commit/e847c8b270150353551d4720e382502c9a916496) fixes the tplg load, but I want to know why this was added (from PR and commit message is not quite clear).

lgirdwood, 2023-09-29:
@iuliana-prodan can you locally revert as @btian1 is OOO next week for comment.

iuliana-prodan, 2023-10-02:
> @iuliana-prodan can you locally revert as @btian1 is OOO next week for comment.

yes, I reverted it locally and moved fwd to another issue :)
Will wait next week for @btian1 comments.

commit bb8d6baf (2023-10-13): Revert "module_adapter: avoid module init crash in case of ipc data invalid"

This reverts 'commit e847c8b27015 ("module_adapter: avoid module
init crash in case of ipc data invalid")'.

No data is not an invalid case for mixer, for example.

Fixes: #8265
closed 2023-10-13 by dbaluta