thesofproject/sof #8275: [BUG] LNL Nocodec has FW reported error: 7 with multiple pipeline playback

fredoh9, 2023-09-29:
**Describe the bug**
First LNL issue.
For LNLM_RVP_NOCODEC, multiple pipeline playback fails with FW reported error: 7 - Unsupported operation requested.

```
[ 1328.146968] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4 set pipeline instance 6 state 3
[ 1328.146983] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x13060003|0x0: GLB_SET_PIPELINE_STATE
[ 1328.147386] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[ 1328.147493] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx done : 0x13060003|0x0: GLB_SET_PIPELINE_STATE
[ 1328.147521] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4 set pipeline instance 6 state 4
[ 1328.147532] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x13060004|0x0: GLB_SET_PIPELINE_STATE
[ 1328.149150] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0: GLB_SET_PIPELINE_STATE
[ 1328.149162] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 1328.149388] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc error for msg 0x13060004|0x0
[ 1328.149420] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at soc_dai_trigger on SSP1 Pin: -22
[ 1328.149439] kernel:  Port1: ASoC: trigger FE cmd: 1 failed: -22
```

**To Reproduce**
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-lnl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline-playback.sh -l 10

**Reproduction Rate**
100%

**Environment**

- kernelBranch : topic/sof-dev
- kernelCommit : b9c99093f5ed-3
- sofBranch : main
- sofCommit : deb7a097e3cf
- TPLG: sof-mtl-nocodec.tplg, symbolic link to /lib/firmware/sof-ace-tplg/sof-lnl-nocodec.tplg

NOTE: There is no LNL topology, MTL nocodec tplg is re-used.

**Screenshots or console output**
[dmesg.txt](https://github.com/thesofproject/sof/files/12772178/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/12772179/mtrace.txt)

keqiaozhang, 2023-10-31:
This issue not only happens on LNL-NOCODEC platform, but also happens on LNL-SDW platform when testing multiple-pipeline or multiple-pause-resume.

abonislawski, 2023-10-31:
@jxstelter please take a look if this is any known issue and we will assign someone if required

softwarecki, 2023-11-15:
@fredoh9 Can you provide logs with ipc payloads?

fredoh9, 2023-11-15:
let me find how to capture logs with ipc payloads

fredoh9, 2023-11-15:
Here you are, this has IPC payloads
[dmesg-mtl-aioc.txt](https://github.com/thesofproject/sof/files/13369197/dmesg-mtl-aioc.txt)

For your information and my memory, this is with LNLM_SDW_AIOC,
cmd: TPLG=/lib/firmware/intel/sof-ace-tplg/sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg MODEL=LNLM_SDW_AIOC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50

Intel internal test link:
planresultdetail/34687?model=LNLM_SDW_AIOC&testcase=multiple-pipeline-playback-50

softwarecki, 2023-11-16:
@fredoh9 This issue will be fixed by https://github.com/thesofproject/sof/pull/8361

plbossart, 2023-11-16:
> @fredoh9 This issue will be fixed by #8361

PR #8361 only deals with SoundWire aggregation. The initial error in this issue is for NOCODEC (i.e. SSP/DMIC).
I really don't see the connection @softwarecki @fredoh9

iganakov, 2023-11-22:
> PR https://github.com/thesofproject/sof/pull/8361 only deals with SoundWire aggregation. The initial error in this issue is for NOCODEC (i.e. SSP/DMIC).

@plbossart Only log for SDW case is captured with IPC payloads so it is hard to tell if it is a common problem for SSP, DMIC and SDW. I'd rather create a separate bugs per interface and then link them if there will be a confirmation that the issue is the same.

In case of SDW PR https://github.com/thesofproject/sof/pull/8361 should help. It deals not only with aggregation but also fixes a couple of problems which I hit when tested those changes in Windows scenarios. I check the log [dmesg-mtl-aioc.txt](https://github.com/thesofproject/sof/files/13369197/dmesg-mtl-aioc.txt) and the problem looks similar to what I found previously.

The problem is related to `dai_config_dma_channel()` which is called twice - first on dai creation, and second time when running pipeline. In the second case in dma channel assignment proccess `dai->host_dma_config` doesn't have a valid pointer because 
`struct ipc4_copier_gateway_cfg` has never being copied and stored in `struct copier_data` and is valid only on Copier creation.

PR https://github.com/thesofproject/sof/pull/8361 has commit https://github.com/thesofproject/sof/pull/8361/commits/415aab5204c772897611e13619f2c6c1dd2307ef which stores gateway config (and DMA config with `dma_channel_id` needed to configure HDA DMA) in `copier_data` so `ipc4_find_dma_config()` could be safely called and `dai->host_dma_config` will be correctly assigned.
Furthermore, ALH dai which is used to configure SDW in SOF could assign only 0 channel instead of retrieving its value from DMA config.

I suppose commit https://github.com/thesofproject/sof/pull/8361/commits/415aab5204c772897611e13619f2c6c1dd2307ef should also help with SSP/DMIC, just `call ipc4_find_dma_config()` with `cd->gtw_cfg->config_data` instead. 

@softwarecki could you please prepare a debug FW with changes mentioned above and check if it helps?

plbossart, 2023-11-27:
best to move commit 415aab5 to  different PR and fast-track it if it also solves other issues.

plbossart, 2023-11-30:
“You are never served better than by yourself.”

@fredoh9 can you try PR https://github.com/thesofproject/sof/pull/8558 - only did a cherry-pick of commit  415aab5

mengdonglin, 2023-12-05:
@iganakov @softwarecki Fred has tried PR #8558 It cannot fix this issue https://github.com/thesofproject/sof/pull/8558#issuecomment-1834724262

fredoh9, 2023-12-06:
https://github.com/thesofproject/sof/issues/8562 blocking us to test further.
I can try to build with the fix for the DSP panic from Zephyr and this PR.

fredoh9, 2023-12-06:
with the fix zephyr and #8574. playback and capture work fine.
multiple pipeline testing are going on. will update the result in the https://github.com/thesofproject/sof/pull/8574

keqiaozhang, 2023-12-18:
This issue can still be reproduced with multiple-pipeline testing. Inner test ID:36046.

wszypelt, 2023-12-22:
@keqiaozhang does this still occur on main or is it different branch?

keqiaozhang, 2023-12-25:
> @keqiaozhang does this still occur on main or is it different branch?

This issue still exists on main branch, the reproduction rate is very high on LNL-SDW platform.

keqiaozhang, 2024-01-03:
Now this issue only happens on LNL-SDW platform now, No reproductions on LNL-NOCODEC for weeks.

plbossart, 2024-01-04:
@keqiaozhang I wouldn't be surprised if this is an issue with the mixer.

````
2023-12-27 22:53:24 UTC [REMOTE_INFO] Testing: Speaker [hw:0,2]
2023-12-27 22:53:24 UTC [REMOTE_COMMAND] aplay   -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/zero -q
2023-12-27 22:53:24 UTC [REMOTE_INFO] Testing: Jack Out [hw:0,0]
2023-12-27 22:53:24 UTC [REMOTE_COMMAND] aplay   -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q
2023-12-27 22:53:24 UTC [REMOTE_INFO] Testing: Deepbuffer Jack Out [hw:0,31]
2023-12-27 22:53:24 UTC [REMOTE_COMMAND] aplay   -D hw:0,31 -c 2 -r 48000 -f S16_LE /dev/zero -q
````

Can we try to put the "Deepbuffer Jack Out" in the deny list to see if independent pipelines work fine?

````
TPLG_IGNORE_LST['pcm']='HDA Digital,Deepbuffer Jack Out' 
````

should do the trick

keqiaozhang, 2024-01-05:
@plbossart , I tried to skip `Deepbuffer` pipeline, but it still failed:
```
2024-01-05 03:05:11 UTC [REMOTE_INFO] Run command to get pipeline parameters
2024-01-05 03:05:11 UTC [REMOTE_COMMAND] sof-tplgreader.py /usr/lib/firmware/intel/sof-ace-tplg/sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg -f 'type:playback & ~pcm:Amplifier Reference & ~pcm:Deepbuffer Jack Out' -b ' pcm:HDA Digital' -s 0 -e
2024-01-05 03:05:12 UTC [REMOTE_INFO] ===== Testing: (Loop: 1/50) =====
2024-01-05 03:05:12 UTC [REMOTE_INFO] /home/ubuntu/sof-test/test-case/multiple-pipeline.sh will use topology /usr/lib/firmware/intel/sof-ace-tplg/sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg to run the test case
2024-01-05 03:05:12 UTC [REMOTE_INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital' in test case
2024-01-05 03:05:12 UTC [REMOTE_INFO] Run command to get pipeline parameters
2024-01-05 03:05:12 UTC [REMOTE_COMMAND] sof-tplgreader.py /usr/lib/firmware/intel/sof-ace-tplg/sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg -f 'type:playback ' -b ' pcm:HDA Digital' -s 0 -e
2024-01-05 03:05:12 UTC [REMOTE_INFO] Testing: Speaker [hw:0,2]
2024-01-05 03:05:12 UTC [REMOTE_COMMAND] aplay   -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/zero -q
2024-01-05 03:05:12 UTC [REMOTE_INFO] Testing: Jack Out [hw:0,0]
2024-01-05 03:05:12 UTC [REMOTE_INFO] wait 0.5s for aplay_opts()
2024-01-05 03:05:12 UTC [REMOTE_COMMAND] aplay   -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q
aplay: pcm_write:2127: write error: Input/output error
2024-01-05 03:05:13 UTC [REMOTE_INFO] checking pipeline status
2024-01-05 03:05:13 UTC [REMOTE_ERROR] Running process count is 1, but 2 is expected
4143 aplay -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q
2024-01-05 03:05:13 UTC [REMOTE_INFO] Starting func_exit_handler(1)
2024-01-05 03:05:13 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2024-01-05 03:05:13 UTC [REMOTE_ERROR]  func_error_exit()  @  /home/ubuntu/sof-test/test-case/multiple-pipeline.sh
2024-01-05 03:05:13 UTC [REMOTE_ERROR]  ps_checks()  @  /home/ubuntu/sof-test/test-case/multiple-pipeline.sh:147
2024-01-05 03:05:13 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/multiple-pipeline.sh:194
/home/ubuntu/sof-test/test-case/../case-lib/lib.sh: line 723:  4143 Killed                  aplay $SOF_ALSA_OPTS $SOF_APLAY_OPTS "$@"
2024-01-05 03:05:13 UTC [REMOTE_INFO] pkill -TERM -f mtrace-reader.py
2024-01-05 03:05:13 UTC [REMOTE_INFO] nlines=177 /home/ubuntu/sof-test/logs/multiple-pipeline/2024-01-05-03:05:11-15024/mtrace.txt
2024-01-05 03:05:13 UTC [REMOTE_INFO] ktime=2994 sof-test PID=3996: ending
2024-01-05 03:05:13 UTC [REMOTE_INFO] Test Result: FAIL!

```

This issue can be easily triggered by:
```
aplay  -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/zero -q -d 10  & aplay -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q -d 10
[1] 4396
aplay: pcm_write:2127: write error: Input/output error
```

plbossart, 2024-01-05:
Thanks @keqiaozhang that's very useful, One last suggestion would be to try with a passthrough topology with simplified pipelines to see if this is the DSP IOs getting corrupted somehow, or really the DSP graph introducing issues.

bardliao, 2024-01-25:
@plbossart @keqiaozhang I tested with https://github.com/thesofproject/sof/pull/8361 and https://github.com/thesofproject/linux/pull/4790 on LNL SDW. And the issue is not reproducible by `aplay  -D hw:0,2 -c 2 -r 48000 -f S16_LE /dev/zero -q -d 10  & aplay -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q -d 10` or `TPLG=/lib/firmware/intel/sof-ipc4-tplg/sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg MODEL=LNLM_SDW_AIOC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50`

wszypelt, 2024-01-26:
root cause: https://github.com/thesofproject/sof/pull/8361 
waiting for merge

wszypelt, 2024-01-30:
Based on the report from Keqiao
the issue does not reproduce