thesofproject/sof #8276: [BUG] LNL Platforms need topologies, nocodec and soundwires

fredoh9, 2023-09-29:
**Describe the bug**
LNL test platforms are getting ready for development. Need SDW and NOCODEC topologies.

For HDA, sof-hda-generic-4ch.tplg was tested with basic playback and capture.
For NOCODEC, sof-mtl-nocodec.tplg was re-used for the test, basic playback and capture are working fine but https://github.com/thesofproject/sof/issues/8275 found
For SDW, sof-mtl-rt711-4ch.tplg was re-used for the test, looks fine

lgirdwood, 2023-10-02:
@jsarha can you convert the above MTL topologies to LNL and upstream? Thanks

jsarha, 2023-10-03:
@ranj063 , so should I simply mimic all nocodec and rt711 mtl topology build rules with mtl replaced with lnl?

What about sof-hda-generic* topologies? There is no mtl on the resulting topology name, and the one created with PLATFORM=mtl is reported to work. Should I still make sof-lnl-hda-generic* variants the same way suggested above?

lgirdwood, 2023-10-04:
> @ranj063 , so should I simply mimic all nocodec and rt711 mtl topology build rules with mtl replaced with lnl?
>
> What about sof-hda-generic* topologies? There is no mtl on the resulting topology [...]

I think so, @fredoh9 will this work for you ?

fredoh9, 2023-10-05:
For hda, I think, 'generic' works for all platforms.

But for ssp and soundwire, I'm not sure lnl is simply same as mtl.

lgirdwood, 2023-10-06:
> For hda, I think, 'generic' works for all platforms.
> 
> But for ssp and soundwire, I'm not sure lnl is simply same as mtl.

@plbossart I think we can resue MTL SSP topology file with LNL (probably without modification ?), but what about resuing MTL SDW topologies for LNL, would you want new topology files for LNL (that can be easily modified and differentiated).

plbossart, 2023-10-06:
I don't think there's any difference for SSP and SoundWire topologies between MTL and LNL, the change with HDaudio DMA is hidden and something the topology writer needs to be aware of.

We can probably use the same MTL RVP topologies for LNL RVP topologies, for real products the configurations will depend on the endless creativity of our hardware partners which keep flipping hardware connections and devices. So most likely we'll have different combinations to handle.

Long term I still think we are going into a wall. The definition of what component is where should be an input to the topology, not hard-coded in the the topology. IOW, the machine driver should update the topology defaults to account for variants and some paths being disabled.

lgirdwood, 2023-10-06:
@fredoh9 it sounds like you can create a LNL topology based on MTL and reuse SSP and SDW and then modify any HDA settings as required.

abonislawski, 2023-10-10:
@jxstelter FYI, maybe you can help with LNL changes

lgirdwood, 2023-10-20:
@fredoh9 is this now done and can be closed ?

keqiaozhang, 2023-10-23:
https://github.com/thesofproject/sof/pull/8344 has been merged. Closing this bug.

closed 2023-10-23 by keqiaozhang