thesofproject/sof #8281: [Lenovo] Update with Meteorlake support not working on (proto system)

mrhpearson, 2023-10-02:
**Describe the bug**
We've got our hands on early MTL HW and are working on bringing up Linux.
Audio wasn't working. Now with the newer sof-bin release adding support available, we tried that - but get the following error message.

I'm limited in what information I can share on this platform on a public forum (as it's not released yet), so if needed please ping me by email (mpearson-lenovo at squebb . ca is best, but markpearson,lenovo,com works if a Lenovo email is preferred).

[    4.857331] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    4.857798] sof-audio-pci-intel-mtl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.858250] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   66.901114] sof-audio-pci-intel-mtl 0000:00:1f.3: couldn't bind with audio component
[   66.901196] sof-audio-pci-intel-mtl 0000:00:1f.3: init of i915 and HDMI codec failed
[   66.903098] sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
[   66.939494] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 1
[   66.939498] sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   66.939502] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   66.945277] sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.7.0.1
[   69.948767] sof-audio-pci-intel-mtl 0000:00:1f.3: hda_cl_copy_fw: timeout with rom_status_reg (0x180000) read
[   69.948779] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   69.948781] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware download failed
[   69.948783] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[   69.948790] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[   69.948791] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000000c, ROM debug error: 0x97
[   69.948798] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[   69.948799] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   69.948811] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to start DSP
[   69.948812] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to boot DSP firmware -110
[   69.953196] sof-audio-pci-intel-mtl 0000:00:1f.3: error: sof_probe_work failed err: -110

**To Reproduce**
Boot System

**Reproduction Rate**
100%

**Expected behavior**
working audio :)

**Impact**
showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: fedora 38 6.5.5
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: < :) >

abonislawski, 2023-10-03:
> Now with the newer sof-bin release adding support available

This build is prepared for production platforms.

lgirdwood, 2023-10-03:
@mrhpearson thanks, let me check - its sounds like the Fedora kernel could be missing a patch if FW loading is failing. There also seems to be an issue with HDMI too.

kv2019i, 2023-10-03:
@mrhpearson So indeed, this looks like you'd need a differently signed binary for this proto system. For display codec, with current vanilla upstream kernel, you need still need to use the "force_probe" i915 module option. With that display audio should work as well.

mengdonglin, 2023-10-22:
@mrhpearson @lgirdwood @kv2019i @marc-hb 
'[ 69.948791] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000000c, ROM debug error: 0x97'

We observed the same issue on another proto device. And @tmleman confirmed this is key verification error.
```
Debug status: 0x0C -> FSR_ROM_VALIDATE_PUB_KEY
Debug error: 0x97 -> ADSP_MANIFEST_KEY_NOT_REGISTERED // Manifest key does not match any registered.

Error is reported by the rom_ext during public kay verification.
```

lgirdwood, 2023-10-23:
This is now a v2.7.1 blocker issue.

abonislawski, 2023-10-27:
Fixed now on stable-v2.7 and v2.7.1 tag

marc-hb, 2023-10-27:
> Fixed now on stable-v2.7 and v2.7.1 tag

More precisely b028b2d5c361a8

marc-hb, 2023-10-30:
> More precisely https://github.com/thesofproject/sof/commit/b028b2d5c361a8cbd6611f1fee6a64d0b5e1d360

> Nothing on the main branch yet.

Now cherry-picked to the main branch:
- #8421

lgirdwood, 2023-10-31:
> > More precisely [b028b2d](https://github.com/thesofproject/sof/commit/b028b2d5c361a8cbd6611f1fee6a64d0b5e1d360)
> 
> > Nothing on the main branch yet.
> 
> Now cherry-picked to the main branch:
> 
> * [rimage: mtl/lnl: fix key slot setup based on imr type #8421](https://github.com/thesofproject/sof/pull/8421)

Now merged in main. @mrhpearson v2.7.2 coming soon.

kv2019i, 2023-10-31:
@lgirdwood wrote:
> Now merged in main. @mrhpearson v2.7.2 coming soon.

Quick correction, @mrhpearson the fix is already in the tagged v2.7.1. Only missing thing is the signed binary and I'll upload a PR to sof-bin as soon as all process steps are done (community signed binaries already uploaded -> https://github.com/thesofproject/sof-bin/pull/151 ).

If there's any hickup with 2.7.1, we'll do a 2.7.2 if needed before we tag sof-bin-2023.09.1 and release that to distros.

keqiaozhang, 2023-11-06:
The fix https://github.com/thesofproject/sof/pull/8421 merged in [stable-v2.7.](https://github.com/thesofproject/sof/tree/stable-v2.7)

keqiaozhang, 2023-11-06:
@mrhpearson please check with https://github.com/thesofproject/sof-bin/releases/tag/v2023.09.1

closed 2023-11-06 by keqiaozhang

mrhpearson, 2023-11-06:
Thanks - we're testing.