thesofproject/sof #8294: [BUG] Very weak sound on XPS 15 model 9530 (2023)

berbmit, 2023-10-06:
**Describe the bug**
XPS 15 laptop, model 9530 (2023) with latest Linux Mint installed and all updates applied, the sound is soft and tinny.  It seems the woofers are not being detected and with only the 2 top tweeters firing.  Have tried suggestions I can find on the web, tried kernels up to 6.5, made sure all updates from repositories are applied.

**Reproduction Rate**
100% Have googled extensively, and seems this problem is prevalent on the XPS 15 9530.

**Expected behavior**
The sound works well under Windows (from testing before deleting windows and installing Linux)  

**Impact**
This is a huge frustration for music and games and Zoom calls for work.

**Environment**
I do not know how to get all the template's information
    * Kernel: tried 6.2 to 6.5
    * SOF: 
    * Topology: 
    * Platform: Dell XPS 15 9530 with latest Linux Mint and all updates applied.
From (https://nanoreview.net/en/laptop/dell-xps-15-9530-2023) it seems the laptop's sound specs are:
Audio chip Realtek ALC3281
Speakers 2.2 (2 tweeters and 2 woofers)
Power 2x2W, 2x3W

From dmesg:
[    4.817359] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.817376] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.817382] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.817579] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.817633] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.825894] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    4.852090] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    4.852093] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.852095] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.854782] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.854784] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    4.854786] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    4.949656] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.949663] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    4.957446] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:20:0 Kernel ABI 3:23:0
[    5.706027] input: sof-hda-dsp Headphone Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    5.706071] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    5.706100] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    5.706130] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20

lgirdwood, 2023-10-06:
@berbmit ok, thanks for the dmesg and closing the duplicate bug.  @bardliao who would be able to help from Realtek, I assume we are missing HDA verbs to enable the amplifiers ?

angioli, 2023-10-08:
I'm exactly in the same situation;  I tried in ubuntu 23.04 and manjaro linux

System:
  Kernel: 6.5.5-1-MANJARO arch: x86_64 bits: 64 compiler: gcc v: 13.2.1
    clocksource: tsc available: acpi_pm
    parameters: BOOT_IMAGE=/boot/vmlinuz-6.5-x86_64
    root=UUID=7b963b00-a2a6-4e87-b512-1ffc311f1c5c rw
    btusb.enable_autosuspend=0 quiet splash
  Desktop: GNOME v: 44.5 tk: GTK v: 3.24.38 wm: gnome-shell dm: GDM v: 44.1
    Distro: Manjaro Linux base: Arch Linux
Machine:
  Type: Laptop System: Dell product: XPS 15 9530 v: N/A
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: Dell model: 0P6GXD v: A00 serial: <superuser required> UEFI: Dell
    v: 1.7.0 date: 08/14/2023

Ruddle, 2023-10-09:
Same here.
The volume is less than a quarter what it is on windows, and there is a frequency cut off at 150hz, instead of 50-60hz.
Let me recap all the info I found on this matter.

I tried the patch from [the previous xps 9520](https://github.com/rikvanderkemp/dell-xps-15-9520-audio) with a 6.5.4 kernel, no luck. It consists in adding this line to `/sound/pci/hda/patch_realtek.c` in the kernel:
```
SND_PCI_QUIRK(0x1028, 0x0beb, "Dell XPS 15 9530", ALC289_FIXUP_DUAL_SPK),
```
(I adapted the SDevice to the xps 9530's one)
Didn't work.

I also tried `hdajackretask` with nearly all possible combinations of enabling disconnected pins. Didn't work.

There are some efforts shared [on reddit here](https://www.reddit.com/r/XPS/comments/1562boo/does_anyone_have_a_patch_for_ubuntu_2304_sound/), and [here](https://www.reddit.com/r/XPS/comments/13p3vp4/comment/jsjf73p/?context=3). But no solutions yet.

Apparently, the xps 15 9530 share some audio hardware with recent Asus laptops, [where the problem was solved](https://asus-linux.org/wiki/cirrus-amps/)

Unfortunately the work done is above my current skills.

berbmit, 2023-10-09:
> Apparently, the xps 15 9530 share some audio hardware with recent Asus laptops, [where the problem was solved](https://asus-linux.org/wiki/cirrus-amps/)
> 
> Unfortunately the work done is above my current skills.

Above mine too ... but thanks for the link.   I was amused by the ASUS link ending with "The goal of this article was to provide an **easily accessible** guide for people to work from".   That's above my pay grade but hopefully this will help others more competent to move this forward to something I can implement.

plbossart, 2023-10-09:
Try this:
https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#disable-sof-on-pci-hdaudio-devices-to-test-audio-playback

If the volume remains low, it's a codec configuration issue and it needs to be worked-around using the other suggestions - it's clearly not simple to reverse-engineer this flow.