thesofproject/sof #8313: [FEATURE]Does “sof host” have any plans to support zephyr?

hongshui3000, 2023-10-12:
Currently, sof hosts are all executed under Linux, which is not friendly to some MCUs, such as the NXP RT600 cross-border processor. They do not have CPUs that can run Linux.

lgirdwood, 2023-10-16:
@hongshui3000 you should be able to run SOF using Zephyr on any supported Zephyr target. You may still need to add drivers for your target DMA and audio endpoints thought.

hongshui3000, 2023-10-17:
I mean running the SOF HOST driver, similar to that in Linux
https://github.com/thesofproject/linux/tree/topic/sof-dev/sound/soc/sof

I saw the support of sof dsp on zephyr. Host generally only seems to be supported by linux. I also want to use zephyr, but I may have to port it myself.

dbaluta, 2023-10-17:
@hongshui3000 this is a topic that is in our area of interest. 

How do you use SOF with MCU systems? (which usually have a bunch of small arm cores (m4, m7) and maybe a separate dsp).

I'm still ramping up in this area but I think traditionally there is no concept of host.

So, we need to do is to make SOF able to run as a standalone firmware one the HIFI4 DSP. The trickies part here would be how do you actually load the topology and get things started.

hongshui3000, 2023-10-17:
They used to use it but seem to have given up on it. It would be great if someone would like to improve it or make it officially supported  https://cloud.listenai.com/zephyr/sof_host

dbaluta, 2023-10-17:
> They used to use it but seem to have given up on it. It would be great if someone would like to improve it or make it officially supported https://cloud.listenai.com/zephyr/sof_host

Who are they? Also, what particular application do you target?

hongshui3000, 2023-10-17:
I'm enabling the cortex m to run "sof host" directly

dbaluta, 2023-10-17:
@hongshui3000 OK, I think then you just need to compile SOF for M core. We enabled some SOF samples for ARM64, but I guess you need arm32.

Then you need to create some sort of an "emulation" of commands received from host (e.g create a list of commands that otherwise would have been sent from Host).

MCU platforms usually load just 1 program with fixed functionality (like just play a song, etc). 

I hope this makes some sense.

hongshui3000, 2023-10-17:
> @hongshui3000 OK, I think then you just need to compile SOF for M core. We enabled some SOF samples for ARM64, but I guess you need arm32.
>
> Then you need to create some sort of an "emulation" of [...]

Yes, it should be smaller than the current Linux sof host (code and data, etc.)

lgirdwood, 2023-10-17:
> I mean running the SOF HOST driver, similar to that in Linux https://github.com/thesofproject/linux/tree/topic/sof-dev/sound/soc/sof
> 
> I saw the support of sof dsp on zephyr. Host generally only seems to be supported by linux. I also want to use zephyr, but I may have to port it myself.

Yes, it can run on host, @andyross did the work and its used for IPC fuzzing today (not for audio). I think there are instructions in sof-docs or in Andy's commit message (search git log).

hongshui3000, 2023-10-18:
> Yes, it can run on host, @andyross did the work and its used for IPC fuzzing today (not for audio). I think there are instructions in sof-docs or in Andy's commit message (search git log).

What you are talking about should be the host running on the POSIX system. In fact, I mean the host running on the cortex M (mcu). The main difference is that the architecture of MCU is different from that of PC（posix generally runs on PC，which may be x86）, and MCU has fewer available resources.

dbaluta, 2023-10-18:
@lgirdwood I think maybe we are talking about different things. 

@hongshui3000 would like to run the SOF firmware on a microcontroller which doesn't have support for an application processor (that in our case runs linux). 

So we need to find a way to run the SOF firmware as a standalone code (on M4 or HIFI4) but without Host driver / ALSA interface.

It looks like @andyross allowed the other way around. That is allowed SOF firmware to be ran inside without an actual "physical" core. That is maybe as a process inside Linux userspace.

lgirdwood, 2023-10-18:
@dbaluta @hongshui3000 got you now. This sould like you need Zephyr support as a minimum on your MCU and you will need to hard code your pipelines.

dbaluta, 2023-10-19:
@lgirdwood exactly.

@hongshui3000 would you like to work on that? We can support you. 

So far we have SOF + Zephyr working on Xtensa and arm64 so it should be easy to get you started.

hongshui3000, 2023-10-20:
> @lgirdwood exactly.
> 
> @hongshui3000 would you like to work on that? We can support you.
> 
> So far we have SOF + Zephyr working on Xtensa and arm64 so it should be easy to get you started.

Yes, but the task may not be completed too quickly

hongshui3000, 2023-10-20:
Maybe I should submit a PR for the external module (host) to zephyr,

lgirdwood, 2023-10-20:
> Maybe I should submit a PR for the external module (host) to zephyr,

Yep, that will start the discussion and allow alignment with Zephyr team.