thesofproject/sof #8320: [BUG][SDW]  failed to prepare widget host-copier.3.capture on IPC4 SDW platforms

keqiaozhang, 2023-10-13:
**Describe the bug**
This is a topology regression and happens on IPC4-SDW platforms. The repro rate is 100%.
arecord doesn't work and the error is:
```
[  237.208709] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: sof_ipc4_init_input_audio_fmt: Unsupported audio format: 48000Hz, 32bit, 4 channels
[  237.208718] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to prepare widget host-copier.3.capture
[  237.208722] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to prepare connected widgets
[  237.208725] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 3 dir 1
[  237.208728] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
```
This issue is introduced by https://github.com/thesofproject/sof/commit/7a11e27bf2a19e9acf643d416fc079f479fd2748

**To Reproduce**
arecord   -Dhw:0,3 -r 48000 -c 4 -f S16_LE -d 3 /dev/null -v -q

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: tip sof-dev
    * SOF: [{SHA}](https://github.com/thesofproject/sof/commit/7a11e27bf2a19e9acf643d416fc079f479fd2748)
2) Name of the topology file
    * Topology: 
    sof-mtl-rt711-l0-rt1316-l23-rt714-l1.tplg
    sof-mtl-rt1318-l12-rt714-l0.tplg
    sof-adl-rt711-l0-rt1316-l12-rt714-l3.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform:IPC4 SDW platforms.

[dmesg.txt](https://github.com/thesofproject/sof/files/12889837/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/12889840/mtrace.txt)

ranj063, 2023-10-13:
@bardliao @keqiaozhang do you have access to one of these devices? Can you please send me the dmesg after boot? I'd like to see what the available formats for host-copier.3.capture are.

fredoh9, 2023-10-13:
@ranj063 yes https://github.com/thesofproject/sof/pull/8323 works fine

fredoh9, 2023-10-18:
Re-run the test after https://github.com/thesofproject/sof/pull/8323 merged. This issue is gone

lgirdwood, 2023-10-20:
> Re-run the test after #8323 merged. This issue is gone

Thanks @fredoh9 - please close after a few more days if no reproduction.

keqiaozhang, 2023-10-23:
No reproductions in CI, closing this bug.

closed 2023-10-23 by keqiaozhang