thesofproject/sof #8336: Writing the sections to .ri file in reversal order in rimage creation process.

saisurya-ch, 2023-10-17:
Hi,

I have observed that the order of writing the sections to .ri got reversed from the [commit ID](https://github.com/thesofproject/rimage/commit/539c2b388cc54ef6e7427ea55c6a77d7636bf99b). 

cause:
At the end of the SOF build process, while finding the sections and listing the valid sections, the address of the last section is stored as the first section. Because of this, the last section is taken first when writing into the output .ri file. This reversal of sections in rimage is causing the SOF loading issues on the AMD platforms.

could you please resolve this issue.

lgirdwood, 2023-10-18:
@saisurya-ch thanks, not sure why this has not hit us on other platforms yet, it could just be the build config and output on your platform triggers it and means a boot failure. Sounds like it has potential to impact all platforms though.
@softwarecki it looks like the new parser is getting the section address wrong for the last ELF section and writing them in reverse order. Can you fix @mwasko fyi

softwarecki, 2023-10-23:
@saisurya-ch Can you provide some more information about this issue which will allow me to reproduce it? What platform is affected? Can you share the input files for rimage? Can you paste output from rimage?

lgirdwood, 2023-10-23:
@marc-hb fyi

marc-hb, 2023-10-23:
> @saisurya-ch Can you provide some more information about this issue which will allow me to reproduce it?

@saisurya-ch do not discard the entire bug report template
https://github.com/thesofproject/sof/issues/new?assignees=&labels=bug&projects=&template=bug_report.md&title=%5BBUG%5D
Specific reproduction steps and sample data are required.

saisurya-ch, 2023-10-26:
Bug:
I have observed that the order of writing the sections to .ri got reversed from the rimage [commit ID](https://github.com/thesofproject/rimage/commit/539c2b388cc54ef6e7427ea55c6a77d7636bf99b).

To Reproduce:
Clone the latest SOF recursively. After adding the GCC toolchain path to the PATH variable, try building for the AMD/rembrandt platform using the command "./scripts/xtensa-build-all.sh rmb". (rmb platform is taken as an example here.)

Reproduction Rate:
The issue gets reproduced every time when the latest SOF is built.

Expected behavior:
The order of the elf sections should not be changed while writing them to the .ri file. They need to be written to the output ri file in the same order they were read.

Impact:
This causes improper SOF loading on all AMD platforms. (I haven't tried on other platforms.)

Environment:
1) Main branch of SOF repo.
2) Platform(s) on which the bug is observed.
    * Platform: rmb, rn, vangogh, acp_6_3 (actually impacts all platforms.)

Screenshots or console output:

[image]
[image]

observation:
https://github.com/thesofproject/sof/blob/a951579bb27e7f8556824ffe4e88705ccee0d024/tools/rimage/src/module.c#L362TThis line is for holding the first section address. But while reading the sections in the loop, at the end of the loop it is actually holding the last section address. So, while writing into the .ri file, it is writing the last section first and so on in the reverse order.

PR #8401 (2023-10-31): rimage: module: Fix section order in output image
Fixed the order in which sections are placed in the output firmware image for platforms using a simple manifest.

Fixes: #8336
commit 3cb13e7c (2023-10-31): rimage: module: Fix section order in output image

Fixed the order in which sections are placed in the output firmware image
for platforms using a simple manifest.

Fixes: #8336
closed 2023-10-31 by kv2019i