thesofproject/sof #8343: [BUG] Topology2: need to set SAMPLE_TYPE_MSB_INTEGER for dai copier

bardliao, 2023-10-18:
**Describe the bug**
Now Firmware adjusts valid format in copier according to sample type.
We need to set SAMPLE_TYPE_MSB_INTEGER for all 24 bit format dai copiers.
https://github.com/thesofproject/sof/pull/8151 did the job for the existing topologies.
However, we need to remember to do the same change when we add new 24 bit dai copier support.
**The issue is just a reminder.**
Playback output will be no sound if we don't do the change.

**To Reproduce**
Test 24 bit format with SAMPLE_TYPE_LSB_INTEGER out_sample_type

**Reproduction Rate**
100%

**Expected behavior**
Playback output is audible.

**Impact**
It is not acceptable if playback output no sound.

**Environment**
1)  * SOF: After v2.7
2) Name of the topology file
    * Topology: Without the out_sample_type change
3) Name of the platform(s) on which the bug is observed.
    * Platform: all

lgirdwood, 2023-10-18:
@bardliao I think @RanderWang did some checking of formats a few months ago. Best to have a quick check with him too.

bardliao, 2023-11-16:
Closing it as https://github.com/thesofproject/sof/pull/8359 has been merged.

closed 2023-11-16 by bardliao