thesofproject/sof #8345: [BUG] DSP panic on LNL-SDW platform

keqiaozhang, 2023-10-18:
**Describe the bug**
Playback failed on LNL-SDW platform with below error:
```
[  523.648750] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x40010004|0x1005d: MOD_INIT_INSTANCE [data size: 372]
[  523.651502] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[  523.651510] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  523.651515] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP panic!
[  523.651518] sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  523.651529] sof-audio-pci-intel-lnl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[  523.651531] sof-audio-pci-intel-lnl 0000:00:1f.3: ROM debug status: 0x0, ROM debug error: 0x0
[  523.651534] sof-audio-pci-intel-lnl 0000:00:1f.3: ROM feature bit enabled
[  523.651550] snd_sof:sof_ipc4_find_debug_slot_offset_by_type: sof-audio-pci-intel-lnl 0000:00:1f.3: Slot type 0x4c455400 is not available in debug window
[  523.651552] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  523.651555] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 7 -> 8
[  523.651581] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[  524.151619] sof-audio-pci-intel-lnl 0000:00:1f.3: ipc timed out for 0x40010004|0x1005d
[  524.151630] sof-audio-pci-intel-lnl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[  524.151633] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  524.151677] sof-audio-pci-intel-lnl 0000:00:1f.3: Host IPC initiator: 0xc0010004|0x1005d|0x0, target: 0x1b0a0000|0x0|0x0, ctl: 0x3
[  524.151681] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  524.151683] sof-audio-pci-intel-lnl 0000:00:1f.3: IPC timeout
[  524.151703] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to create module alh-copier.SDW0-Playback.0
[  524.151768] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget gain.1.1 freed
[  524.151773] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x12010000|0x0: GLB_DELETE_PIPELINE
[  524.151797] sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12010000|0x0 failed: -19
[  524.151801] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to free pipeline widget pipeline.1
[  524.151806] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget mixin.0.1 freed
[  524.151809] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget gain.0.1 freed
[  524.151811] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x12000000|0x0: GLB_DELETE_PIPELINE
[  524.151813] sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12000000|0x0 failed: -19
[  524.151816] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to free pipeline widget pipeline.0
[  524.151819] sof-audio-pci-intel-lnl 0000:00:1f.3: Failed to set up connected widgets
[  524.151824] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 0
[  524.151828] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[  524.151846]  Jack Out: ASoC: error at __soc_pcm_hw_params on Jack Out: -110
[  524.151854] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[  524.151873]  Jack Out: ASoC: error at dpcm_fe_dai_hw_params on Jack Out: -110
[  524.151880] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: free stream 0 dir 0
[  524.151920] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 6
[  524.152123] snd_sof:sof_pcm_close: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: close stream 0 dir 0
```

**To Reproduce**
speaker-test -Dhw:0,0 -t wav -c 2 

**Reproduction Rate**
100%.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {tip sof-dev}
    * SOF: {tip sof-main}
2) Name of the topology file
    * Topology: {sof-lnl-rt711-4ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL}

plbossart, 2023-10-18:
@keqiaozhang sof-main is missing a patch, using the upstream branch does not work (yet)

fredoh9, 2023-10-18:
I'm able to reproduce the problem, 100%.
It wasn't the case when I tested LNL Suondwire in a month ago. I checked my note. It was working fine with this hashes.

- Kernel : 0912756595d5 (6.5-rc6)
- SOF FW/TPLG : 0912756595d5

plbossart, 2023-10-19:
Things work fine for me with SOF FW/TPLG 054003f2926b835d7e7ca2db8e602a7f2f9faeab with this patch added for zephyr

[0001-ace-alh-Only-ACE1.5-has-OSEL-feature.patch.txt](https://github.com/thesofproject/sof/files/13035783/0001-ace-alh-Only-ACE1.5-has-OSEL-feature.patch.txt)

kv2019i, 2023-10-20:
@jxstelter This is the missing "ace: alh: Only ACE1.5 has OSEL feature" patch

keqiaozhang, 2023-10-30:
Fixed by https://github.com/thesofproject/sof/pull/8386. Closing this bug.

closed 2023-10-30 by keqiaozhang