thesofproject/sof #8356: [BUG] Way too many DT/Kconfig build warnings

plbossart, 2023-10-19:
**Describe the bug**
 lots of warning messages for Device tree and Kconfig. Am I the only one that looks at build logs?

**To Reproduce**
./sof/scripts/xtensa-build-zephyr.py --use-platform-subdir --key-type-subdir dbgkey -k ~/mtl_debug_private_key.pem -d lnl

**Reproduction Rate**
100%

**Expected behavior**
no warnings

**Impact**
Annoyance

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * zephyr: 3563347b108a8fe323595060576572fccdb81be6
    * SOF: 054003f2926b835d7e7ca2db8e602a7f2f9faeab

2) Name of the platform(s) on which the bug is observed.
    * Platform: lnl

**Screenshots or console output**

````
-- Including generated dts.cmake file: /data/pbossart/SOF-zephyr-main/build-lnl/zephyr/dts.cmake
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:382.16-385.5: Warning (simple_bus_reg): /soc/timer: missing or empty reg/ranges property
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:439.25-444.4: Warning (unique_unit_address_if_enabled): /memory@a0020000: duplicate unit-address (also used in node /virtualmemory@a0020000)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:31.26-38.5: Warning (unique_unit_address_if_enabled): /soc/dai-dmic0@10100: duplicate unit-address (also used in node /soc/dai-dmic1@10100)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh1@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh2@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh2@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh3@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh3@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:71.20-75.5: Warning (unique_unit_address_if_enabled): /soc/alh2@24400: duplicate unit-address (also used in node /soc/alh3@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh4@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh4@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:71.20-75.5: Warning (unique_unit_address_if_enabled): /soc/alh2@24400: duplicate unit-address (also used in node /soc/alh4@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:76.20-80.5: Warning (unique_unit_address_if_enabled): /soc/alh3@24400: duplicate unit-address (also used in node /soc/alh4@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh5@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh5@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:71.20-75.5: Warning (unique_unit_address_if_enabled): /soc/alh2@24400: duplicate unit-address (also used in node /soc/alh5@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:76.20-80.5: Warning (unique_unit_address_if_enabled): /soc/alh3@24400: duplicate unit-address (also used in node /soc/alh5@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:81.20-85.5: Warning (unique_unit_address_if_enabled): /soc/alh4@24400: duplicate unit-address (also used in node /soc/alh5@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh6@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh6@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:71.20-75.5: Warning (unique_unit_address_if_enabled): /soc/alh2@24400: duplicate unit-address (also used in node /soc/alh6@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:76.20-80.5: Warning (unique_unit_address_if_enabled): /soc/alh3@24400: duplicate unit-address (also used in node /soc/alh6@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:81.20-85.5: Warning (unique_unit_address_if_enabled): /soc/alh4@24400: duplicate unit-address (also used in node /soc/alh6@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:86.20-90.5: Warning (unique_unit_address_if_enabled): /soc/alh5@24400: duplicate unit-address (also used in node /soc/alh6@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:71.20-75.5: Warning (unique_unit_address_if_enabled): /soc/alh2@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:76.20-80.5: Warning (unique_unit_address_if_enabled): /soc/alh3@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:81.20-85.5: Warning (unique_unit_address_if_enabled): /soc/alh4@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:86.20-90.5: Warning (unique_unit_address_if_enabled): /soc/alh5@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:91.20-95.5: Warning (unique_unit_address_if_enabled): /soc/alh6@24400: duplicate unit-address (also used in node /soc/alh7@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:66.20-70.5: Warning (unique_unit_address_if_enabled): /soc/alh1@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:71.20-75.5: Warning (unique_unit_address_if_enabled): /soc/alh2@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:76.20-80.5: Warning (unique_unit_address_if_enabled): /soc/alh3@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:81.20-85.5: Warning (unique_unit_address_if_enabled): /soc/alh4@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:86.20-90.5: Warning (unique_unit_address_if_enabled): /soc/alh5@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:91.20-95.5: Warning (unique_unit_address_if_enabled): /soc/alh6@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:96.20-100.5: Warning (unique_unit_address_if_enabled): /soc/alh7@24400: duplicate unit-address (also used in node /soc/alh8@24400)
/data/pbossart/SOF-zephyr-main/build-lnl/zephyr/zephyr.dts:61.20-65.5: Warning (unique_unit_address_if_enabled): /soc/alh0@24400: duplicate unit-address (also used in node /soc/alh9@24400)
[... cut ...]
got the value 'n'. Check these unsatisfied dependencies: ((DMA_INTEL_ADSP_GPDMA && DMA_DW && DMA) ||
(DMA_INTEL_ADSP_GPDMA && DMA_INTEL_ADSP_GPDMA && DMA) || (DMA_INTEL_ADSP_GPDMA && DMA_INTEL_LPSS &&
DMA)) (=n). See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_DMA_DW_SUSPEND_DRAIN and/or
look up DMA_DW_SUSPEND_DRAIN in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.
````

plbossart, 2023-10-19:
@kv2019i @marc-hb @jxstelter FYI

marc-hb, 2023-10-19:
Reproduced in today's daily build (and probably many others) https://github.com/thesofproject/sof/actions/runs/6568221442/job/17842304700

I don't know how to make the DT warnings fatal. We always had some and not just in LNL. @nashif any idea?

I'm more surprised about the Kconfig warning slipping through. I think Zephyr CI catches these, someone would have to go and find how it promotes them to errors.

> Am I the only one that looks at build logs?

No, but you maybe the only one building LNL _and_ looking at build logs.

kv2019i, 2023-10-20:
@plbossart Guilty as charged (w.r.t. anyone looking at build output).

Fix for the Kconfig warning pushed as https://github.com/thesofproject/sof/pull/8362

The DT warnings are a more complicated thing. This is a known feature/side-effect of the approach we use to describe the ALH/SDW nodes in our DT files. We use the same register for multiple nodes (alh0, alh1) and this triggers a warning. This has been discussed multiple times, but so far no solution. FYI to at least @serhiy-katsyuba-intel who has been in the loop in the discussion.

marc-hb, 2023-11-13:
Thanks @kv2019i

> This has been discussed multiple times

Any references or just lost in email?

Short of fixing it, is there a way to make it less verbose so it does drown other, unrelated issues in the build log? Any idea why it's seems so repetitive?

Due to the volume I don't think it should be P3

kv2019i, 2023-11-14:
@marc-hb @plbossart Discussed last time here:
https://github.com/zephyrproject-rtos/zephyr/pull/52048

"OK for now, but we need to rework this completely in the future."

kv2019i, 2023-11-16:
No owner and P3, moving to v2.9.

kv2019i, 2024-03-04:
Stable-v2.9 branched, this didn't make the cut, bumping to 2.10.

kv2019i, 2024-09-05:
No owner -> v2.12

kv2019i, 2024-12-13:
P3 bug that has not seen activity since 2022, I'll close this as WONTFIX.

closed 2024-12-13 by kv2019i