thesofproject/sof #8357: [BUG] topology2/rtc-aec: missing include file

plbossart, 2023-10-19:
**Describe the bug**
error in topology build ....

**To Reproduce**
./sof/scripts/docker-run.sh ./scripts/build-tools.sh

**Reproduction Rate**
100%

**Expected behavior**
no warnings on topology

**Impact**
annoyance or showstopper, who knows what is generated

**Environment**
    * SOF: 054003f2926b835d7e7ca2db8e602a7f2f9faeab

**Screenshots or console output**
````
[584/601] Generating sof-tgl-nocodec-rtcaec.tplg
ALSA lib conf.c:838:(get_char_skip_comments) Cannot access file include/components/google-rtc-aec/rtc-aec-blob.conf
saving nhlt as binary in nhlt-sof-tgl-nocodec-rtcaec.bin
````

plbossart, 2023-10-19:
@singalsu @ranj063 fyi

singalsu, 2023-10-20:
Seems I have missed to include this blob conversion from tplg1 to 64fcebbd64bf8e8be4ec7098215d7281fa6a4791. Thanks for reporting @plbossart !

singalsu, 2023-10-20:
Actually it seems that commit 9e9120a1549f48e8365a90a6bb0b65d9b36b5456 removed the file.

But I think it's a mistake, the blob should be duplicated from tplg1 to tplg2: 

https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/m4/google_rtc_audio_processing_default.m4

marc-hb, 2023-10-20:
Why is this error not fatal? As of version 1.2.10, `alsatplg` returns success status `0` and generates a topology file anyway!

Much more than an SOF bug, this looks like a serious `alsatplg` include bug. Who knows what other includes are missing in other files and what content gets generated. I know error handling is rarely ever tested but this seems to go a bit too far.

plbossart, 2023-10-20:
@ranj063 any comments?

ranj063, 2023-11-06:
> Actually it seems that commit [9e9120a](https://github.com/thesofproject/sof/commit/9e9120a1549f48e8365a90a6bb0b65d9b36b5456) removed the file.
> 
> But I think it's a mistake, the blob should be duplicated from tplg1 to tplg2:

@singalsu thats not correct. The blob was needed for IPC3 to set the audio format for the reference. But with IPC4 we send all audio formats with the init module instance IPC. So, no it wasnt a mistake. It is the right thing to do

singalsu, 2023-11-06:
Thanks, I'll do another fix, just not reference the missing blob in test topology.

marc-hb, 2023-11-10:
> Why is this error not fatal? As of version 1.2.10, alsatplg returns success status 0 and generates a topology file anyway!

Ping?

singalsu, 2024-02-08:
This was fixed by #8453.

PR #8453 (2023-11-09): Tools: Topology2: Remove bytes control from cavs-nocodec-rtcaec.conf
It was confirmed that the IPC4 version of Google RTC AEC is not using the bytes control, so it can be safely removed from topology.

This fixes topologies build error:

ALSA lib conf.c:838:(get_char_skip_comments)
Cannot access file include/components/google-rtc-aec/rtc-aec-blob.conf

Fixes: 	64fcebbd64bf8e8be4ec7098215d7281fa6a4791
	("Tools: Topology2: Add nocodec topology to test google-rtc-aec")

Fixes:	#8357
commit 304dfd79 (2023-11-09): Tools: Topology2: Remove bytes control from cavs-nocodec-rtcaec.conf

It was confirmed that the IPC4 version of Google RTC AEC
is not using the bytes control, so it can be safely removed
from topology.

This fixes topologies build error:

ALSA lib conf.c:838:(get_char_skip_comments)
Cannot access file include/components/google-rtc-aec/rtc-aec-blob.conf

Fixes: 	64fcebbd64bf8e8be4ec7098215d7281fa6a4791
	("Tools: Topology2: Add nocodec topology to test google-rtc-aec")

Fixes:	#8357
closed 2024-02-08 by singalsu