thesofproject/sof #8380: [FEATURE] Switch to Zephyr native drivers on i.MX platforms

dbaluta, 2023-10-24:
**Is your feature request related to a problem? Please describe.**

Completely switch to Zephyr native drivers on i.MX platforms.

Supported boards:

- [ ] i.MX8QM 
- [ ] i.MX8QXP
- [ ] i.MX8MP
- [ ] i.MX8ULP
- [ ] i.MX93

Issue tracking overall work from community is here: https://github.com/thesofproject/sof/issues/5794

**Describe the solution you'd like**

IPs drivers:
- DMA
  - [ ] SDMA
  - [ ] EDMA
- DAI
  - [ ] SAI
  - [ ] ESAI
- [ ] IRQSteer - https://github.com/zephyrproject-rtos/zephyr/pull/62776
- [ ] Messaging Unit / IPC

lgirdwood, 2023-10-24:
> Open questions:
> 
> * should we use IPC3 or IPC4?

I would recommend IPC4 as that's what's had most integration, validation and testing effort.  i.e. most APIs around IP integration already have glue to IPC4.

lgirdwood, 2023-10-24:
Adding @nashif and @andyross

kv2019i, 2024-03-04:
@dbaluta Ok to bump this to v2.10 ? I think lot of progress has been done, so maybe add the ticket with current status and then bump the milestone tag...?

dbaluta, 2024-03-04:
@kv2019i  do we have a timeline for v2.9 ? I think we are in the final steps of merging support for imx8/8x/8ulp and later 8mp for this. We estimate around 1 month for this.

No problem for us to bump this for v2.10. I will post a table with the progress here.

dbaluta, 2024-03-04:
OK, bump this to v2.10

kv2019i, 2024-06-05:
@dbaluta v2.10 branched today, can you give a status update and move to 2.11 if there's work left?

dbaluta, 2024-06-05:
Move to v2.11 for now.

kv2019i, 2024-09-13:
@dbaluta  To v2.12? Stable branch is today (#9360).

dbaluta, 2024-09-16:
@kv2019i  yes, moved it.

kv2019i, 2024-12-13:
Feature cutoff for v2.12, moving this to v2.13. imx8m still in hybrid mode, https://github.com/thesofproject/sof/pull/9608 pending. @dbaluta Would be nice to get this done in 2.13 cycle. The imx8m keeps tripping me over :) e.g. yesterday in https://github.com/thesofproject/sof/pull/9719 it's the only fail I have left.

dbaluta, 2024-12-18:
@kv2019i it will be in v2.13. We spent a lot of time merging zephyr changes.

dbaluta, 2025-02-20:
Changes merged and queued for v2.13

closed 2025-02-20 by dbaluta