thesofproject/sof #8396: [BUG] SSP tplg2 built with upstream alsa-utils fail

kv2019i, 2023-10-26:
**Describe the bug**
With merge of https://github.com/alsa-project/alsa-utils/pull/226 , any targetting Intel devics with SSP, will no longer work with upstream SOF driver.

Following change is needed in kernel to use the upstream-built topologies:
```
diff --git a/sound/soc/sof/ipc4-topology.c b/sound/soc/sof/ipc4-topology.c index 6139a122d337..7d09710c5a90 100644
--- a/sound/soc/sof/ipc4-topology.c
+++ b/sound/soc/sof/ipc4-topology.c
@@ -1367,7 +1367,7 @@ static int snd_sof_get_nhlt_endpoint_data(struct snd_sof_dev *sdev, struct snd_s
 	/* find NHLT blob with matching params */
 	cfg = intel_nhlt_get_endpoint_blob(sdev->dev, ipc4_data->nhlt, dai_index, nhlt_type,
 					   bit_depth, bit_depth, channel_count, sample_rate,
-					   dir, 0);
+					   dir, 4);
 
 	if (!cfg) {
 		dev_err(sdev->dev,

```

kv2019i, 2023-10-26:
FYI @ujfalusi @plbossart @brentlu @ranj063 @lgirdwood

kv2019i, 2023-10-26:
P2 as we don't have this alsa-utils patch in our toolchain we use to create SOF releases (yet).

ujfalusi, 2023-10-26:
@marc-hb, FYI

marc-hb, 2023-10-26:
Comment https://github.com/alsa-project/alsa-utils/pull/226#issuecomment-1668814096 said "this was only a preview that can't be merged yet, please revert it". Was it not reverted?

mengdonglin, 2023-10-30:
@jsarha @ranj063 Can you help look at this issue?

btian1, 2023-11-07:
1. https://github.com/alsa-project/alsa-utils/pull/226，this was closed not merge.
2. local tplg2 built file will failed to launch with latest kernel.

lgirdwood, 2023-11-10:
@jsarha @ranj063 did anyone get a chance to check this ?

lgirdwood, 2023-11-10:
Adding to v2.8 for tracking, although not a v2.8 blocker

jsarha, 2023-11-19:
Is this bug still valid. I see that https://github.com/thesofproject/linux/pull/4667 was merged, even if https://github.com/alsa-project/alsa-utils/pull/226 was not reverted.

At least I saw no problems when I built alsa-utils from upstream master (with alsa-lib from v1.2.10 tag), and used the produced alsatplg to build our topologies sof main, and the used the produced sof-mtl-nocodec.tplg with our daily build on a Nest RVP, and tried both "aplay -D hw:0,0 -vv -d 5 -f cd /dev/zero" and "arecord -D hw:0,0 -vv -d 5 -f cd /dev/null".

Is there something more to this bug, or should this be closed?

lgirdwood, 2023-11-23:
@kv2019i still valid with latest update from @jsarha ?

kv2019i, 2023-11-23:
@lgirdwood @jsarha Not sure. Now the problem is that unless you have a very, very new kernel (i.e. latest sof-dev), a topology built with upstream alsa-utils will NOT work. 

If we need to update the docker image for any reason, this requirement will kick in for SOF official releases as well, and bump the minimum required Linux kernel version to v6.7+ something. 

I'm not comfortable with this.

lgirdwood, 2023-11-29:
We either submit a revert to alsa-utils repo (and possibly request a point release) OR submit the above to Linux as a "Fix" patch for backporting.

If alsa-utils has NOT released this update then easiest is to revert and fix in alsa-utils ?

kv2019i, 2023-12-07:
No more comments, so I'll close this via https://github.com/thesofproject/linux/pull/4667 . This is now sent upstream and we need to ensure it gets backported - FYI to @ujfalusi . This only affects I2S configs and if hit, happens 100% so hard to miss.

closed 2023-12-07 by kv2019i