thesofproject/sof #8414: [BUG][LNL] DSP panic during the stress test

keqiaozhang, 2023-10-30:
**Describe the bug**
This issue happens on LNL-NOCODEC platforms. It runs to DSP panic after dozens of recording tests. The reproduction rate is almost 100%.

**dmesg**
```
[ 2519.683135] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx done : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 2519.683141] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4 set pipeline instance 0 state 4
[ 2519.683145] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[ 2519.686193] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 2519.686197] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2519.686201] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: DSP panic!
[ 2519.686203] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 2519.686216] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[ 2519.686218] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM debug status: 0x0, ROM debug error: 0x0
[ 2519.686221] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM feature bit enabled
[ 2519.686252] kernel: snd_sof:sof_ipc4_find_debug_slot_offset_by_type: sof-audio-pci-intel-lnl 0000:00:1f.3: Slot type 0x4c455400 is not available in debug window
[ 2519.686254] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 2519.686256] kernel: snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 7 -> 8
[ 2519.686321] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 2520.188239] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc timed out for 0x13000004|0x0
[ 2520.188251] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 2520.188255] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2520.188285] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x0|0x0, target: 0x1b0a0000|0x0|0x0, ctl: 0x3
[ 2520.188289] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2520.188291] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: IPC timeout
[ 2520.188315] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at soc_dai_trigger on SSP2 Pin: -110
[ 2520.188323] kernel:  Port2: ASoC: error at dpcm_be_dai_trigger on Port2: -110
[ 2520.188327] kernel:  Port2: ASoC: trigger FE cmd: 1 failed: -110
```

**mtrace**
```
[   75.929828] <inf> ipc: ipc_cmd: rx	: 0x13000003|0x0
[   75.930236] <inf> ipc: ipc_cmd: rx	: 0x13000004|0x0
[   75.930285] <err> os: xtensa_excint1_c:  ** FATAL EXCEPTION
[   75.930296] <err> os: xtensa_excint1_c:  ** CPU 0 EXCCAUSE 6 (divide by zero)
[   75.930298] <err> os: xtensa_excint1_c:  **  PC 0xa0061c98 VADDR (nil)
[   75.930301] <err> os: xtensa_excint1_c:  **  PS 0x60c20
[   75.930305] <err> os: xtensa_excint1_c:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:12 CALLINC:2)
[   75.930308] <err> os: z_xtensa_dump_stack:  **  A0 0xa00702a7  SP 0xa00e1e80  A2 (nil)  A3 0xa00c60e8
[   75.930311] <err> os: z_xtensa_dump_stack:  **  A4 0x2  A5 0xa00c60e8  A6 0x1  A7 0xa00e1fc0
[   75.930315] <err> os: z_xtensa_dump_stack:  **  A8 0xa0061c7c  A9 0xa00e1e60 A10 (nil) A11 0xa0100540
[   75.930316] <err> os: z_xtensa_dump_stack:  ** A12 (nil) A13 (nil) A14 0x1 A15 0x1
[   75.930320] <err> os: z_xtensa_dump_stack:  ** LBEG 0xa0057ab3 LEND 0xa0057ac3 LCOUNT 0xa0057973
[   75.930323] <err> os: z_xtensa_dump_stack:  ** SAR 0x20

Backtrace:0xa0061c95:0xa00e1e80 0xa00702a4:0xa00e1ea0 0xa0070721:0xa00e1ec0 0xa006e550:0xa00e1ee0 0xa006dcae:0xa00e1f10 0xa006dba5:0xa00e1f40 0xa006258d:0xa00e1f80 0xa0062fdb:0xa00e1fa0 0xa0062f69:0xa00e1fc0 0xa0062d91:0xa00e2030 0xa003afb0:0xa00e2070 0xa00635d6:0xa00e20a0 0xa005a87f:0xa00e20f0 

[   75.930413] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[   75.930418] <err> os: z_fatal_error: Current thread: 0x400f8d78 (unknown)
[   75.932920] <err> zephyr: k_sys_fatal_error_handler: Halting system
Terminated
```

**To Reproduce**
~/sof-test/test-case/check-capture.sh -d 1 -l 100 -r 1

**Reproduction Rate**
Almost 100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[7ee647149d40](https://github.com/thesofproject/linux/commit/7ee647149d40f49c28f63dc2cec62b2c559ac382)}
    * SOF: {[31140b132c1c](https://github.com/thesofproject/sof/commit/31140b132c1cbffde524b767954670fe49a5fdad)}
2) Name of the topology file
    * Topology: {development/sof-lnl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL-RVP-NOCODEC}

[dmesg.txt](https://github.com/thesofproject/sof/files/13202177/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/13202178/mtrace.txt)

abonislawski, 2023-10-31:
@jxstelter please take a look if this is any known issue and we will assign someone if required

jxstelter, 2023-11-09:
Can you provide FW symbol file exactly from failing run? I would like to corelate with exception data.

jxstelter, 2023-11-20:
The crash dump points to location where according to provided symbol file the EQFIR uuid = "43A90CE7-f3A5-41Df-AC06-BA98651AE6A3" module code is located. However I do not see it's sings in captured log. 

Can you repeat the test capturing IPC payloads and provide all binaries from the build including symbols and map files? 
Is the issue ends always with the same exception:
 75.930285] <err> os: xtensa_excint1_c:  ** FATAL EXCEPTION
[   75.930296] <err> os: xtensa_excint1_c:  ** **CPU 0 EXCCAUSE 6 (divide by zero)**
[   75.930298] <err> os: xtensa_excint1_c:  **  PC 0xa0061c98 VADDR (nil)
[   75.930301] <err> os: xtensa_excint1_c:  **  PS 0x60c20

lgirdwood, 2023-11-23:
@jxstelter did a quick grep in the FIR for / and % but empty, must be an inline or macro from generic convenience APIs. as this is all division in macros/inlines.
```
lrg@sif:~/work/sof/sof$ grep -rn " \/ " src/audio/eq_fir/
lrg@sif:~/work/sof/sof$ grep -rn " \/ " src/include/
src/include/user/eq.h:146:	(sizeof(struct sof_eq_iir_header) / sizeof(int32_t))
src/include/user/eq.h:152:	(sizeof(struct sof_eq_iir_biquad) / sizeof(int32_t))
src/include/user/fir.h:30:	(sizeof(struct sof_fir_coef_data) / sizeof(int16_t))
src/include/sof/trace/trace.h:511:/** Direct, low-level access to mbox / shared memory logging when DMA
src/include/sof/trace/preproc.h:15: * directly / independently.
src/include/sof/drivers/dmic.h:433:	((int64_t)-LOGRAMP_START_DB * DMIC_UNMUTE_RAMP_US / 1000))
src/include/sof/common.h:101:#define SOF_DIV_ROUND_UP(val, div) (((val) + (div) - 1) / (div))
src/include/sof/common.h:111:#define ARRAY_SIZE(x) (sizeof(x) / sizeof((x)[0]))
src/include/sof/samples/audio/kwd_nn/kwd_nn_config.h:12:#define KWD_NN_MS_TO_SAMPLES(samplerate, time) ((samplerate) * (time) / 1000)
src/include/sof/math/window.h:16:#define WIN_BLACKMAN_A0 Q_CONVERT_FLOAT(7938.0 / 18608.0, 15) /* For "exact" blackman */
src/include/sof/math/numbers.h:48:	c = a / b;
src/include/sof/math/cordic.h:17: * Inverse CORDIC Gain,inverse_cordic_gain = 1 / cordic_gain
src/include/sof/math/auditory.h:47:	int half_fft_bins; /**< In, fft_bins / 2 + 1 */
src/include/sof/math/auditory.h:64: * hz = 700 * (exp(mel / 1126.9941805389) - 1)
src/include/sof/audio/coefficients/fft/twiddle_16.h:16:/* in Q1.15, generated from cos(i * 2 * pi / FFT_SIZE_MAX) */
src/include/sof/audio/coefficients/fft/twiddle_16.h:1044:/* in Q1.15, generated from sin(i * 2 * pi / FFT_SIZE_MAX) */
src/include/sof/audio/coefficients/fft/twiddle_32.h:16:/* in Q1.31, generated from cos(i * 2 * pi / FFT_SIZE_MAX) */
src/include/sof/audio/coefficients/fft/twiddle_32.h:1044:/* in Q1.31, generated from sin(i * 2 * pi / FFT_SIZE_MAX) */
src/include/sof/audio/coefficients/up_down_mixer/up_down_mixer.h:14:#define COMPUTE_COEFF_32BIT(counter, denominator) ((0x7fffffffULL * (counter)) / (denominator))
src/include/sof/audio/coefficients/up_down_mixer/up_down_mixer.h:15:#define COMPUTE_COEFF_16BIT(counter, denominator) ((0x7fffULL * (counter)) / (denominator))
src/include/sof/audio/coefficients/volume/windows_fade.h:13:	Q_CONVERT_FLOAT(1.0 / (WINDOWS_FADE_LUT_SIZE - 1), 30)  /* Q2.30 */
src/include/sof/audio/module_adapter/module/generic.h:196:	/* number of sinks / sources and (when in use) input_buffers / input_buffers */
src/include/sof/audio/module_adapter/iadk/processing_module_interface.h:160:	 * SetConfiguration() / GetConfiguration().
src/include/sof/audio/asrc/asrc_farrow.h:176:					  /*!< data arrives (interleaved / */
src/include/sof/audio/kpb.h:39:#define	KPB_SAMPLES_PER_MS (KPB_SAMPLNG_FREQUENCY / 1000)
src/include/sof/audio/kpb.h:42:#define KPB_MAX_BUFFER_SIZE(sw, channels_number) ((KPB_SAMPLNG_FREQUENCY / 1000) * \
src/include/sof/audio/kpb.h:43:	(KPB_SAMPLE_CONTAINER_SIZE(sw) / 8) * KPB_MAX_BUFF_TIME * \
src/include/sof/audio/kpb.h:51:	((bytes) / ((KPB_SAMPLE_CONTAINER_SIZE(sample_width) / 8) * \
src/include/sof/audio/audio_stream.h:385:	uint32_t frame_num = byte_align / gcd(byte_align, frame_size);
src/include/sof/audio/audio_stream.h:388:	return frame_align_req * frame_num / gcd(frame_num, frame_align_req);
src/include/sof/audio/audio_stream.h:945:	return bytes / frame_bytes;
src/include/sof/audio/cadence/src_pp/xa_src_pp_api.h:100:xa_codec_func_t xa_src_pp; // Standard SRC / SRCPlus
src/include/sof/audio/cadence/aac_dec/xa_aac_dec_api.h:331:(Refer Section C.4.2.13 / C.4.2.14 and C.5.2.4 of ETSI TS 101 154 V1.9.1 document )
src/include/sof/audio/format.h:71:#define Q_CONVERT_QTOF(x, ny) ((float)(x) / ((int64_t)1 << (ny)))
src/include/ipc4/fw_reg.h:20: * \brief IPC4 fw registers in mailbox for host. Fw exposes dsp / fw
src/include/ipc4/fw_reg.h:143:/* FW Registers exposes additional DSP / FW state information to the driver. */
src/include/ipc4/fw_reg.h:145:	/* Current ROM / FW status(at 0x0) */
src/include/ipc4/fw_reg.h:148:	/* Last ROM / FW error code(at 0x4). */
src/include/ipc4/module.h:194:			/**< ModuleMsg::BIND / UNBIND. */
src/include/ipc4/module.h:230:			/**< ModuleMsg::LARGE_CONFIG_GET / LARGE_CONFIG_SET */
src/include/ipc4/module.h:246:			/**< param type : VENDOR_CONFIG_PARAM / GENERIC_CONFIG_PARAM */
src/include/ipc4/module.h:263:			/**< ModuleMsg::LARGE_CONFIG_GET / LARGE_CONFIG_SET */
src/include/ipc4/module.h:279:			/**< param type : VENDOR_CONFIG_PARAM / GENERIC_CONFIG_PARAM */
src/include/ipc4/asrc.h:91: *		round_up( 115 * 48000 / 22050 ) * 8 + 1 * 8 = 251 * 8 + 8 = 2016
src/include/ipc4/asrc.h:126:	 *	a. 0 1 - Playback mode / PUSH mode
src/include/ipc4/asrc.h:127:	 *	b. 1 0 - Capture mode / PULL mode
```

jxstelter, 2023-11-23:
Therefore we need logs with IPC payload, to try to reproduce locally.

keqiaozhang, 2023-11-24:
dmesg log with IPC payload:
```
[  241.982462] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: open stream 0 dir 0
[  241.982468] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: period min 192 max 16384 bytes
[  241.982471] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: period count 2 max 16
[  241.982472] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: buffer max 65536 bytes
[  241.982605] kernel: snd_sof:sof_ipc4_pcm_dai_link_fixup: sof-audio-pci-intel-lnl 0000:00:1f.3: Set NoCodec-0 to 32 bit format
[  241.982670] kernel: snd_sof_intel_hda_common:generic_calc_stream_format: sof-audio-pci-intel-lnl 0000:00:1f.3: format_val=0x41, rate=48000, ch=2, format=10
[  241.982679] kernel: snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  241.982691] kernel: snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1e0]=0x40000 successful
[  241.982716] kernel: snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1e0]=0x40000 successful
[  241.982721] kernel: snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-lnl 0000:00:1f.3: period_bytes:0x4000
[  241.982722] kernel: snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-lnl 0000:00:1f.3: periods:4
[  241.982735] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-lnl 0000:00:1f.3: copier host-copier.0.playback, type 23
[  241.982738] kernel: snd_sof:sof_ipc4_init_input_audio_fmt: sof-audio-pci-intel-lnl 0000:00:1f.3: matched audio format index for 48000Hz, 16bit, 2 channels: 0
[  241.982740] kernel: snd_sof:sof_ipc4_init_input_audio_fmt: sof-audio-pci-intel-lnl 0000:00:1f.3: Init input audio formats for host-copier.0.playback
[  241.982742] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-lnl 0000:00:1f.3: Pin index #0: 48000Hz, 16bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x11002) buffer size 192
[  241.982745] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-lnl 0000:00:1f.3: copier host-copier.0.playback: reference output rate 48000, channels 2 valid_bits 32
[  241.982747] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-lnl 0000:00:1f.3: Output audio format for host-copier.0.playback
[  241.982748] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-lnl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x12002) buffer size 384
[  241.982751] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-lnl 0000:00:1f.3: copier host-copier.0.playback, dma buffer: 2 ms (384 bytes)
[  241.982753] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-lnl 0000:00:1f.3: copier host-copier.0.playback, IPC size is 84
[  241.982754] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC value available in the firmware file's manifest (ibs/obs: 192/384)
[  241.982757] kernel: snd_sof:sof_ipc4_update_resource_usage: sof-audio-pci-intel-lnl 0000:00:1f.3: host-copier.0.playback: ibs / obs / cpc: 192 / 384 / 0
[  241.982759] kernel: snd_sof:sof_ipc4_init_input_audio_fmt: sof-audio-pci-intel-lnl 0000:00:1f.3: Init input audio formats for gain.1.1
[  241.982761] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-lnl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x12002) buffer size 384
[  241.982763] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: GAIN (UUID: 61BCA9A8-18D0-4A18-8E7B-2639219804B7): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  241.982764] kernel: snd_sof:sof_ipc4_update_resource_usage: sof-audio-pci-intel-lnl 0000:00:1f.3: gain.1.1: ibs / obs / cpc: 384 / 384 / 0
[  241.982766] kernel: snd_sof:sof_ipc4_init_input_audio_fmt: sof-audio-pci-intel-lnl 0000:00:1f.3: Init input audio formats for mixin.1.1
[  241.982767] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-lnl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x12002) buffer size 384
[  241.982769] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: MIXIN (UUID: 39656EB2-3B71-4049-8D3F-F92CD5C43C09): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  241.982770] kernel: snd_sof:sof_ipc4_update_resource_usage: sof-audio-pci-intel-lnl 0000:00:1f.3: mixin.1.1: ibs / obs / cpc: 384 / 384 / 0
[  241.982772] kernel: snd_sof:sof_ipc4_init_input_audio_fmt: sof-audio-pci-intel-lnl 0000:00:1f.3: Init input audio formats for mixout.2.1
[  241.982774] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-lnl 0000:00:1f.3: Pin index #0: 48000Hz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x12002) buffer size 384
[  241.982775] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: MIXOUT (UUID: 3C56505A-24D7-418F-BDDC-C1F5A3AC2AE0): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  241.982777] kernel: snd_sof:sof_ipc4_update_resource_usage: sof-audio-pci-intel-lnl 0000:00:1f.3: mixout.2.1: ibs / obs / cpc: 384 / 384 / 0
[  241.982778] kernel: snd_sof:sof_ipc4_init_input_audio_fmt: sof-audio-pci-intel-lnl 0000:00:1f.3: Init input audio formats for gain.2.1
[... cut ...]
[  242.495917] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[  242.496063] kernel:  Port0: ASoC: error at __soc_pcm_hw_params on Port0: -110
[  242.496124] kernel:  Port0: ASoC: error at dpcm_fe_dai_hw_params on Port0: -110
[  242.496145] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: free stream 0 dir 0
[  242.497523] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: close stream 0 dir 0
```

jxstelter, 2023-11-27:
The original log attached at the top contains flow with operations like SET_PIPELINE_STATE, MOD_BIND and MOD_UNBIND. The log with IPC payload is much shorter and there is even no single SET_PIPELINE_STATE? 
Is this log from whole test or just last operations?

I created python test from that log and run more than 2000 iterations without any failure. But I am not sure if it replies the original failure flow.

keqiaozhang, 2023-11-28:
> Is this log from whole test or just last operations?

This log is based on the latest test, comparing with the original log, there're some differences, but please refer the new log for debugging.

jxstelter, 2023-12-04:
Can you provide remote access to the setup and BKM's how to reproduce the failure? 
I have no reproduction with script created from your logs.

keqiaozhang, 2023-12-05:
@jxstelter please check issue https://github.com/thesofproject/sof/issues/8562 first. This is the latest error on LNL-NOCODEC when checking single playback/capture.

jxstelter, 2023-12-05:
The issue seems to be caused by inverted condition for SSP CPA check, that was introduced during some code improvements. 
Test code with fix available on my private branch:
https://github.com/jxstelter/sof/tree/lnl-rvp-d3-workarround
SOF: 27cdad0aa5efac94de50c73a7fb4b6e191e263b2

I tested reproduction scenario on this build many times without reproduction. 
Please verify if it on Your side.

lgirdwood, 2023-12-18:
@keqiaozhang @kv2019i can we close ?

keqiaozhang, 2023-12-19:
> @keqiaozhang @kv2019i can we close ?

Yes, this bug can be closed now.

closed 2023-12-19 by keqiaozhang