thesofproject/sof #8418: [BUG][ARL] no FW_READY at boot

kv2019i, 2023-10-30:
**Describe the bug**
FW build from latest SOF main does not boot on ARL system.

Last know good commit 98a20ce546ca09b36ef8639c058196e938b04a7a (tag: releases/mtl/v3.1.1, origin/mtl-003-drop-stable)

No FW_READY message is receive by kernel, FW state is FW_ENTERED, so ROM
has completed.

**To Reproduce**
Build firmware and boot system.

**Reproduction Rate**
100%

**Expected behavior**
FW should boot.

**Impact**
Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: -
    * SOF: v2.7.1, main, latest mtl-006-drop-stable
2) Name of the topology file
    * Topology: sof-hda-generic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: arl

kv2019i, 2023-10-30:
FYI @arunt1 @mengdonglin @ujfalusi

kv2019i, 2023-10-30:
First bisect results:
- v2.4 still work
- v2.5 fails to boot

kv2019i, 2023-10-30:
Closing as invalid, incorrect configuration targetted.

kv2019i, 2023-10-31:
Reopening, the Zephyr soc code has support for dynamic lookup of the core count, so the mtl built FW is expected to boot.

abonislawski, 2023-10-31:
Worth to check WOVCRO commits on 006 zephyr

kv2019i, 2023-10-31:
Base Zephyr smoke test boot up with SOF main, so this is potentially something on the SOF side. Still trying to get logs out or a proper panic dump.

kv2019i, 2023-10-31:
@abonislawski and @tmleman root-caused, addition of "watchdog2: watchdog@78500" is the thing that started causing this problem in this configuration.

abonislawski, 2023-10-31:
Great, so we just need to limit core count in watchdog

lgirdwood, 2023-11-10:
Tagged to v2.9 since not needed in v2.8

keqiaozhang, 2023-11-20:
This issue can be closed now.

closed 2023-11-20 by keqiaozhang