thesofproject/sof #8445: Stopping a CHAIN_DMA stream takes 100ms

plbossart, 2023-11-06:
Adding a Linux issue based on https://github.com/thesofproject/sof-test/pull/1126 comments

"ChainDMA enabled PCMs where the time to stop the stream takes more than 100ms (from trigger stop to PCM close)."

this needs to be root-caused, 100ms is not an acceptable response time.

@ujfalusi @lgirdwood we might transfer this issue to SOF at some point.

plbossart, 2023-11-06:
nice work @ujfalusi. The response time should not depend on the number of active CHAIN_DMA streams, that's a code flow problem clearly.

marc-hb, 2023-11-10:
There was one failure in daily tests with commit 1bd9e0d2c7a853d4c9, right before the #8451 fix was merged:
34373?model=TGLU_UP_HDA-ipc4&testcase=multiple-pause-resume-50

This failure was extremely rare. Rather than letting GitHub's auto-close this issue (anti--feature https://github.com/orgs/community/discussions/23476), let's make sure it does not happen for about a week and only then close it.

plbossart, 2024-07-23:
@marc-hb @kv2019i  I don't think this issue should remain open, no one has reported any updated in 8+ months.

PR #8451 (2023-11-09): schedule: cancel a task before freeing it
When a chain DMA stream is the only one running, stopping it currently takes 100ms. The reason is that the chain DMA driver is trying to stop the task directly from the IPC context instead of letting the task exit on its next scheduling event. The IPC thread is indeed trying to wait on a semaphore for the task to exit, but that doesn't work either, because zephyr_domain_unregister() terminates the thread before it signals the semaphore. To fix this we swap the order and signal the semaphore before terminating the thread. Fixes #8445
closed 2024-08-05 by lgirdwood