thesofproject/sof #8464: [BUG] Add CPC data for HDA generic for DMIC capture

kv2019i, 2023-11-08:
**Describe the bug**
Kernel emits the following warnings:

```
[  100.637496] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC value available in the firmware file's manifest (ibs/obs: 768/768)
[  100.637506] sof-audio-pci-intel-mtl 0000:00:1f.3: Please try to update the firmware.
[  100.637508] sof-audio-pci-intel-mtl 0000:00:1f.3: If the issue persists, file a bug at
[  100.637510] sof-audio-pci-intel-mtl 0000:00:1f.3: https://github.com/thesofproject/sof/issues/
[  100.637513] sof-audio-pci-intel-mtl 0000:00:1f.3: EQIIR (UUID: 5150C0E6-27F9-4EC8-8351-C705B642D12F): No CPC value available in the firmware file's manifest (ibs/obs: 768/768)
[  100.637517] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC value available in the firmware file's manifest (ibs/obs: 768/768)
[  100.637520] sof-audio-pci-intel-mtl 0000:00:1f.3: GAIN (UUID: 61BCA9A8-18D0-4A18-8E7B-2639219804B7): No CPC value available in the firmware file's manifest (ibs/obs: 768/768)
[  100.637524] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC value available in the firmware file's manifest (ibs/obs: 768/768)
```

**To Reproduce**
arecord -fS32_LE -c4 -r48000 -traw -Dhw:0,6 /dev/null

**Reproduction Rate**
100%

**Expected behavior**
No warning in kernel log.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev 20231108
    * SOF: SOF2.7.1
2) Name of the topology file
    * Topology: intel/sof-ace-tplg/sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel mtl

kv2019i, 2023-11-08:
FYI @mengdonglin @btian1 @andrula-song  @abonislawski

kv2019i, 2023-11-08:
This is fixed in SOF main with commit 352f01fee4ba0 , but filing this to track the issue as this can be seen in 2.7.1 released binaries.

kv2019i, 2023-11-08:
This is fixed in SOF main with commit https://github.com/thesofproject/sof/commit/352f01fee4ba0b1282091bb41dd08c51c353d79a , but filing this to track the issue as this can be seen in 2.7.1 released binaries.

btian1, 2023-11-13:
this is similar issue with #8465, fixed with PR merge: https://github.com/thesofproject/rimage/pull/175/ to main.
Release after Aug 28th should already have this fix, please check next release.