thesofproject/sof #8465: [BUG] Missing CPC data for HDA generic for HDA playback

kv2019i, 2023-11-08:
**Describe the bug**
Kernel emits the following warnings:
```
[  305.952146] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  305.952157] sof-audio-pci-intel-mtl 0000:00:1f.3: GAIN (UUID: 61BCA9A8-18D0-4A18-8E7B-2639219804B7): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  305.952160] sof-audio-pci-intel-mtl 0000:00:1f.3: MIXIN (UUID: 39656EB2-3B71-4049-8D3F-F92CD5C43C09): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  305.952163] sof-audio-pci-intel-mtl 0000:00:1f.3: MIXOUT (UUID: 3C56505A-24D7-418F-BDDC-C1F5A3AC2AE0): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  305.952166] sof-audio-pci-intel-mtl 0000:00:1f.3: GAIN (UUID: 61BCA9A8-18D0-4A18-8E7B-2639219804B7): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
[  305.952168] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC value available in the firmware file's manifest (ibs/obs: 384/384)
```

**To Reproduce**
aplay  -fS32_LE -c2 -r48000 -traw -Dhw:0,0 /dev/zero

**Reproduction Rate**
100%

**Expected behavior**
No warnings in kernel log.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev 20231108
    * SOF: SOF2.7.1
2) Name of the topology file
    * Topology: intel/sof-ace-tplg/sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform:  Intel mtl

kv2019i, 2023-11-08:
FYI @mengdonglin @btian1 @andrula-song @abonislawski @ujfalusi

btian1, 2023-11-09:
@kv2019i , main or mtl_006 branch?

kv2019i, 2023-11-09:
This bug occures with SOF2.7.1 released, but it seems this is fixed in both SOF main and mtl-006.

btian1, 2023-11-10:
> This bug occures with SOF2.7.1 released, but it seems this is fixed in both SOF main and mtl-006.

yes, it is not reproducible on SOF main with latest on MTL HDA rvp, dmesgshowed CPC are correctly loaded:
[image]

@kv2019i @mengdonglin , can we close it for now?

kv2019i, 2023-11-10:
@btian1 Thanks for double-checking, let's close but good to have this as a reference if any SOF2.7 user hits this. Fix is coming!

closed 2023-11-10 by kv2019i