thesofproject/sof #8476: [BUG] 2 abnormal sampling points during alsabat testing on MTL-HDA

keqiaozhang, 2023-11-13:
**Describe the bug**
This issue happens on MTL-006 branch, the reproduction rate is low, but it can be manually reproduced. No reproductions on main branch so far. 

[image]

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -p hw:sofhdadsp,0 -c hw:CODEC,0 -C 2 -F 821

**Reproduction Rate**
less than 10%.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {sof-dev/[453fdbbb8fce](https://github.com/thesofproject/linux/commit/453fdbbb8fced8d547e7fac1583eb485a5e0cf42)}
    * SOF: {mtl-006-drop-stable/https://github.com/thesofproject/sof/commit/df1a295fc81b534b5fdf1e33f2c4b018401d6212}
2) Name of the topology file
    * Topology: {sof-ace-tplg/sof-hda-generic-4ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL-RVP-HDA w/ Gen3.1 AIOC}

lgirdwood, 2023-11-13:
@keqiaozhang just at the start of the playback/capture or randomly during playback/capture ?

keqiaozhang, 2023-11-14:
> @keqiaozhang just at the start of the playback/capture or randomly during playback/capture ?

It happened during playback test. From the 2 waveforms, it happens between 0.46s~0.47s.

abonislawski, 2023-11-14:
@wszypelt please assign someone from validation team to check if we are testing exactly the same scenario in python tests (and have repro)

lgirdwood, 2023-11-14:
> > @keqiaozhang just at the start of the playback/capture or randomly during playback/capture ?
> 
> It happened during playback test. From the 2 waveforms, it happens between 0.46s~0.47s.

Interesting, Btw, what's the host side buffer size in this test ?

keqiaozhang, 2023-11-16:
> 
> Interesting, Btw, what's the host side buffer size in this test ?

Buffer size is 2 ms (384 bytes).

`kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier host-copier.0.playback, dma buffer: 2 ms (384 bytes)`

lrudyX, 2023-11-21:
@keqiaozhang please provide logs with payloads

lgirdwood, 2023-11-24:
> > Interesting, Btw, what's the host side buffer size in this test ?
> 
> Buffer size is 2 ms (384 bytes).
> 
> `kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier host-copier.0.playback, dma buffer: 2 ms (384 bytes)`

Ok, so this is the DSP side buffer and it looks correct. The aplay command line can dump the host side buffer size, my feeling is that the buffer size could be 0.46 seconds long.

kv2019i, 2023-11-29:
Only on the release branch, so removing the v2.8 milestone.