thesofproject/sof #8478: [BUG] LNL has FW reported error: 116 Invalid target code ID provided

fredoh9, 2023-11-13:
**Describe the bug**
LNLM_RVP_NOCODEC has new firmware error 116, Invalid target code ID provided.

```
[  245.761516] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-lnl 0000:00:1f.3: pipeline: 5 memory pages: 5
[  245.761520] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-lnl 0000:00:1f.3: Create widget pipeline.5 instance 0 - pipe 5 - core 2
[  245.761524] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x11000005|0x200000: GLB_CREATE_PIPELINE
[  245.761788] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x31000074|0x200000: GLB_CREATE_PIPELINE
[  245.761792] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: FW reported error: 116 - Invalid target code ID provided
[  245.764783] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc error for msg 0x11000005|0x200000
[  245.768322] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: failed to create module pipeline.5
[  245.772154] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Failed to set up connected widgets
[  245.776196] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed widget list set up for pcm 2 dir 0
[  245.780925] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  245.786529] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: free stream 2 dir 0
[  245.786806] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: close stream 2 dir 0
```

**To Reproduce**
Can be reproducible with simple playback
TPLG=/lib/firmware/intel/development/sof-lnl-nocodec.tplg MODEL=LNLM_RVP_NOCODEC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F

**Reproduction Rate**
100%

**Environment**

- Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
- Linux Commit: [453fdbbb8fce](https://github.com/thesofproject/linux/commit/453fdbbb8fced8d547e7fac1583eb485a5e0cf42)
- SOF Branch: [main](https://github.com/thesofproject/sof/tree/main)
- SOF Commit: [9eae757947d4](https://github.com/thesofproject/sof/commit/9eae757947d4c3a7cd127112d5d70ddcc514acda)
- TPLG: development/sof-lnl-nocodec.tpl

Similar issue is https://github.com/thesofproject/sof/issues/8275

jxstelter, 2023-11-14:
Are there any FW logs collected for this issue?

fredoh9, 2023-11-14:
Yes, wasn't sure we can upload any log. Can you access the internal link?

Intel internal daily test link:
planresultdetail/34523?model=LNLM_RVP_NOCODEC&testcase=check-playback-all-formats

softwarecki, 2023-11-15:
I'm unable to access to this link. Please grab logs with ipc payloads.

fredoh9, 2023-11-15:
@softwarecki I enabled IPC payload.
[dmesg.txt](https://github.com/thesofproject/sof/files/13369137/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/13369138/mtrace.txt)

keqiaozhang, 2023-11-20:
This issue happens after switching to multicore tplg.

softwarecki, 2023-11-20:
@keqiaozhang @fredoh9 Please indicate in the log the IPC that activates core 2, because I don't see it.

keqiaozhang, 2023-11-20:
> @keqiaozhang @fredoh9 Please indicate in the log the IPC that activates core 2, because I don't see it.

Here is the log when running audio test on core 2:
```
[  131.753095] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  131.753107] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  131.756653] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  131.756672] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[  131.756684] snd_sof_intel_hda_common:hda_dsp_state_log: sof-audio-pci-intel-lnl 0000:00:1f.3: Current DSP power state: D0I0
[  131.756701] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 0 -> 2
[  131.756705] snd_sof:snd_sof_load_firmware: sof-audio-pci-intel-lnl 0000:00:1f.3: loading firmware
[  131.756708] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 2 -> 3
[  131.757761] snd_sof_intel_hda_common:mtl_dsp_pre_fw_run: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x1010000 successful
[  131.758821] snd_sof_intel_hda_common:mtl_dsp_pre_fw_run: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1d1c]=0x8161 successful
[  131.758831] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-lnl 0000:00:1f.3: booting DSP firmware
[  131.758836] snd_sof_intel_hda_common:hda_dsp_cl_boot_firmware: sof-audio-pci-intel-lnl 0000:00:1f.3: IMR restore supported, booting from IMR directly
[  131.759887] snd_sof_intel_hda_common:mtl_dsp_core_power_up: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000101 successful
[  131.759894] snd_sof_intel_hda_common:mtl_dsp_cl_init: sof-audio-pci-intel-lnl 0000:00:1f.3: Primary core power up successful
[  131.759908] snd_sof_intel_hda_common:mtl_dsp_cl_init: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x73214]=0x80000000 successful
[  131.759919] snd_sof_intel_hda_common:mtl_enable_interrupts: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x41 successful
[  131.759928] snd_sof_intel_hda_common:mtl_enable_interrupts: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x1 successful
[  131.980659] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx      : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  131.980687] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 3 -> 6
[  131.980705] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx done : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  131.980786] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-lnl 0000:00:1f.3: firmware boot complete
[  131.980797] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 6 -> 7
[  131.980864] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[  131.981199] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x64000000|0x31400008: MOD_LARGE_CONFIG_SET
[  131.981313] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx done : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[  131.981324] Message payload: 00000000: 2c2a021e ffeddff1
[  131.981331] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  131.981837] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[  131.981995] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  131.982007] Message payload: 00000000: 00000100 00001000 00000001 000001ef
[  131.982009] Message payload: 00000010: 00000000 00000000 00000000 00000000
[  131.982011] Message payload: 00000020: 00000000 00000000 00000000 00000000
[  131.982014] Message payload: 00000030: 00000000 00000000 00000000 00000000
[  131.982016] Message payload: 00000040: 00000000 00000000 00000000
[  131.982036] snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: open stream 2 dir 0
[  131.982040] snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: period min 192 max 16384 bytes
[  131.982043] snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: period count 2 max 16
[  131.982046] snd_sof:sof_pcm_open: sof-audio-pci-intel-lnl 0000:00:1f.3: buffer max 65536 bytes
[  131.982481] snd_sof:sof_ipc4_pcm_dai_link_fixup: sof-audio-pci-intel-lnl 0000:00:1f.3: Set NoCodec-2 to 32 bit format
[  131.982518] snd_sof_intel_hda_common:generic_calc_stream_format: sof-audio-pci-intel-lnl 0000:00:1f.3: format_val=0x41, rate=48000, ch=2, format=10
[  131.982529] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: hw params stream 2 dir 0
[  131.982537] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1e0]=0x40000 successful
[  131.982563] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1e0]=0x40000 successful
[... cut ...]
[  134.687873] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  134.687878] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
[  134.688758] snd_sof_intel_hda_common:hda_dsp_state_log: sof-audio-pci-intel-lnl 0000:00:1f.3: Current DSP power state: D3
[  134.688766] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 7 -> 0
```

TPLG:
[image]

softwarecki, 2023-11-20:
@keqiaozhang Once again. There is no MOD_SET_DX in the log, which would activate core 2 before trying to create a pipeline on core 2 via GLB_CREATE_PIPELINE.

mengdonglin, 2023-11-21:
> @keqiaozhang Once again. There is no MOD_SET_DX in the log, which would activate core 2 before trying to create a pipeline on core 2 via GLB_CREATE_PIPELINE.

@RanderWang Can you help check this open on kernel side?

plbossart, 2023-11-22:
Can someone verify that PR thesofproject/linux/#4711 fixes this issue?

keqiaozhang, 2023-11-22:
> Can someone verify that PR [thesofproject/linux#4711](https://github.com/thesofproject/linux/pull/4711) fixes this issue?

Confirmed that https://github.com/thesofproject/linux/pull/4711 can fix this issue, but the code base of this PR is too old, need to rebase it first.

closed 2023-11-29 by plbossart