thesofproject/sof #8497: [BUG] HP Omen 15-ek0010ca Comet Lake - consitent gap in audio when sending audio to left channel, delayed audio starts

RustyClanker, 2023-11-18:
This issue has been plaguing me for a long time, I have scoured the Internet and can not find anything related or a solution that works. I have stripped the sound system down to just ALSA with the raw driver outputs (no Pulse, no Pipewire, no JACK) and the issue is still present. I can plug headphones into the headphone jack and the issue is not present. I have been very hesitant to open an issue on this, but I have run out of stuff to dig into or try.

Retasking of the pins does nothing, shutting off power save does not correct the issue, using legacy drivers vs SOF does nothing, looping a silent sound in the background does not fix it.

There is a delayed start to audio if sending through both left and right channels, as long as the sound stays constant (through both channels) it will plays without issue. Once the sound stops, after a few seconds the delay will happen again. This is not related to powering save or idling, as disabling either does nothing to correct it.

If I start up speaker-test and send audio only though the left channel, there is a consistent 1 second gap in the sound output (plays for 1 second, silence for one second). If I do the same through the right channel, the output is constant with no issue. Using the GNOME control panel and doing a sound test, either Front or Left or Front or Right is cut off or their is a cut in audio in the middle of one of the words. I believe where it is cutting is related to the timing of pressing the test button in relation to when the test button was pressed in previous tests (pressing it right away after the audio stops or waiting a few seconds).

If a put a constant audio stream up (I was using random music from soundcloud), and move the balance toward the left channel, it starts the 1 second gap interval not quite all the way to left on the slider.

At this point, it is either the hardware or the driver. Audio in Windows works without issue (left and right channel play audio with no gap interval). Booting into Windows and then back into Linux does not fix the issue. It also happens regardless of the distro, Ubuntu, Pop_OS, Manjaro, Endeavour, Elementary, Debian, and ones I've forgotten. I think the only one I didn't try was Fedora.

Right now I am using a fresh install of Manjaro, all of the below is with Pipewire running using wireplumber as the media session manager. Again, it running has no bearing on the issue, I can turn it all off and the issue persists.

- HP Omen 15-ek0010ca, hybrid GPU/Audio -> Intel & NVIDIA
- Firmware version F.15
- Only has the two front speakers, no center channel or rear speakers.
- Intel Core i7-10750H
- Intel UHD Graphics (CML GT2)
- NVIDIA GTX 1660 Ti Mobile
- sof-firmware 2023.09.1-1 [installed]
- linux-firmware 20231030.2b304bfe-1 [installed]
- 6.5.11-1-MANJARO #1 SMP PREEMPT_DYNAMIC

```
> dmesg | grep -E 'nvidia|skl|snd|sof'
[    2.617250] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.617354] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.617427] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    2.617532] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    2.928771] nvidia: loading out-of-tree module taints kernel.
[    2.928779] nvidia: module license 'NVIDIA' taints kernel.
[    2.928783] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[    2.928783] nvidia: module license taints kernel.
[    3.136677] nvidia-nvlink: Nvlink Core is being initialized, major device number 509
[    3.137416] nvidia 0000:01:00.0: enabling device (0000 -> 0003)
[    3.137509] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    3.174746] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.174842] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.295998] sof-audio-pci-intel-cnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.296111] sof-audio-pci-intel-cnl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.296124] sof-audio-pci-intel-cnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.296292] sof-audio-pci-intel-cnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.301109] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  535.129.03  Thu Oct 19 18:42:12 UTC 2023
[    3.336003] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    3.336005] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0
[    3.439461] sof-audio-pci-intel-cnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.510191] sof-audio-pci-intel-cnl 0000:00:1f.3: use msi interrupt mode
[    3.540223] sof-audio-pci-intel-cnl 0000:00:1f.3: hda codecs found, mask 5
[    3.540228] sof-audio-pci-intel-cnl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.540232] sof-audio-pci-intel-cnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    3.545758] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    3.545762] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    3.545768] sof-audio-pci-intel-cnl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    3.676251] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    3.676254] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    3.689259] sof-audio-pci-intel-cnl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[    3.689457] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
[    3.707159] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC245: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    3.707162] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.707163] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    3.707164] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    3.707165] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    3.707166] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    3.750001] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
[    3.751457] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    3.751463] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    3.751570] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
[    3.751574] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
[    3.751577] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
[... 132 lines cut ...]
			 10BitTagComp- 10BitTagReq- OBFF Via message, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
```

Let me know if anything else is required.

Thanks!

plbossart, 2023-11-20:
@RustyClanker Can you clarify this 

````
If I start up speaker-test and send audio only though the left channel, there is a consistent 1 second gap in the sound output (plays for 1 second, silence for one second). If I do the same through the right channel, the output is constant with no issue. 
````

if you use 'speaker-test -Dhw:0,0 -c2 -r48000 -FS16_LE -t sine' you'll get sine wave playing for one second on left and one second on right. That's the point of speaker-test, to make sure we hear what speaker is used.

From your description I wonder if it's actually the right speaker that's problematic and playing both L+R

RustyClanker, 2023-11-20:
I was using, 'speaker-test -Dhw:1,0 -c2 -s1' to test out just the left speaker with just the default noise. The noise was consistent out the right (if I use -s2) and gaps in audio out the left.

I used what you provided above (just changing the card, 0 is nvidia HDMI, 1 is Intel), it just sounds like a busy tone on a phone. When I plug my headphones into the jack, it's a consistent tone on both channels using the same command.

```
> speaker-test -D hw:1,0 -c 2 -r48000 -FS16_LE -t sine

speaker-test 1.2.10

Playback device is hw:1,0
Stream parameters are 48000Hz, S16_LE, 2 channels
Sine wave rate is 440.0000Hz
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16384
Period size range from 48 to 4096
Using max buffer size 16384
Periods = 4
was set period_size = 4096
was set buffer_size = 16384
 0 - Front Left
 1 - Front Right
Time per period = 5.632568
 0 - Front Left
 1 - Front Right
Time per period = 5.974272
 0 - Front Left
 1 - Front Right
Time per period = 5.972868
 0 - Front Left
 1 - Front Right
```

plbossart, 2023-11-20:
Sorry, I am not following. 

With the `speaker-test -D hw:1,0 -c 2 -r48000 -FS16_LE -t sine` command, you should hear a signal on only one of the two headphones at a time.

If yes, is the problem only with speakers?

if yes, can you please check with the legacy HDaudio driver, without SOF.

It's likely that you have a codec configuration issue and not an SOF driver issue. See https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#disable-sof-on-pci-hdaudio-devices-to-test-audio-playback

RustyClanker, 2023-11-20:
> Sorry, I am not following.
> 
> With the `speaker-test -D hw:1,0 -c 2 -r48000 -FS16_LE -t sine` command, you should hear a signal on only one of the two headphones at a time.
> 
> If yes, is the problem only with speakers?

The problem is with the internal speakers only, when running that command with the internal speakers I hear what sounds like a busy tone on a phone; this happens on both the left and right speaker. During the test, the audio starts out of the left speaker, 1 second on, 1 second off, 1 second on. It then switches to the right speaker and the same audio pattern, 1 second on, 1 second off, 1 second on.

My second test was to plug my headphones into the jack on the left side of my laptop and repeat the same command. During the test, the audio is a constant tone out of the left headphone earpiece. Then then switches to the right headphone earpiece and is a constant tone.

Hopefully this will be more clear:

Internal laptop speakers
 0 - Front Left    -> broken tone, 1 second on, 1 second off
 1 - Front Right -> broken tone, 1 second on, 1 second off

Headphones plugged into physical jack
 0 - Front Left    -> constant tone
 1 - Front Right -> constant tone

> if yes, can you please check with the legacy HDaudio driver, without SOF.
>
> It's likely that you have a codec configuration issue and not an SOF driver issue. See [...]

I had tried the legacy drivers before using dsp_driver=1, but I will do it again in order to run the above speaker-test vs what I had been running before and let you know the results. But having just typed that, I have a feeling you are right, the issue persisted when I bypassed the SOF drivers before.

RustyClanker, 2023-11-20:
So I rebooted with only the legacy drivers and while the right speaker now has a constant tone in the output, the left still has the broken tone. So you are correct, this is not a SOF issue.

Internal laptop speakers
0 - Front Left -> broken tone, 1 second on, 1 second off
1 - Front Right -> constant tone

```
> dmesg | grep -E 'nvidia|skl|snd|sof'

[    2.995154] snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.996444] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    2.996554] snd_hda_intel 0000:01:00.1: Disabling MSI
[    2.996564] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    3.289070] nvidia: loading out-of-tree module taints kernel.
[    3.289080] nvidia: module license 'NVIDIA' taints kernel.
[    3.289084] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[    3.289086] nvidia: module license taints kernel.
[    3.616869] nvidia-nvlink: Nvlink Core is being initialized, major device number 509
[    3.617879] nvidia 0000:01:00.0: enabling device (0000 -> 0003)
[    3.618030] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    3.742303] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  535.129.03  Thu Oct 19 18:42:12 UTC 2023
[    3.782192] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    3.782195] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0
[    3.899404] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.007178] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC245: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    4.007183] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.007185] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.007187] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[    4.007188] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[    4.007189] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19
[    4.237786] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
[    4.362208] nvidia-uvm: Loaded the UVM driver, major device number 507.
```

plbossart, 2023-11-21:
thanks for the clarifications @RustyClanker. The SOF driver provides a single output and there's nothing that changes the data pushed to the HDaudio codec. The speaker/headphone handling is entirely done in the codec.

If it's a periodic problem, then it's likely that there's some sort of timer that mucks with the output. Possibly a jack detection issue with a calibration gone bad.

Transferring this to Linux repo.