thesofproject/sof #8524: [FEATURE] Support ssp early start of MCLK/BCLK in IPC4, FW/tplg part

kv2019i, 2023-11-24:
**Is your feature request related to a problem? Please describe.**
FW counterpart to Linux kernel side issue https://github.com/thesofproject/linux/issues/4514

**Describe the solution you'd like**
Need definition of the IPC4 interface for this feature and potentially split work to Zephyr (for ssp driver) and alsa-utils (possible changes to tplg config blob generation).

kv2019i, 2023-11-24:
FYI @Vamshigopal @plbossart @mwasko @ranj063

kv2019i, 2023-11-24:
Based on quick look, the alsa-utils NHLT generator does fill 

Furthermore, current ssp driver for Intel hardware, enables MCLK and BCLK always in early mode, triggering the clocks when module is initialized. The driver does parse the aux part of the blob (that contains the controls to configure aspects of early clocks), but there is no functionality here, just a debug print:

zephyr/drivers/dai/intel/ssp/ssp.c
```
»       »       case SSP_DMA_CLK_CONTROLS_EXT:
»       »       »       ext = (struct ssp_intel_ext_ctl *)&aux_tlv->val;
»       »       »       LOG_INF("%s ext ext_data %u", __func__, ext->ext_data);
»       »       »       break;
```

plbossart, 2023-11-27:
And for the record the solution MUST scale to ACE2.0/LNL with the link DMA triggered by the host.

kv2019i, 2024-03-04:
Stable-v2.9 branched, this didn't make the cut, bumping to 2.10.

lrudyX, 2024-05-10:
@ujfalusi need you  to check if this is already done

ujfalusi, 2024-05-10:
@kv2019i, I'm genuinely surprised and don't get how this works in IPC4.
We unconditionally set SSP_CLK_MCLK_ES_REQ and SSP_CLK_BCLK_ES_REQ in `dai_ssp_set_config_blob()` but for the BCLK to be really running the port also needs to be enabled, which is only done at start.

@lrudyX, I don't think anyone worked on this feature and I think it is not supported.

kv2019i, 2024-06-05:
@marcinszkudlinski No update, clearing v2.10 milestone.

kv2019i, 2024-12-13:
Feature cutoff for v2.12. No owner/activity in 2.10/2.11 cycles, so clearing the milestone for now.