thesofproject/sof #8543: Noise in the left microphone input

Omrigan, 2023-11-28:
**Describe the bug**
Hi!

I have a Lenovo ThinkPad X1 Extreme Gen 5. The issue is that on the left microphone input I hear a loud noise, which has a period of 22ms.
[image]

If I mute the left channel, on the right the voice is fine.

The laptop recently had some issues with sound, but:
1. It is different
2. Seem to be fixed by now
[1]: https://forums.lenovo.com/t5/Other-Linux-Discussions/X1-Extreme-and-P1-gen5-sound-issues-on-Linux/m-p/5221589?page=3

**To Reproduce**
pw-record test.mp3

**Reproduction Rate**
Always

**Expected behavior**
No noise.

Or, can you help me with using the right input channel as a mono mic in alsa/pipewire, if the solution for this is not found? 

**Impact**
Mic is unusable.

**Environment**
Arch linux, stock kernel and sof

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.6.2-arch1-1
    * Firmware info: version 2:2:0-57864

I tried `options snd_sof_intel_hda_common dmic_num=4`, but the additional two dmics are silent except for the short beep in the beginning of the recording.

**Screenshots or console output**
```
sudo dmesg | grep sof
[ 1854.181051] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[ 1854.181114] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[ 1854.181326] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[ 1854.430901] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[ 1854.612600] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[ 1854.644654] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[ 1854.644656] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[ 1854.644658] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[ 1854.648102] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[ 1854.648104] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[ 1854.648106] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[ 1854.754553] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[ 1854.754556] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[ 1854.773628] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[ 1854.862943] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input27
[ 1854.862967] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input28
[ 1854.862985] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input29
[ 1854.863003] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input30
[ 1854.863019] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input31
```

lgirdwood, 2023-11-29:
@Omrigan it looks like the NHLT table (the ACPI table that describes the DMIC and I2S HW) is telling the kernel we have 4 DMICs on the laptop but that should not stop stereo recording. 

@kv2019i @ujfalusi @plbossart this is on RPL with SOF v2.2.

plbossart, 2023-11-29:
@Omrigan if you have Windows installed, it's worth checking if you see the same behavior, just to rule out a hardware issue.

What is really surprising is that if this was a 2-mic device, the two mics would share the same data pin, so if there was any sort of interference at the pin level both channels would be corrupted.

It wouldn't hurt to remove all options, add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/13504401/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the dmesg log.

Also try to record with arecord -Dhw:0,1 -c4 -r48000 -fS32_LE -d 20 test.wav and attach the file or the waveform. I am not following how a 4ch streams was reduced to 2ch.

Omrigan, 2023-12-02:
> @Omrigan if you have Windows installed, it's worth checking if you see the same behavior, just to rule out a hardware issue.

I don't have at the moment, but I plan to install it eventually. Once I get around to it, I will test.

In the meantime, I've done the following workaround:
```~> cat .config/pipewire/pipewire.conf.d/mic.conf
context.modules = [
    # plenty of existing { ... } blocks, then paste this:
    # Alternate microphone-only mono source
    {   name = libpipewire-module-loopback
        args = {
            node.description = "Mono Mic"
            capture.props = {
                node.name = "capture.Mono_Mic"
                audio.position = [ FR ]
                stream.dont-remix = true
                node.target = "alsa_input.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp_6__source"
                node.passive = true
            }
            playback.props = {
                node.name = "capture.Mono_Mic"
                media.class = "Audio/Source"
                audio.position = [ MONO ]
            }
        }
    }
]```

closed 2025-07-24 by github-actions[bot]